nRF24/RF24 #55: Data (float, char) becomes garbled when received on Due

NeoTokyoNori, 2015-01-03:
I have a sketch that has been working properly on Uno, but is not working on Due, 
Where the data (float, char) becomes garbled when received on Due. 

I am sending the following struct which includes IMU data plus some chars,
and receiving it with the following receive function. 

typedef struct{
  byte PIPE;
  float YAW;
  float PCH;
  char MSG[15]; 
}
A_t;
A_t RFL_STRCT;  

void NRF_RECEIVE(){

    if ( radio.available(&pipe_num) )  {

            radio.read( &RFL_STRCT, sizeof(RFL_STRCT) );

            RFL_YAW_STR = String(RFL_STRCT.YAW);
            RFL_PCH_STR = String(RFL_STRCT.PCH);
            RFL_PRX_MSG = String(RFL_STRCT.MSG);

            Serial.print("RFL_YAW_STR: ");
            Serial.println(RFL_YAW_STR);
            Serial.print("RFL_PRX_MSG: ");
            Serial.println(RFL_PRX_MSG);

	}

On the UNO, I receive the data properly formatted as expected, like so;
RFL_YAW_STR: -168.25
RFL_PCH_STR :  132.65
RFL_PRX_MSG: GTRK,

But on the DUE, I get the same exact data showing up garbled or wrongly terminated/formatted.
RFL_YAW_STR: 0.00
RFL_PCH_STR : 227634315264.00
RFL_PRX_MSG: K,

The sketch is only changed where necessary, so that it compiles and runs on Due.

TMRh20, 2015-01-03:
1. **First rule of troubleshooting code:**  *Always* assume user error first, 99% of problems are pebcak and id_10T related errors  
2. Second rule of troubleshooting code: Don't ask the developer unless you went through this list and then some. See rule no.1. Provide all the results and info resulting from this list  
3. Test with known working code, see rules 1,2  
4. Test the totally obvious and provide the results with your inquiry, see rules 1,2  
    ie: I can do x by modifying known working code....
5. Test likely things: ```Serial.println...``` , see rules 1,2  
6. Search [google](google.com), see rules 1,2  
7. Consult various forums and online support communities , see rules 1,2  
8. Ask your neighbor, see rules 1,2  
9. Have children? Ask them, they are usually smarter than you. No, *really*, see rules 1,2  
10. Religious? Maybe pray. see rules 1,2  
11. Consider the fact that, unless you provide a bug report or question with some reasonable amount of info, you will either be flamed or met with considerable sarcasm. Since you have not provided any information related to troubleshooting the issue, I can only assume you have not, and if you couldn't be bothered to consider simple troubleshooting steps, why would I? , see rules 1,2  
12. Last but not least, get a bigger hammer. see rules 1,2  

Hint: https://www.youtube.com/watch?v=FClGhto1vIg

Note: I will post the solution, if you provide some indication that you have done some troubleshooting as per the above list, but I think you will figure it out by then...  ;)

mz-fuzzy, 2015-01-03:
You pass a structure into RF24 calls. Serializing might be tricky. Implicitly serialized structure could look different on different platforms. 

For this particular issue - my guess is that arm gcc aligns structure members differently than avr does. Try adding this into structure definition this to both platform's sketch to avoid aligning:
```
typedef struct __attribute__((packed)) {
...
```

If this does not help, try explicit serializing of the structure into a string.

Beside this also read something about [C/C++ naming conventions](http://en.wikipedia.org/wiki/Naming_convention_%28programming%29#C_and_C.2B.2B)

NeoTokyoNori, 2015-01-03:
Hello @mz-fuzzy
Thanks for you input
 
I will give a shot at what you have suggested, and let you know how it goes. 

I am glad that you were able to grasp the issue so astutely from the information provided. 
This message board does not have any formatting/editing tags, so posting the entire code would get messy, so I thought it would be better to start off with just the essential parts. 

I had tried everything that I could think of, and your suggestion would have never occurred to me.  

Thanks again.

NeoTokyoNori, 2015-01-04:
Hello @mz-fuzzy 

You were absolutely right about the fix for the issue I had. 
I tried it and the data is now properly formatted ! 
Just to note, I only had to do it on the Due side. 

I have been trying to get this code running on Due for weeks.
It was so close, but this issue was like a "small stone in your shoe" that was ruining the whole trip,
so it was driving me mad, and I really have to thank you for solving this. 
Otherwise I would have had to listen to the "twelve commandments" and subject myself to self torture 
: P 

Anyway, so would you say this is a bug, and not a feature? 
It would certainly be wrathful to say everyone that encounters this issue is just an id_10T

Regards

mz-fuzzy, 2015-01-04:
Yes, this is a bug. In your sketch. Structure memory representation is platform specific and you have to care for proper serializing when transporting via network. This has nothing to do with RF24. 

Maybe RF24 is too permissive when allowing a void pointer as its payload and encouraging to pass any data type this way.

TMRh20, 2015-01-05:
@NeoTokyoNori 

My apologies if you took my comments personally, they were intended to simply state that most technical problems are user related, not system related. 

Having handled literally 10s of thousands of tech support issues over the past years, I have a very distinct hatred for tech. support.  
  
A bit of a story, one of my old high-school teachers related to me:
While she was working for a unionized grocery store in her youth, they went on strike, and a number of 'scab' workers were brought in to do their work. The existing employees were advised to answer any questions they are asked, and handle things for the 'scab' workers as part of a strategy.

The moral of the story: If you simply give people the answers etc., and they put zero effort into understanding what is going on and finding the answer for themselves, they may achiev their goal, but have learned nothing, and will struggle with similar issues in the future.

NeoTokyoNori, 2015-01-05:
@TMRh20 
Message received. no worries. 
Thanks for working on the library.
Regards

closed 2015-01-23 by TMRh20