nRF24/RF24 #56: Fixes to DYNPD/FEATURE on some NRF24L01+ Modules

dsbaha, 2015-01-05:
Hi, I have no clue how to submit a patch, but I wanted to submit the following for inclusion.  I have two different types of NRF14L01+ modules that behave slightly different (this is what the patch addresses).

In RF24.cpp 
```
/****************************************************************************/

void RF24::enableDynamicPayloads(void)
{
  // Enable dynamic payload throughout the system

    //toggle_features();
    write_register(FEATURE,read_register(FEATURE) | _BV(EN_DPL) );

	if (!read_register(FEATURE)) {
		toggle_features();
		write_register(FEATURE, read_register(FEATURE) | _BV(EN_DPL));
	}

  IF_SERIAL_DEBUG(printf("FEATURE=%i\r\n",read_register(FEATURE)));

  // Enable dynamic payload on all pipes
  //
  // Not sure the use case of only having dynamic payload on certain
  // pipes, so the library does not support it.
  write_register(DYNPD,read_register(DYNPD) | _BV(DPL_P5) | _BV(DPL_P4) | _BV(DPL_P3) | _BV(DPL_P2) | _BV(DPL_P1) | _BV(DPL_P0));

  dynamic_payloads_enabled = true;
}

/****************************************************************************/

void RF24::enableAckPayload(void)
{
  //
  // enable ack payload and dynamic payload features
  //

    //toggle_features();
    write_register(FEATURE,read_register(FEATURE) | _BV(EN_ACK_PAY) | _BV(EN_DPL) );

	if (!read_register(FEATURE)) {
		toggle_features();
		write_register(FEATURE, read_register(FEATURE) | _BV(EN_ACK_PAY) | _BV(EN_DPL));
	}

  IF_SERIAL_DEBUG(printf("FEATURE=%i\r\n",read_register(FEATURE)));

  //
  // Enable dynamic payload on pipes 0 & 1
  //

  write_register(DYNPD,read_register(DYNPD) | _BV(DPL_P1) | _BV(DPL_P0));
  dynamic_payloads_enabled = true;
}

/****************************************************************************/

void RF24::enableDynamicAck(void){
  //
  // enable dynamic ack features
  //
    //toggle_features();
    write_register(FEATURE,read_register(FEATURE) | _BV(EN_DYN_ACK) );

	if (!read_register(FEATURE)) {
		toggle_features();
		write_register(FEATURE, read_register(FEATURE) | _BV(EN_DYN_ACK));
	}

  IF_SERIAL_DEBUG(printf("FEATURE=%i\r\n",read_register(FEATURE)));

}

/****************************************************************************/
```

dsbaha, 2015-01-05:
Hmmm, only works on Arduino and not RPi, needs further investigation.

TMRh20, 2015-01-05:
Hi there, I've looked at this one in the past, but haven't noticed problems with the modules I've gotten so far.  
  
You need to specifically look at the [revision 2.0 of the nrf24l01 datasheet](http://www.nordicsemi.com/jpn/content/download/2730/34105/file/nRF24L01_Product_Specification_v2_0.pdf) since the ACTIVATE command was previously undocumented.  
  
I have a feeling that, it might be best to just put the toggle_features(); function into radio.begin, and that should leave the features activated and ready to use, although disabled. I haven't tested this though, because as I said, I haven't noticed any problems with my hardware, but if you could try that, I would be grateful.

TMRh20, 2015-01-05:
[Here](https://github.com/TMRh20/RF24/archive/Updates.zip) is a link to the RF24 updates branch. I committed a change that doesn't appear to cause any other immediate problems, and works the same on RPi and Arduino.

Idea:
```
In radio.begin():

1. Attempt writing to the FEATURE register
2. Read the FEATURE register
3. If not 0, set it to 0, leave it enabled
4. If 0, just ACTIVATE it
5. Leave it activated, do not toggle it any further.
```
All I can test is that it doesn't cause any other problems, but I am fairly confident this should address all scenarios.  
  
Please let me know if this works, or if you have found another or better solution.

dsbaha, 2015-01-05:
Thanks for the submission!  It doesn't seem to work though.  I have two sets of NRF24L01+ modules.  They both look the same (board and layout) but the "non-working" one has less resistors than the "working" one.  Its weird, utilizing the gettingstarted app with RF24 & RF24Network-DEV, they both work.  When trying to use the example RF24Mesh app, that is when things break.  I thought I tracked it down to the DYNPD/FEATURES setting, but it appears to be additional problems.  For instance, I can see multicast/fragmented packets going out, but the "non-working" modules never receive them.  I'll provide a link to the NRF24L01+ modules I bought later today.

dsbaha, 2015-01-08:
[These are the NRF24L01+ modules I got which do not behave like the working modules.]
(http://www.ebay.com/itm/10pcs-New-NRF24L01-2-4GHz-Antenna-RF-Wireless-Transceiver-Module-US-FLY-/400768784248?pt=LH_DefaultDomain_0&hash=item5d4fae5778)

TMRh20, 2015-01-08:
https://github.com/TMRh20/RF24/wiki/GettingStarted-sketch-failing-to-send:-how-to-diagnose%3F

http://en.wikipedia.org/wiki/Wikipedia:Reference_desk/How_to_ask_a_software_question

closed 2015-01-08 by TMRh20