nRF24/RF24 #57: ImportError: No module named RF24

WERAQS, 2015-01-11:
did you changed something on the code, I've tried to import RF24 to new RasPi but when run it says

"Traceback (most recent call last):
   File "pingpair_dyn.py", line 10, in <module>
          from RF24 import *
ImportError: No module named RF24"

*make'd and installed all

weird?

mz-fuzzy, 2015-01-11:
I went through recent commits related to pyRF24 and nothing suspicious.
what can be a problem: in readme there is instructed:

```
./setup.py install
```

instead of

```
sudo ./setup.py install
```

Check this.

WERAQS, 2015-01-11:
in fact, "sudo ./gettingstarted" works, rest not, tried installing RF24 for 5 times now, reinstalling raspbian all over again.

spaniakos, 2015-01-11:
What cide is not running ? I have tested just my custom code with it . I
can do a check for you .

WERAQS, 2015-01-11:
I'm restarting all coding process, thanks for help

closed 2015-01-11 by WERAQS