nRF24/RF24 #59: Will the IRQ pin fire if a radio is powerdown() or stoplistening() ?

lakidd, 2015-01-12:
I'm trying to understand how to make a battery powered system. I'm assuming that using the IRQ pin tied to the arduino is the best way to do this ?
If the radio is set to powerdown I'm guessing the IRQ pin wont change state if traffic is rx'd ?
If it's stoplistening() ?
From the doco
After calling startListening(), a basic radio will consume about 13.5mA at max PA level. During active transmission, the radio will consume about 11.5mA, but this will be reduced to 26uA (.026mA) between sending
Is it possible to have a radio consume 26uA and stil respond to incoming data ?

TMRh20, 2015-01-12:
```If the radio is set to powerdown I'm guessing the IRQ pin wont change state if traffic is rx'd ?
If it's stoplistening() ?```
Nope

```Is it possible to have a radio consume 26uA and stil respond to incoming data ?```
No, except for ACK payloads.

http://tmrh20.github.io/RF24/GettingStarted_CallResponse_8ino-example.html
http://tmrh20.github.io/RF24/classRF24.html#abf8efced2ee9edbcc6510878b20edc1b

lakidd, 2015-01-13:
Cool, got it, thanks

closed 2015-01-13 by lakidd