nRF24/RF24 #64: Error compiling Python Wrapper

sebpwhite, 2015-01-20:
When trying to setup pyRF24 as per the readme I get the following errors. Not sure if I'm doing something silly?

running build
running build_ext
building 'RF24' extension
gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-protpes -fPIC -I/usr/include/python2.7 -c pyRF24.cpp -o build/temp.linux-armv6l-2pyRF24.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for Ada/bjC but not for C++ [enabled by default]
pyRF24.cpp: In function ‘void init_module_RF24()’:
pyRF24.cpp:276:81: error: no matches converting function ‘txStandBy’ to type ol (class RF24::*)(uint32_t) {aka bool (class RF24::*)(unsigned int)}’
/usr/local/include/RF24/RF24.h:450:9: error: candidates are: bool RF24::txStay()
/usr/local/include/RF24/RF24.h:468:9: error:                 bool RF24::txStay(uint32_t, bool)
error: command 'gcc' failed with exit status 1

mz-fuzzy, 2015-01-20:
It seems that you have a strange RF24.h installed on the system?
Make and install RPi's RF24 library again.
https://github.com/TMRh20/RF24/blob/master/README.md#rpi---rf24-quick-start

TMRh20, 2015-01-20:
This is my bad. I slipped in a small change that I **assumed** wouldnt change the API for anybody. I wanted to get it in sooner rather than later since I'm hoping to use it in RF24Network_DEV. Too bad I didn't think of the python wrapper...

The new definitions are as follows, but I have no idea how to declare an optional parameter in python.
```bool txStandBy(uint32_t timeout, bool startTx = 0);```
```void startFastWrite( const void* buf, uint8_t len, const bool multicast, bool startTx = 1 );```

If you could clue me in @mz-fuzzy , I could make the needed change right away, or if you would be willing to to, that would be excellent also. Otherwise it might take me a bit to figure out hehe.

sebpwhite, 2015-01-20:
Thanks very much for the very fast response (and also thanks to both of you for all your work on this fantastic library!)
Glad to hear I wasn't doing anything silly!

mz-fuzzy, 2015-01-20:
@TMRh20 I'll check that.
anyway symptoms this seems suspicious, I've never seen such RF24 method there:
/usr/local/include/RF24/RF24.h:450:9: error: candidates are: bool RF24::**txStay()**
/usr/local/include/RF24/RF24.h:468:9: error: bool RF24::**txStay(uint32_t, bool)**

TMRh20, 2015-01-20:
Thanks @mz-fuzzy , but I assumed that was just a copy/paste thing, because the line numbers matched.
I did manage to figure it out, and it ended up being much easier than it seemed initially, just by specifying it similar to in c++ in the wrapper. ie: ```bp::arg("startTx")=0``` 

@sebpwhite It should compile nicely now.

mz-fuzzy, 2015-01-20:
@TMRh20 thanks; however there is one detail in your fix which I don't like: you have defined default values once again there. They are now on 2 places, in .h and in the wrapper. If it's fine with you, I'll pull request for a bit different fix.

TMRh20, 2015-01-20:
@mz-fuzzy Bad habits die hard, that would be great.

mz-fuzzy, 2015-01-20:
Pull request #65.

By the way - nice to see that people use the wrapper and detect errors.

TMRh20, 2015-01-20:
I'm not too surprised, my favorite part is not having to recompile things when testing small changes. Python seems to have its advantages. Thanks for the wrapper and support.

sebpwhite, 2015-01-21:
All working now. Thanks again to both of you!

closed 2015-01-21 by sebpwhite