nRF24/RF24 #68: **Change freeze/hold on master branch**

TMRh20, 2015-02-01:
Due to the number of changes lately, and expanding hardware support, please create all new pull requests against the [updates](https://github.com/TMRh20/RF24/tree/Updates) branch, except for 'show-stopping' bug fixes etc. for the time being.

[Updates](https://github.com/TMRh20/RF24/tree/Updates) will be synced with the master after testing etc is completed, with timeframes etc. TBD. The intent is to allow more consistency in the master branch without stopping development.

Downloads:
[Master](https://github.com/TMRh20/RF24/archive/master.zip) (stable)
[Updates](https://github.com/TMRh20/RF24/archive/Updates.zip) (dev/testing)

Materdaddy, 2015-02-19:
Do you have any updates on timeframe for a merge to master?  I cherry-picked 25d379c from Updates to master for a project on a Pi/Pi2 and it seems to work with a quick smoke-test on a Pi2.  (I didn't re-test on a Pi yet).

If the timeline stretches out, would you be open to strange work-flows like individual confirmation of a patch and cherry-picks to master or branches per platform where it's master+platform patch?  If not, no big deal, I'll probably fork your repo until such time you close this ticket with a merge to master.  (The more I think about this one, the more I think it makes sense for me to temporarily fork so you can keep making progress without CM work).

Thank you for your work on this project!

TMRh20, 2015-02-19:
Well, I don't have a specific timeline in mind, but I'm really enjoying the freedom of having a dev branch to work out of, and a master branch for finished, stable releases only.

I assume the number of updates will enventually peter out to the point where this is not necessary, but with all of the hype and interest surrounding IoT systems and devices lately, there seems to have been an increased demand for wireless devices like nrf24l01, and extended hardware support.
  
I also don't really expect any functional or API changes, since the lib has been around for such a long time, we are really just keeping drivers updated and adding additional hardware support. Once the details of this new hardware support template are all in place, and docs updated, I would like to push to master ASAP.  
  
In the end, it really depends on the number of changes coming in, and my ability to keep up with testing and change/configuration management. I'm targetting monthly or bi-monthly releases, if/when there are enough changes, but that is a loose target. I would prefer that all changes be made against the 'Updates' branch, and would allow minimal changes to master, but generally only in the case of fixes or emergency patches.  In the case of driver updates like https://github.com/TMRh20/RF24/commit/25d379c187d673c775a7d68702f6f811405f1bdc I would probably be ok with some cherry picked pushes to master after testing, but again, the RPi should remain pretty stable in both branches, so the need is questionable unless the merge is delayed.

With that in mind, the dev/updates branch should remain fairly stable, except for maybe the newly supported platforms.

closed 2015-09-24 by TMRh20