nRF24/RF24 #69: Porting TMRh20 RF24 Libraries to the BeagleBone Black

desertgourd, 2015-02-02:
Would like to help port your libraries over to the BeaglBone Black (BBB). This issue started with a request comment on the TMRh20's Project Blog for March 2014 and a reply from TMRh20 with some initial suggestions.  

I am going over the reply from TMRh20 and looking at the original port to the BBB of the maniacbug/RF24 library done by Purinda Gunasekara  https://bitbucket.org/purinda/rf24bb/wiki/Home, and a more recent fork  https://github.com/mlsorensen/RF24/tree/master/librf24-bbb/librf24 of the stanleyseow/RF24 for the BBB.

It is going to take a few days to digest all this,  I am willing to test any updates on my system,  I have the BBB, several Arduino Nano's and an Arduino Due with several nRF24L01 boards.  I hope to help with the coding as I come up to speed.

TMRh20, 2015-02-02:
Hi, since I'm kind of asking you to follow a template that hasn't been created yet, I'll try to put something together for this in the [updates](https://github.com/TMRh20/RF24/tree/Updates) branch, starting with the BBB, and do a rough port of the BBB stuff. It most likely won't work, but should hopefully give you a better idea of what I'm talking about, and from there we can figure out the details. I'll post back here with more info once I have something in place.

desertgourd, 2015-02-02:
That sounds good.  If you setup the template I think I can handle modifying the configuration changes necessary for the BBB with some help.  One question where did you get the "#if defined _ARM_ARCH_7A"?  The BBB uses the ARM_CORTEX_A8.

TMRh20, 2015-02-02:
From [google](http://electron14.com/?p=463) of course.

I created a new folder in the RF24 directory:  ```arch/BBB```
Within that directory are all the custom files specific to the BBB, including the Makefile.
If you take a look at the commit referenced above, it will show the specific changes.

A: The RF24_config.h file has been modified to simply detect the BBB, and include the config file from the arch/BBB directory (RF24_arch_config.h)
B: As per your comment, I'm not sure of the best method for detecting the BBB, but I will leave that up to you to determine.
C: The RF24.cpp and RF24.h files should contain all the necessary changes, it should just be a matter of getting the Makefiles, examples, and pin configuration figured out.

Moving forward:
1. We will probably want to merge the BBB Makefile with the file in the RF24 directory, instead of having separate files.
2. I included a gettingstarted sketch, under the BBB directory, which I was able to get working with the RPi. The pin selection would need to be modified for the BBB.
3. You may need to modify the chosen spidev in spi.cpp ```this->device = "/dev/spidev1.0";;```

I was able to get the BBB code running on the RPi, by copying the BBB Makefile into the main RF24 directory, renaming the original, and modifying the RF24_config.h file to detect RPi as BBB. The code seems to work, but the response times range from 15-20ms, so either the GPIO is really slow, or there is some other issue.

It may be worthwhile to see if there is a pure hardware SPI driver for the BBB, or faster GPIO options, to pick the performance up.  

Hopefully this all makes sense, but if you have any questions or problems, feel free to post here or drop me an email.

dromer, 2015-02-03:
Any chance this lays some groundwork for porting to, for instance, Allwinner? (I'm looking for a port to A10/A20)

TMRh20, 2015-02-03:
I believe porting to A10/A20 would be very similar to BBB. We could probably configure this as general Linux/ARM template, but I'm not familiar enough with the different platforms etc. to know if this would be a good idea or not.

Main Components/Needed Options:
1. A method to detect the specific hardware/platform
2. Functions to replicate basic Arduino GPIO functions ( pinMode(), digitalWrite() ) , see the RF24_arch_config.h file under the BBB directory along with gpio.h and gpio.cpp
3. Functions to replicate basic Arduino timing functions ( millis() , delay() ) See compatibility.h and compatiility.c
4. Functions to replicate standard Arduino SPI functions ( SPI init, SPI transfer )

From there it is mainly device specific details, defining the correct pins, choosing the correct spidev in spi.cpp, etc. Technically, the current BBB code could be modified to work on any linux system using spidev, as I mentioned above, it was fairly straightforward to modify for Raspberry Pi. You could probably just create a copy of the BBB files/folder, edit the files as required, and update the RF24_config.h file to detect the platform correctly.  The Makefile would need to be merged into the main one as well.

See [here](https://github.com/TMRh20/RF24/tree/Updates/arch/BBB) for the BBB/Linux specific files.

desertgourd, 2015-02-04:
@ TRMh20
Just a note to let you know I am still here.  I have been reviewing the work of all the previous contributors and getting a better understanding of how your libraries are put together.  I am a newbie at this,  I have been mainly involved in instrumentation design dealing with the hardware end and leaving the software interface to others.  This is giving me a chance to develop a whole new set of skills.  At this point the post you made on creating the new arch/BBB directory and moving forward makes a lot of sense.  I had questions but have been finding the answers while digging deeper into the research. If I hit any road blocks I will let you know.  I am going to start to put together a Nano and BBB with nRF24 transceivers for testing.  I am going to install the mlsorensen RF24 library for the BBB to test the hardware and have a base to work from where the SPI and GPIO are working. Time to go do the farm chores now.

TMRh20, 2015-02-04:
Sounds like you are on the right track, glad I could help!

desertgourd, 2015-02-13:
I copied the contents of the RF24-Updates directory to /root/rf24libs/ on the BBB.
I renamed the /rf24libs/RF24/makefile and copied the /rf24libs/RF24/arch/BBB/makefile over to /rf24libs/RF24/makefile.

When I compile I get a lot of warnings and errors such as:

root@beaglebone:~/rf24libs/RF24# make all
g++ -Wall -fPIC  -c RF24.cpp
In file included from RF24.h:20:0,
                 from RF24.cpp:11:
RPi/bcm2835.h:336:0: warning: "HIGH" redefined [enabled by default]
arch/BBB/RF24_arch_config.h:54:0: note: this is the location of the previous definition
RPi/bcm2835.h:338:0: warning: "LOW" redefined [enabled by default]
arch/BBB/RF24_arch_config.h:53:0: note: this is the location of the previous definition
In file included from RF24.h:20:0,
                 from RF24.cpp:11:

From the RF24.h file

18  #include "RF24_config.h"
19  //#if ( defined (__linux) || defined (LINUX) ) && defined( __arm__ ) && !defined (RF24_BBB)
20  #if ( defined (__linux) || defined (LINUX) ) && defined( __arm__ ) && !defined(RF24_BBB) // Mod by DG 2/13/15
21  #include "RPi/bcm2835.h"

line 19 was not detecting the defined RF24_BBB that had been set in the RF24_config.h file and was including the RPi/bcm2835.h file.

I commented out line 19 and changed it as in line 20 to eliminate the space between !defined and (RF24_BBB) and that fixed the problem. I was then getting the following on make all:

root@beaglebone:~/rf24libs/RF24# make all
g++ -Wall -fPIC  -c RF24.cpp
RF24.cpp: In member function 'uint8_t RF24::read_register(uint8_t, uint8_t*, uint8_t)':
RF24.cpp:93:73: error: 'SPI_CONTINUE' was not declared in this scope
RF24.cpp:97:38: error: no matching function for call to 'SPI::transfer(uint8_t&, int)'
RF24.cpp:97:38: note: candidate is:
arch/BBB/spi.h:31:10: note: uint8_t SPI::transfer(uint8_t)
arch/BBB/spi.h:31:10: note:   candidate expects 1 argument, 2 provided
RF24.cpp: In member function 'uint8_t RF24::read_register(uint8_t)'

The new errors are again associated with the "!defined (RF24_BBB)".  If this is indeed an error then it is extensive through out the source files, for example from above some times you include the space "defined (__linux) and  
defined (LINUX)" and some times you do not have the space as in above "defined( __arm__ )".

I am just learning C++ and before I go changing everything I wanted your input.

TMRh20, 2015-02-13:
I believe the main issue with the BBB code right now is the detection. I've never known spaces to make a difference, so not sure on that. I've updated the lib to include Due and ATTiny in the arch/ structure, so a lot of the ```!defined (RF24_BBB)``` have been removed from the main source. You may want to update to the latest code in the updates branch.

Steps:

1. Copy the makefile from arch/BBB to RF24 root
2. Edit RF24_config.h 
   a: Change the top line from ```#if defined __ARM_ARCH_7A__ //|| defined __linux || defined (linux)// BeagleBone Black running GNU/Linux``` to ``` #if defined __ARM_ARCH_7A__ || defined __linux || defined (linux)// BeagleBone Black running GNU/Linux``` to ensure BBB is detected 
3. On RPi, I need to edit ```arch/BBB/spi.cpp``` and set the correct spidev : ```this->device = "/dev/spidev0.0";;``` - on BBB you can probably leave it as is
4. Run ```make``` then  ```sudo make install```

Compiling an Example:
1. Edit the makefile in the example directory
   a: Find the line starting with g++, change lrf24-bcm to just lrf24
2. Edit the gettingstarted.cpp file
   a: Change the line ```RF24 radio(RPI_V2_GPIO...```  to something like ```RF24 radio(115, 117);``` to select your CS and CE pins respectively on BBB

Notes:
1. The only lines in RF24_config.h that should matter for BBB are lines 0 to 35
2. The BBB detection needs to be cleaned up/fixed. This could either be done by specifically identifying the RPI, and using the spidev code for every other linux system, or by identifying the BBB via #if defined statements

Since this driver works with spidev, I should be able to test most things on my RPi. If you are ok with it, I could clean this up/integrate it a bit more for BBB/Linux. If you want to continue working on it, that is ok by me as well.

desertgourd, 2015-02-13:
Thanks for the feed back.  I will wait for what changes you make and test them out on the BBB.  I want to learn but I also want to get this up and running as soon as possible so any input from you is fine.  There is enough to learn without being over overwhelmed.

TMRh20, 2015-02-14:
OK, I've modified the code enough that it **should** work directly on the BBB.

1. Update your code from the 'Updates' branch
2. Run ```sudo make install -B``` as usual
3. You will need to modify examples to declare the correct pins like ```RF24 radio(115, 117); ```
4. You may need to modify the arch/BBB/spi.cpp file to select the correct spi_dev.  ```this->device = "/dev/spidev1.0";;```

The pins and spi_dev defaults are the same in the mlsorensen fork you linked above

Please let me know if/how this works for you. The latency on RPi seems pretty slow (10ms using gettingstarted example, compared to 1-2ms using BCM2835 driver), but I'm hoping it works better on BBB. From what I can tell so far, the GPIO seems to be what is slowing things down on the RPi when using spi_dev.

Side note: A RPi can also be forced to use SPI_DEV by calling ```sudo make install -B RF24_SPIDEV=1```. The arch/BBB/spi.cpp file would need to be modified to use the correct spi_dev as above.

TMRh20, 2015-02-14:
@dromer This update should also work with any Linux/ARM device that supports spi_dev. As per my last post, the specific spidev needs to be selected in spi.cpp, and the correct CE & CS pins chosen in the examples via ```RF24 radio(CE_PIN,CS_PIN)```

dromer, 2015-02-14:
Nice, I will see if I can try something out on a Cubieboard/A10 tonight :)

desertgourd, 2015-02-14:
This is the output from make install -B
root@beaglebone:~/rf24libs/RF24# make install -B
cp arch/BBB/includes.h arch/includes.h
g++ -Wall -fPIC -Ofast -mfpu=vfp -mfloat-abi=hard -march=armv6zk -mtune=arm1176jzf-s -c RF24.cpp
In file included from /usr/include/c++/4.6/bits/char_traits.h:41:0,
                 from /usr/include/c++/4.6/string:42,
                 from arch/BBB/spi.h:11,
                 from arch/BBB/RF24_arch_config.h:16,
                 from arch/includes.h:6,
                 from RF24_config.h:35,
                 from RF24.cpp:10:
/usr/include/c++/4.6/bits/stl_algobase.h: In function 'int std::__lg(int)':
/usr/include/c++/4.6/bits/stl_algobase.h:981:15: sorry, unimplemented: Thumb-1 hard-float VFP ABI
arch/BBB/compatibility.h: At global scope:
arch/BBB/compatibility.h:18:23: warning: 'start' defined but not used [-Wunused-variable]
arch/BBB/compatibility.h:18:30: warning: 'end' defined but not used [-Wunused-variable]
arch/BBB/compatibility.h:19:13: warning: 'mtime' defined but not used [-Wunused-variable]
arch/BBB/compatibility.h:19:20: warning: 'seconds' defined but not used [-Wunused-variable]
arch/BBB/compatibility.h:19:29: warning: 'useconds' defined but not used [-Wunused-variable]
make: *** [RF24.o] Error 1
root@beaglebone:~/rf24libs/RF24#

TMRh20, 2015-02-14:
Looks like i overlooked the CCFLAGS defined for RPi.
The Makefile here should fix this issue:
https://gist.github.com/TMRh20/2bee6acf13c86cb5e70f#file-bbb_rpi-makefile

Please let me know if this works, and I can push the change. Hopefully there aren't too many more compatibility issues.

desertgourd, 2015-02-14:
Copied over "bbb_rpi-makefile" to replace the .../rf24libs/RF24/Makefile.
Ran "make install -B" and the output looks OK.  Got warnings but no errors.

root@beaglebone:~/rf24libs/RF24# make install -B
cp arch/BBB/includes.h arch/includes.h
g++ -Wall -fPIC  -c RF24.cpp
arch/BBB/compatibility.h:18:23: warning: 'start' defined but not used [-Wunused-variable]
arch/BBB/compatibility.h:18:30: warning: 'end' defined but not used [-Wunused-variable]
arch/BBB/compatibility.h:19:13: warning: 'mtime' defined but not used [-Wunused-variable]
arch/BBB/compatibility.h:19:20: warning: 'seconds' defined but not used [-Wunused-variable]
arch/BBB/compatibility.h:19:29: warning: 'useconds' defined but not used [-Wunused-variable]
gcc -Wall -fPIC  -c arch/BBB/spi.cpp
g++ -Wall -fPIC  -c arch/BBB/gpio.cpp
gcc -Wall -fPIC   -c arch/BBB/compatibility.c
arch/BBB/compatibility.c: In function '__msleep':
arch/BBB/compatibility.c:13:2: warning: implicit declaration of function 'nanosleep' [-Wimplicit-function-declaration]
g++ -shared -Wl,-soname,librf24-bcm.so.1  -o librf24-bcm.so.1.0 RF24.o spi.o gpio.o compatibility.o
[Installing Libs]
[Installing Headers]
root@beaglebone:~/rf24libs/RF24#

xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

Tried to make the gettingstarted.cpp and the output had an error.  

root@beaglebone:~/rf24libs/RF24/arch/BBB/examples# make install -B
g++ -Ofast -mfpu=vfp -mfloat-abi=hard -march=armv6zk -mtune=arm1176jzf-s -Wall -I../ -lrf24 gettingstarted.cpp -o gettingstarted
In file included from /usr/include/c++/4.6/cstdlib:66:0,
                 from gettingstarted.cpp:22:
/usr/include/stdlib.h: In function 'double atof(const char*)':
/usr/include/stdlib.h:279:1: sorry, unimplemented: Thumb-1 hard-float VFP ABI
make: *** [gettingstarted] Error 1
root@beaglebone:~/rf24libs/RF24/arch/BBB/examples#

TMRh20, 2015-02-14:
*facepalm* This can be a frustrating process without access to the hardware.

Its the same problem with the example makefile, you would need to add a # in front of CCFLAGS to comment it out in the Makefile. I guess this means the examples either need an updated makefile, or we need separate examples for Linux w/spidev

desertgourd, 2015-02-15:
I put a # in front of CCFLAFS to comment out the line and got the following on make 

root@beaglebone:~/rf24libs/RF24/arch/BBB/examples# make install -B
g++  -Wall -I../ -lrf24 gettingstarted.cpp -o gettingstarted
/usr/bin/ld: cannot find -lrf24
collect2: ld returned 1 exit status
make: *** [gettingstarted] Error 1
root@beaglebone:~/rf24libs/RF24/arch/BBB/examples# 

Do not get discouraged.  I think you have made quite a bit of progress today.  Maybe it would be easier if the process was broken down into the different hardware sets instead of one grand file set that compiled for all the different hardware boards. Not as eloquent but may be easier to maintain.  On the other hand I don't think we are to far away from having the gettingstarted example working and then the hardware can be tested.

TMRh20, 2015-02-15:
It might seem like it, but the RPi and Arduino code was originally separate, and it was a nightmare to maintain separate source files. This way, the install process should end up exactly the same on RPi and BBB from a user perspective, except for device specific pin configuration in examples.  
  
I should have noticed, and mentioned it, but you will need to run the examples from the main examples_RPi folder. The gettingstarted.cpp file has been updated with an example pin config for BBB and the Makefiles are updated with the correct CCFLAGS.

1. To keep it simple, just update the whole library
2. ```sudo make install```
3. Edit gettingstarted.cpp, uncomment example pin config for BBB, comment out RPI pin config. Do the same for any other examples you wish to compile.
4. ```Make``` the example(s)
5. ```sudo ./gettingstarted```

desertgourd, 2015-02-15:
Great success!!
I made the changes to the example files in the /RF24/examples_RPi directory.

I am using the following Pins for the BBB 

// Setup for ARM(Linux) devices like BBB using spidev
RF24 radio(125,117);

from the following blog
http://electron14.com/wp-content/uploads/2014/06/Screen-Shot-2014-06-01-at-3.50.18-PM.png
http://electron14.com/?cat=28

Make install output:
root@beaglebone:~/rf24libs/RF24/examples_RPi# make install
g++  -Wall -I../ -lrf24-bcm gettingstarted.cpp -o gettingstarted
g++  -Wall -I../ -lrf24-bcm gettingstarted_call_response.cpp -o gettingstarted_call_response
g++  -Wall -I../ -lrf24-bcm transfer.cpp -o transfer
g++  -Wall -I../ -lrf24-bcm pingpair_dyn.cpp -o pingpair_dyn
.././arch/BBB/compatibility.h:18:23: warning: 'start' defined but not used [-Wunused-variable]
.././arch/BBB/compatibility.h:18:30: warning: 'end' defined but not used [-Wunused-variable]
.././arch/BBB/compatibility.h:19:13: warning: 'mtime' defined but not used [-Wunused-variable]
.././arch/BBB/compatibility.h:19:20: warning: 'seconds' defined but not used [-Wunused-variable]
.././arch/BBB/compatibility.h:19:29: warning: 'useconds' defined but not used [-Wunused-variable]
test -d /usr/local || mkdir /usr/local
test -d /usr/local/bin || mkdir /usr/local/bin
for prog in gettingstarted gettingstarted_call_response transfer pingpair_dyn; do \
          install -m 0755 $prog /usr/local/bin; \
        done
root@beaglebone:~/rf24libs/RF24/examples_RPi#

Running the example:
root@beaglebone:~/rf24libs/RF24/examples_RPi# ./gettingstarted
RF24/examples/GettingStarted/
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xe7e7e7e7e7 0xc2c2c2c2c2
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0xe7e7e7e7e7
RX_PW_P0-6       = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x03
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX

 ************ Role Setup ***********
Choose a role: Enter 0 for pong_back, 1 for ping_out (CTRL+C to exit)

As you can see I am now communicating with the nRF24L01+ chip and can read/write the registers. Is there one place where the Radio pipe addresses or node addresses are set.

I next tried to compile the /RF24/examples/GettingStarted/GettingStarted.ino and got the following errors that appear to be associated with the Radio pipe addresses.  I am using Sublime Text with the Arduino plugin to upload examples to the Arduino Nano's

Upload Output:
Compiling GettingStarted...
Creating /home/robert/Arduino_Build/GettingStarted/GettingStarted.ino.cpp.o...
/home/robert/Arduino_Build/GettingStarted/GettingStarted.ino.cpp: In function 'void setup()':
/home/robert/Arduino_Build/GettingStarted/GettingStarted.ino.cpp:40:38: error: invalid conversion from 'byte* {aka unsigned char*}' to 'uint64_t {aka long long unsigned int}' [-fpermissive]
     radio.openWritingPipe(addresses[1]);
                                      ^
In file included from /home/robert/Arduino_Build/GettingStarted/GettingStarted.ino.cpp:10:0:
/home/robert/Arduino/libraries/RF24/RF24.h:326:8: error:   initializing argument 1 of 'void RF24::openWritingPipe(uint64_t)' [-fpermissive]
   void openWritingPipe(uint64_t address);
        ^
/home/robert/Arduino_Build/GettingStarted/GettingStarted.ino.cpp:41:40: error: invalid conversion from 'byte* {aka unsigned char*}' to 'uint64_t {aka long long unsigned int}' [-fpermissive]
     radio.openReadingPipe(1,addresses[0]);
                                        ^
In file included from /home/robert/Arduino_Build/GettingStarted/GettingStarted.ino.cpp:10:0:
/home/robert/Arduino/libraries/RF24/RF24.h:355:8: error:   initializing argument 2 of 'void RF24::openReadingPipe(uint8_t, uint64_t)' [-fpermissive]
   void openReadingPipe(uint8_t number, uint64_t address);
        ^
/home/robert/Arduino_Build/GettingStarted/GettingStarted.ino.cpp:43:38: error: invalid conversion from 'byte* {aka unsigned char*}' to 'uint64_t {aka long long unsigned int}' [-fpermissive]
     radio.openWritingPipe(addresses[0]);
                                      ^
In file included from /home/robert/Arduino_Build/GettingStarted/GettingStarted.ino.cpp:10:0:
/home/robert/Arduino/libraries/RF24/RF24.h:326:8: error:   initializing argument 1 of 'void RF24::openWritingPipe(uint64_t)' [-fpermissive]
   void openWritingPipe(uint64_t address);
        ^
/home/robert/Arduino_Build/GettingStarted/GettingStarted.ino.cpp:44:40: error: invalid conversion from 'byte* {aka unsigned char*}' to 'uint64_t {aka long long unsigned int}' [-fpermissive]
     radio.openReadingPipe(1,addresses[1]);
                                        ^
In file included from /home/robert/Arduino_Build/GettingStarted/GettingStarted.ino.cpp:10:0:
/home/robert/Arduino/libraries/RF24/RF24.h:355:8: error:   initializing argument 2 of 'void RF24::openReadingPipe(uint8_t, uint64_t)' [-fpermissive]
   void openReadingPipe(uint8_t number, uint64_t address);
        ^
[Stino - Error 1]
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
I have the BBB and 2 Arduino Nano's all with the nRF24L01+ units.  My plan is to load one Nano with the GettingStarted.ino and test it with the BBB.  Once That is working I would like to setup the BBB and 2 Nano's as a network for testing.

TMRh20, 2015-02-15:
Excellent. The addresses/pipes are configured according to the defined
radio number in the getting started sketch.

It looks like Sublime is picking up an old version of the rf24 library, so
you would need to update that for Arduino to work.

http://electron14.com/wp-content/uploads/2014/06/Screen-Shot-2014-06-01-at-3.50.18-PM.png
http://electron14.com/?cat=28

desertgourd, 2015-02-15:
Further testing of the GettingStarted example on the BBB

When I run gettingstarted and go into the ping mode I get the following errors.

root@beaglebone:~/rf24libs/RF24/examples_RPi# ./gettingstarted
RF24/examples/GettingStarted/
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0x65646f4e32
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x03
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX

 ************ Role Setup ***********
Choose a role: Enter 0 for pong_back, 1 for ping_out (CTRL+C to exit)
>1
Role: Ping Out, starting transmission

Now sending...
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.

I believe this is being outputted from the following function:
/******************************************************************/
#if defined (FAILURE_HANDLING) || defined (RF24_LINUX)
void RF24::errNotify(){
	#if defined (SERIAL_DEBUG) || defined (RF24_LINUX)
	  printf_P(PSTR("RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.\r\n"));
	#endif
	#if defined (FAILURE_HANDLING)
	failureDetected = 1;
	#else
	delay(5000);
	#endif
}
#endif
/******************************************************************/

Isn't the 'defined (RF24_LINUX)' true from being a BBB system and the RF24::errNotify() would run?  I have checked the pin connections and wiring and we are reading and writing the nRF24L01+ chip registers.

I have also copied the RF24 directory over to the Arduino folder and can compile and upload the GettingStarted.ino example from the Arduino IDE to 2 Arduino Nano units.  They run but I cannot get any ping/pong response between the Nano's or a Nano and the BBB.  Output from the Nano's is:

RF24/examples/GettingStarted
*** PRESS 'T' to begin transmitting to the other node
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
Now sending
failed
Failed, response timed out.
Now sending

The 2 Nano units are good and I have tested them with other RF24 libraries.
I found where you setChannel(76); in function void RF24::begin(void) but have not found where Pipes 0x65646f4e32 0x65646f4e31 are set.  Where can those be changed?

TMRh20, 2015-02-15:
The ```RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.``` message means the radio is not responding. This is usually due to incorrect pins/connections/wiring, etc..

The http://electron14.com/?cat=28 example uses  RF24 radio(115, 117);

desertgourd, 2015-02-16:
The  http://electron14.com/?cat=28  
"1. Wire up equipment" diagram shows GPIO_125 as CE and GPIO_117 as CSN.   
This would need  RF24 radio(125, 117); 
I changed to that pin configuration on the files I found that needed the pin #.  
If there was a wiring error would the radio.printDetails(); be giving the RF24 chip register output at the start of the program? 

root@beaglebone:~/rf24libs/RF24/examples_RPi# ./gettingstarted
RF24/examples/GettingStarted/
STATUS = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1 = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR = 0x65646f4e32
RX_PW_P0-6 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA = 0x3f
EN_RXADDR = 0x03
RF_CH = 0x4c
RF_SETUP = 0x07
CONFIG = 0x0e
DYNPD/FEATURE = 0x00 0x00
Data Rate = 1MBPS
Model = nRF24L01+
CRC Length = 16 bits
PA Power = PA_MAX

TMRh20, 2015-02-16:
```If there was a wiring error would the radio.printDetails(); be giving the RF24 chip register output at the start of the program? ```
Yes.
  
The CE pin is toggled when writing, where it hangs.

The GPIO code is copied exactly from the other fork, so if you had it working with that fork, the pin cofnig is the same. If you have a way to test the selected CE pin to see if it actually is toggled, that would give you more info also.

desertgourd, 2015-02-16:
I was not able to get the BBB to work with the mlsorensen fork.  He was testing between a BBB and a RPi.  I was testing between the BBB and a Nano and it was not working and so I thought I would give your fork a try.  You are right that the mlsorensen fork dose use the RF24 radio(115, 117);  On his blog he specifies pins RF24 radio(125, 117); from the wiring up diagrams and the picture at the top showing the BBB header #9 connected by ribbon cable to the RF24 board.  In my search of the web I have not been able to find any information on GPIO_115 being available on the BBB headers.  I will take the issue as being a hardware problem and continue testing the BBB and cabling.  I have a scope but won't have access to it for a few days.  I have swaped the RF24 board with the other 2 boards on the Nano's and they all work with the Maniacbug RF24 master.

TMRh20, 2015-02-16:
I didn't notice your comments about the Nano. Have you modified the pins in the User Configuration to match your hardware? ```RF24 radio(9,10);``` Since it works with the maniacbug fork this is pretty much a user/hardware configuration issue, which I won't help to fix.  
  
The GPIO and BBB code is tested and works on RPi, and does everything in a very generic, ARM/Linux friendly way, so the remaining problems are *likely* user/hardware configuration issues as well.  
  
You could connect an LED+resistor to your CE pin and use Arduino type delay(), pinMode() and digitalWrite(); commands in gettingstarted to test the GPIO and pin selection.  

At this point, you basically need to verify that the basic functionality (GPIO, etc) is working on your end with the correct pins etc. which again, I won't really help with.

desertgourd, 2015-02-16:
I checked in with Mlsorensen and he said that their is great confusion out their that CE pin is given as both GPIO 115 and 125.  The correct pin number is GPIO 115.  You were correct the RF24 radio(115,117) is the right version.  After correcting that I still had an intermittent on the cabling to RF24 board.  For 2 Nano boards to communicate with each other of I had to set one to 'radioNumber = 1'.  Below is the output for the gettingstarted ping/pong between the BBB and a Nano.  Is the delay comparable to what you are getting with the RPi?  Now that this is working I will try some of the other examples.  Is there anything you want tested on the BBB in particular?  Thanks for your help, I am sure as I go along there will be more questions. One application I am locking at is sending jpg files back from a remote camera on an Arduino Due board.  Do you have a file transfer example yet?

BBB Ping Role
 ************ Role Setup ***********
Choose a role: Enter 0 for pong_back, 1 for ping_out (CTRL+C to exit)
>1
Role: Ping Out, starting transmission
Now sending...
Got response 17026, round-trip delay: 16
Now sending...
Got response 18051, round-trip delay: 14
Now sending...
Got response 19074, round-trip delay: 15
Now sending...
Got response 20102, round-trip delay: 32
Now sending...
Got response 21140, round-trip delay: 9
Now sending...
Got response 22155, round-trip delay: 10

BBB Pong Role
 ************ Role Setup ***********
Choose a role: Enter 0 for pong_back, 1 for ping_out (CTRL+C to exit)
>0
Role: Pong Back, awaiting transmission
Got payload(4) 350873064...
Got payload(4) 351886420...
Got payload(4) 352901732...
Got payload(4) 353934912...
Got payload(4) 354949776...
Got payload(4) 355964776...
Got payload(4) 356979920...

Arduino Nano output for Pong and Ping
RF24/examples/GettingStarted
*** PRESS 'T' to begin transmitting to the other node
Sent response 17026
Sent response 18051
Sent response 19074

*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
Now sending
Sent 350878136, Got response 350873064, Round-trip delay 5072 microseconds
Now sending
Sent 351899236, Got response 351886420, Round-trip delay 12816 microseconds
Now sending
Sent 352932416, Got response 352901732, Round-trip delay 30684 microseconds
Now sending
Sent 353947280, Got response 353934912, Round-trip delay 12368 microseconds

TMRh20, 2015-02-17:
Good to hear its finally working. Thanks for putting in all the time and effort to make this work

The BBB pin documentation is sure odd. The only reference I could find to pin 115 was in the mlsorenson fork itself, but at least it works.

To ansewr your questions, the response times are similar to RPi, but they seem a bit inconsistent. 

RPi:
```
Got response 346, round-trip delay: 12
Now sending...
Got response 1364, round-trip delay: 12
Now sending...
Got response 2382, round-trip delay: 11
Now sending...
Got response 3399, round-trip delay: 12
Now sending...
Got response 4417, round-trip delay: 11
```

Nano:
```
Sent 450120, Got response 440024, Round-trip delay 10096 microseconds
Now sending
Sent 1462376, Got response 1452452, Round-trip delay 9924 microseconds
Now sending
Sent 2474208, Got response 2464152, Round-trip delay 10056 microseconds
Now sending
Sent 3486348, Got response 3476028, Round-trip delay 10320 microseconds
```

If your devices are any distance away, it may help to comment out ```radio.setPALevel(RF24_PA_LOW);```

To give a comparison, using the bcm2835 library for RPi, response times are consistent between 1 to 2 ms, and same with Arduino, so using SPIDEV is not an ideal solution.

Some of your response times seem to be quicker, so I'm assuming that the SPI and GPIO is a bit faster on the BBB.

```Now that this is working I will try some of the other examples. Is there anything you want tested on the BBB in particular? ```

Since I can test most things on the RPi using spidev, I'm pretty confident the BBB will perform as expected. I usually run through most of the examples to test things out, since they cover most of the features and functionality. The main ones to test are gettingstarted, gettingstarted_call_response and the transfer example.

```One application I am locking at is sending jpg files back from a remote camera on an Arduino Due board. Do you have a file transfer example yet?```

Not specifically, but the transfer.ino example should give you a general idea. You would need to create a basic protocol, and usually want to include some sort of incremental value with your payloads, to prevent sequencing errors when the data (CRC) is exactly the same from payload to payload. You would also need a method for handling dropped/missed packets, retries, duplicates, etc.

Depending on the file size, you could also use RF24Network with a really big MAX_PAYLOAD size, and just use fragmentation to send the whole file at once. (This is limited by the size of the file and error rate during transmission)  Note: If you retry a failed large fragment, without creating a new RF24NetworkHeader, the re-assembly code should pick up where if failed last.

If it was me, I would probably look at using TCP (RF24Ethernet lib) with a large max payload size to make the end process simple and reliable, but it really depends on you and your preferences. This way may be the most reliable and simple method in the end, but is more difficult to configure, and definitely not as fast.

desertgourd, 2015-02-17:
I found Derek Molloy's blog and video @  http://derekmolloy.ie/gpios-on-the-beaglebone-black-using-device-tree-overlays/  to be a definitive source.  He explains the topic very well and in the the P9 header tables on his github site https://github.com/derekmolloy/boneDeviceTree he list P9 header pin # 27 as gpio 115.

desertgourd, 2015-02-17:
I am playing with the spi.cpp settings and the speed is set to this->speed = 8000000; // 8Mhz  which gave an average delay of 15ms with the min/mas = 5/34 ms.  Setting speed to this->speed = 2000000; // 2Mhz avg. 15ms min/max 8/19ms.  A speed of this->speed = 24000000; // 24Mhz gave an avg. of 15ms with about half of the responses failing. To me this indicates that the BBB hardware is capable of higher spi throughput but the spidev kernal has a lot of overhead and that there is some possibilities for improvement. I will look into it.

desertgourd, 2015-02-17:
The spidev kernel is designed to interface with linux so it has to have a lot of excess overhead to comply with all the different linux environments it could be in.  The Arduino spi is very simple and has a round trip delay of 2ms between two Nano's.  Using the Arduino spi libraries that interface with your RF24 library as a quide it might be possible to bit bang a spi driver for the BBB that would be compatible with your RF24 library.  I have bit banged spi drivers for Microchip cpu's in the past to interface with various sensors and think this may be possible.  I may be over my head on this but I will give it a try.

TMRh20, 2015-02-17:
Excellent information. The max SPI speed of the RF24 modules should be about 8mhz if I remember correctly, so errors will likely crop up with speeds over that.
  
BIt-banging might work, but the GPIO speed is part of the latency problem I believe. Take a look at [this](https://github.com/TMRh20/RF24/pull/74#issuecomment-74689288) thread. The mraa layer should work with a number of linux platforms including BBB, and judging from the comments, we might be able to bring the performance up a bit more.

desertgourd, 2015-02-17:
The MRAA layer looks very interesting.  I have added it to my list of things to research.  I have been going over the Arm Cortex 8 reference manual and it has hardware implemented SPI.  It is just a mater of setting it up and reading and writing registers. That may be over simplification but it is easier then writing everything in software.  At this stage I am researching all the possibilities before committing to one method.  I will be following what is happening on MRAA.

desertgourd, 2015-02-18:
I tested out the other 3 examples in the examples_RPi directory and got the following.

gettingstarted_call_response:
gettingstarted_call_response BBB-ping to Nano-pong round trip delay avg. 9ms  min/max 5/21 ms
gettingstarted_call_response Nano-ping to BBB-pong round trip delay avg. 788us  min/max 760/804us

Round trip delay with Nano as ping device was consistently less then 1 ms.  I think this is an important piece of information on where the delays are coming from.  Just have to figure out what it means.  

pingpair_dyn: works fine. Data scrolls up screen fast.

transfer: 
transfer BBB->Nano transfer rate avg. 2.5 KB/S
transfer Nano->BBB transfer rate avg. 0.03 KB/S 
Something wrong with the Nano->BBB transfer.  Will take a look the programs.  What transfer rates did you get with the RPi.

It looks like you are making progress on the MRAA layers with the RPi. Can that be applied to the BBB?

TMRh20, 2015-02-18:
I just committed an update that seems to address some timing issues with the transfer example.

getstarted = 10-11ms on both
getstarted_call_response: RPi 6-7ms Nano 6-700us

The reason for the gettingstarted_call_response differences is that the ACK_PAYLOAD is actually written to the radio long before it is requested, so the RPI/BBB does not have to do anything. Thats why the Uno/Nano will receive it in 700us.
In reverse, the payload is delivered just as quickly, but it takes a while for the RPi/BBB to read it in, so this is exactly as expected given the circumstances.

transfer: 10-11KB/s as received on Arduino
               13-14KB/s as received on RPi

With a datarate of 1MBPS the radios are capable of about 47-48KB/s with autoack enabled.

Note: In the transfer example, change the line:
```
if(role == role_pong_back){
     while(radio.available()){
```
to
```
if(role == role_pong_back){
     if(radio.available()){
```
Since the device is somewhat slow to respond, it will appear to hang when the Arduino is hammering it with data without the above changes. It is just stuck in the while(radio.available()) loop.

MRAA does appear to support BBB, but it looks like the documentation is not quite updated for BBB.
Just call ```sudo make install -B RF24_MRAA``` to build with libmraa, but I'm unsure of pin config.

desertgourd, 2015-02-18:
When I make install -B RF24_MRAA=1 I get an error missing mraa.h file a search dose not find it in the RF24-Updates.

prophet-roshak, 2015-02-18:
You should first install mraa to your BBB, using instruction from
https://github.com/intel-iot-devkit/mraa
You can use opkg, or you can build and make install it from that repo, so
it will install needed headers and shared libraries.

prophet-roshak, 2015-02-19:
Hi!

-lmraa linker switch tells linker to look for lib named mraa, e.g.
```libmraa.so``` (shared) or ```libmraa.a``` (static) in you libraries
path, so it looks like either file is on some other path which is not used
by linker, or it is not installed into system libs. If you built lib
yourself, you should find lib in your build dir.
Also, try doing ```ldconfig```, to update shared libraries cache

desertgourd, 2015-02-19:
@prophet-roshak Thanks for the info I will check it out.

desertgourd, 2015-02-19:
@prophet-roshak 
Hi, I found it   /mraa/build/src/libmraa.so, now the question is where do I need to put it so the linker can find it.  I also tried ldconfig and ran make again but still the same error.  This is my first time through this process.  Also when I ran the cmake .. and make from the mraa/build it generated a bunch of *.h files that were being called for when I ran make install -B RF24_MRAA=1.  I copied them over to the directory they were being look for in.  What is the best way to insure all the header files generated by the cmake mraa are included is there a way to set up a search path

TMRh20, 2015-02-19:
On RPi I had to do the following:

``` nano /etc/ld.so.conf  ```
Add the line  ``` /usr/local/lib/arm-linux-gnueabihf ```
Run ``` sudo ldconfig ```

desertgourd, 2015-02-19:
I copied libmraa.so to /usr/local/lib/libmraa.so
with nano  /etc/ld.so.conf 
added the line /usr/local/lib/libmraa.so
ran sudo ldconfig
ran /rf24libs/RF24# make install -B RF24_MRAA=1

still getting error:

/usr/bin/ld: cannot find -lmraa
collect2: ld returned 1 exit status
make: *** [librf24-bcm] Error 1

desertgourd, 2015-02-20:
I found that  /etc/ld.so.conf.d/arm-linux-gnueabihf.conf already exist and would be loaded by ldconfig with the first line of the  /etc/ld.so.conf file "include /etc/ld.so.conf.d/*.conf".  I still can't link -lmraa, do not know where to put it so linker finds it.

desertgourd, 2015-02-20:
I ended up putting libmraa.so in the /usr/lib/arm-linux-gnueabihf/ directory 
Ran  /rf24libs/RF24# make install -B RF24_MRAA=1 and it compiled without any errors.

prophet-roshak, 2015-02-20:
Have you tried to run ```make install``` from mraa build dir? As far as I know, cmake generated makefile should also have this build target. Can you test this to clarify mraa installation on BBB? (they still have not perfect docs on this, if you're not using repository they provide.

===
On my platform (galileo gen1), it will copy headers to ```/usr/local/include``` and lib to ```/usr/local/lib``` by default. The ```cmake_install.cmake``` however specifies this default path only if ```CMAKE_INSTALL_PREFIX``` is not defined, and it can be defined for you platform separately, on makefile generation step.
)

TMRh20, 2015-02-20:
``` I found that /etc/ld.so.conf.d/arm-linux-gnueabihf.conf already exist and would be loaded by ldconfig with the first line of the /etc/ld.so.conf file "include /etc/ld.so.conf.d/*.conf". I still can't link -lmraa, do not know where to put it so linker finds it. ```

Just to note, I actually saw the same thing on my system, so assumed that this step was not necessary. If you look at the original MRAA post, I posted my steps, but left this out, because I later assumed it was not necessary for the same reason. In reviewing my steps, removing and re-installing, I found that it seemed to be necessary. I don't understand the linking etc. well enought to know why, but that's my observation.

desertgourd, 2015-02-20:
The first time I tried to install the mraa master branch on the BBB I used the Build guide from  mraa/docs/building.md
Basic build steps:
mkdir build
cd build
cmake ..
make

The steps did not include the make install.  For a beginner this was something I overlooked and that is why none of the libraries and headers were installed.

I rebuilt using TMRh20's steps and everything went fine with the build. 
TMRh20's steps to install mraa master branch on BBB:
1. mkdir build
2. cd build
3. cmake .. -DBUILDSWIGNODE=OFF
4. sudo make install
5. sudo ldconfig

One observation, the mraa/docs/building.md steps took around 45 minuets to build and included the SWIG part.  TMRh20's method took less then a minute.  Question to TMRh20, if I use any of your python examples will I have to NOT use the option -DBUILDSWIGNODE=OFF on cmake in step 3?

In the gettingstarted.cpp file I used the following lines
//RPi Alternate, with mraa
RF24 radio(115,117);

It built and linked with no errors.  When I try to run ./gettingstarted I get the following error.

root@beaglebone:/home/rnh/rf24libs/RF24/examples_RPi# ./gettingstarted
*** glibc detected *** ./gettingstarted: realloc(): invalid next size: 0x00019a08 ***
Aborted

From what I have read this has something to do with trying to free memory that was already free.  Any suggestions.

desertgourd, 2015-02-20:
I have a feeling that the running ./gettingstarted error
root@beaglebone:/home/rnh/rf24libs/RF24/examples_RPi# ./gettingstarted
*** glibc detected *** ./gettingstarted: realloc(): invalid next size: 0x00019a08 ***
Aborted
is caused by the lines
GPIO::~GPIO() {
	// Prophet: this should free memory, and unexport pins when RF24 and/or GPIO gets deleted or goes out of scope
	this->close(gpio_ce_pin);
	this->close(gpio_cs_pin);
}
from the /arch/MRAA/gpio.cpp file.  I do not fully understand what is happening here or how to fix it.

The RF24 library was built with the command  /rf24libs/RF24# make install -B RF24_MRAA=1

TMRh20, 2015-02-20:
I'm guessing it just doesn't like the assigned pin numbers. If that is the case, it might be worth trying the [p9 pin numbers](http://electron14.com/wp-content/uploads/2014/08/beagleboneblack-pins.jpg) instead and/or testing with the included examples like blink-io.c.

``` Question to TMRh20, if I use any of your python examples will I have to NOT use the option -DBUILDSWIGNODE=OFF on cmake in step 3 ```
The RF24 python wrapper is independant of mraa, just a wrapper on top of RF24, so won't affect anything with building MRAA. The swignode option is disabled because I have an old version, and per the documentation ```You'll need at least SWIG version 3.0.2 and we recommend 3.0.5 to build the javascript & python modules. If your version of SWIG is older than this then please see above for disabling SWIGNODE. Otherwise you will get a weird build failure when building the javascript module. The python module builds with SWIG 2.x.```

desertgourd, 2015-02-21:
Changed RF24 radio(115,117);  to RF24 radio(27,25);  and still getting the same error with ./gettingstarted.  I will try testing the blink-io.c example and research what the correct pin numbers are.

prophet-roshak, 2015-02-21:
@desertgourd
I've checked platform definitions for mraa (/mraa/src/arm/) and it looks like they currently have only RPi officially supported. Code that was meant to support BBB wasn't pushed (or wasn't completed first) to repo yet. Sorry for misleading you into testing with uncomplete low level lib.

From what's in code, it is initializing BBB as it is a RPi, and should make a syslog message (you can check it with ```dmesg | tail``` after running any mraa using executable).

desertgourd, 2015-02-23:
Hi,
I have been working with michael-ring on testing his mraa fork - arm_beaglebone branch 
https://github.com/michael-ring/mraa/tree/arm_beaglebone
We have been making some progress and the GPIO examples are working.  The SPI examples are initializing but the device tree overlays have to be configured for the BBB.  The branch also uses different for the SPI CSN and there may be some errors in the P9 header Pin#27 GPIO assignment.  Will probably have it worked out in a few days.

desertgourd, 2015-02-25:
I have the TMRh20 RF24 library built with the MRAA drivers working with the /examples_RPi/gettingstarted  and have a round trip delay of a steady 7ms +- 1ms.  I am testing the other examples and will write up a summery of what has to be done to get it working.

TMRh20, 2015-02-26:
Well that was only 2 days, so you still have a day left hahaha. Seriously though, nice work.
I'm guessing we might be able to adjust the timing of things in RF24 to improve on that slightly, but don't hold me to it.

desertgourd, 2015-02-26:
I am testing and getting a better understanding of the hardware and software.  I would like to up the data rate from 1MBPS to 2MBPS but am not sure where that is set for the MRAA configuration.  I had to load a different SPI.dts board overlay file and on that you set the spi-max-frequency = MHz.  I have not seen any change in performance between 8MHz and 16MHz.  From what I understand the MRAA drivers default to the max-frequency independent of what you pass to them.  I am getting an avg. of 25KB/s transfer from the Nano to the BBB with 0 dropped packets, independent of the max-freq.  5KB/s BBB to Nano with avg. 680  dropped packets.  Did you say that was as timing issue.  It looks like it might be able to handle a data rate of 2MBPS.

desertgourd, 2015-02-26:
@michael-ring and @TMRh20

I have the MRAA working with the TMRh20 RF24 libraries and have tested the 'gettingstarted' and 'transfer' examples in the /examples_RPi directory. In the process I moved the RF24 board to the default SPI0 pins:
P9-17___CSN
P9-18___D1_MOSI
P9-21___D0_MISO
P9-22___SCLKP
The CE pin was assigned to:
P9-13___CE

The TRMh20 RF24 constructor sets up the CE and CSN with the command 'RF24 radio(CE-pin, CSN-pin);' with the pin numbering format as follows:
pin#=P8 Header Pin
pin#=P9 Header Pin + 46

So the RF24 constructor would be 'RF24 radio(59,63);'. The problem is that this dose not work because pin 63 is not a valid GPIO pin at this stage. Trying to run the transfer example generates the error:

root@beaglebone:/home/rnh/rf24libs/RF24/examples_RPi# ./transfer
RF24/examples/Transfer/
terminate called after throwing an instance of 'std::invalid_argument'
what(): Invalid GPIO pin specified
Aborted

gdb debuger output:
(gdb) r
Starting program: /home/rnh/rf24libs/RF24/examples_RPi/transfer
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
RF24/examples/Transfer/
terminate called after throwing an instance of 'std::invalid_argument'
what(): Invalid GPIO pin specified

Program received signal SIGABRT, Aborted.
0xb6db9f96 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
(gdb) Quit
(gdb)

syslog output:
Feb 26 16:39:30 beaglebone libmraa[2402]: libmraa version v0.5.4-142-gb46146a initialised by user 'root' with EUID 0
Feb 26 16:39:30 beaglebone libmraa[2402]: gpio: pin 63 not capable of gpio

Even when the BB-SPI0-01-00A0.dtbo overlay file has been enabled to setup the SPIO pins I still get the error. I got around the problem by assigning the CSN-pin to P8-8 which is a valid GPIO pin at boot with 'RF24 radio(59,8);'. The SPIO still outputs the CSN signal on P9-17 that was setup by the BB-SPI0-01-00A0.dtbo overlay file and where the CSN wire for the RF24 board is connected.

For the TMRh20 with the MRAA framework the CSN pin in the 'RF24 radio(59,8);' statement is redundent and generates errors. Is there any way to change the MRAA program so only the CE pin is passed 'RF24 radio(59);'? This fix will also require a change in the TMRh20 RF24 library.

The same problem was occurring when I had the CE pin assigned to P9-15 that was an invalid GPIO pin at boot and I moved it to P9-13.

TMRh20, 2015-02-27:
To set the datarate on the radios, just call ```radio.setDataRate(RF24_2MBPS);``` after initializing the radio via ```radio.begin();``` 

With the transfer example, for example, the radio retries are set at (2,15), which means the radios will retry each payload 15 times at an interval of 500us. This means that technically, the receiving device only has to handle 1 payload per 7.5ms, without dropping any payloads. The radios also have three 32-byte buffers, giving the receiving device slightly more time before payloads are lost.

With the above information noted, you may not see a large increase in speed using 2mbps, because the radio is capable of 47-48KB/s at 1MBPS datarate. There also may be a limit to SPI speed on BBB or possibly specific frequencies that it accepts for SPI, but I'm not that familiar with it. (nrf24l01 datasheet specifies a 0-10mbps SPI interface)

Regarding the timing, I was mostly referring to the delays in the RF24 driver, which may not be needed if the SPI/GPIO interface is not quite as fast as is with a microcontroller. This can be tweaked and tested once everything is working.

Regarding the CSN/constructor issue, I'm not sure I fully understand the cause of the problem, but RF24 generally needs to know the CSN pin unless mraa automatically toggles it for each transfer. Hopefully the mraa folks can provide more insight.

desertgourd, 2015-02-27:
I did some testing to check performance,  Their were 3 variables; the spi-max-frequency set in the device tree overlay file with a default of 24 MHz, The mspi frequency set in the /arch/mraa/spi.cpp, and the Air Data Rate set with radio.setDataRate(RF24_2MBPS); in the gettingstarted and transfer example files the /examples_RPi directory.

BB-SPI0-01-00A0.dtbo device tree overlay spi-max-frequency = 24000000
/arch/mraa/spi.cpp  mspi->frequency(8000000);
radio.setDataRate(RF24_2MBPS);
gettingstarted__Nano->BBB  6ms delay, no fails
gettingstarted__BBB->Nano  7ms delay, no fails
transfer________Nano->BBB  36KB/S no drops steady rate
transfer________BBB->Nano  36KB/S no drops 10K packets, 1K packets 22-50KB/S
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
BB-SPI0-01-00A0.dtbo device tree overlay spi-max-frequency = 24000000
/arch/mraa/spi.cpp  mspi->frequency(8000000);
radio.setDataRate(RF24_1MBPS);
gettingstarted__Nano->BBB  6ms delay, no fails
gettingstarted__BBB->Nano  7ms delay, no fails
transfer________Nano->BBB  25KB/S no drops steady rate 
transfer________BBB->Nano   5KB/S 600 dropped packets out of 10K packets
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
BB-SPI0-01-00A0.dtbo device tree overlay spi-max-frequency = 24000000
radio.setDataRate(RF24_2MBPS);
gettingstarted__Nano->BBB  6ms delay, no fails
gettingstarted__BBB->Nano  7ms delay, no fails
transfer________Nano->BBB  37KB/S no drops steady rate
transfer________BBB->Nano  39KB/S no drops 10K packets, 1K packets 34-54KB/S
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
BB-SPI0-01-00A0.dtbo device tree overlay spi-max-frequency =  24000000
radio.setDataRate(RF24_1MBPS);
gettingstarted__Nano->BBB  6ms delay, no fails
gettingstarted__BBB->Nano  8ms delay, no fails
transfer________Nano->BBB  25KB/S no drops steady rate
transfer________BBB->Nano   5KB/S 400 dropped packets out of 10K packets
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
BB-SPI0-01-00A0.dtbo device tree overlay spi-max-frequency =  8000000
/arch/mraa/spi.cpp  mspi->frequency(8000000);
radio.setDataRate(RF24_2MBPS);
gettingstarted__Nano->BBB  6ms delay, no fails
gettingstarted__BBB->Nano  7ms delay, no fails
transfer________Nano->BBB  36KB/S no drops steady rate
transfer________BBB->Nano  32KB/S no drops 10K packets, 1K packets 28-52KB/S
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
BB-SPI0-01-00A0.dtbo device tree overlay spi-max-frequency = 8000000
/arch/mraa/spi.cpp  mspi->frequency(16000000);
radio.setDataRate(RF24_1MBPS);
gettingstarted__Nano->BBB  6ms delay, no fails
gettingstarted__BBB->Nano  8ms delay, no fails
transfer________Nano->BBB  25KB/S no drops steady rate 
transfer________BBB->Nano   5KB/S 400 dropped packets out of 10K packets
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

Conclusions:
1. The device tree overlay spi-max-frequency has no effect and can be left at the 24 MHz default.
2. The mspi frequency set in the /arch/mraa/spi.cpp file has no effect and can be left a 8 MHz.
3  The Air data rate has a grate effect with 1MBPS having lower transfer rates and considerable dropout. The 2MBPS gives an average 10K packet rate of 36 KB/S and the 1K packet rate at the receiver varied from 22-54 KB/S. Their are probably timing issues with the 1MBPS rate.  The 2MBPS may have some timing issues preventing a steady 54 KB/S transfer rate.

TMRh20, 2015-02-28:
Thanks to comments by michael-ring I realized that both SPIDEV and MRAA should be able to toggle the CS pins  via kernel. I initially had problems getting it working with SPIDEV because the pin had previously been configured for GPIO.

This allowed me to change the API for SPI on linux devices, and have the SPI BUS specified in the constructor instead of the specific pin number. (defaults to 0 if number > 1 is specified)

``` RF24 radio(ce_pin,SPI_BUS_NO); ```
SPIDEV with BBB: ``` RF24 radio(115,0); ```  or ``` RF24 radio(115,1); ```

I assume it will work the same with BBB using MRAA, but with a different CE pin specified, and should resolve the GPIO conflicts/errors.

This also seems to bring performance up for both SPIDEV and MRAA, and I think may resolve some of the issues when sending from BBB to Arduino as well. The code still needs some cleanup, but this should have it working as it should I believe.

[9 more replies]

closed 2015-10-20 by Avamander