nRF24/RF24 #71: RPI <-> RPI communication

uggima, 2015-02-05:
loving your/everyone's work on this, but slight issue with communication between two rpi's running your code. i know that setup is not the main scope of the project but if someone could point me in the right direction would be happy to fork the code and do the work myself

between a rpi and arduino packets are fragmented at 90bytes, between two rpi's they aren't (or i've got something very wrong :S). im not great at C, but i've had a read through RF24Network RF24 and RF24toTAP's main files and can't find which part does the fragmenting

libraries were installed using your install.sh and manually cloning/building/installing them, then editing rf24totun to enable the printing of the packet tx attempts. 

i include some error logs - http://pastebin.com/7XE9ANGF

TMRh20, 2015-02-05:
Hello, thanks for the well thought out trouble report and error logs, more information always makes things easier to figure out.  
  
The default MAX_PAYLOAD_SIZE in RF24Network is 144 bytes, and I think that default size limit is your only problem.
  
With ICMP packets, using TAP/Ethernet, a standard ping of 56 bytes will require an additionlal 28 bytes of payload size + 14 bytes for Ethernet Headers = 42 bytes of overhead.

With TCP/IP packets, using TAP/Ethernet, a standard request of 90 bytes will require an additional 54 bytes in header overhead. (20 - IP + 20 - TCP + 14 - Ethernet)

This typically makes 90 bytes the maximum message size, given the default MAX_PAYLOAD_SIZE of 144 bytes.
  
To test this, just edit RF24Network_config.h with a text editor, and raize the MAX_PAYLOAD_SIZE to 1514 on the Raspberry Pi devices. From there, you should be able to connect from RPi to RPi using any standard methods.  
  
One note: Technicaly this issue could be filed under [RF24toTUN](https://github.com/TMRh20/RF24toTUN) or [RF24Ethernet](https://github.com/TMRh20/RF24Ethernet) but not a big deal.

uggima, 2015-02-05:
Thanks for the quick reply and information! 
its been a while since i read up about the lower levels of networking, packet headers etc... will test that, n report back

and wasn't sure where to post the issue so shoved it at the center of it and hoped for the best :)

uggima, 2015-02-05:
ok! works perfectly! thanks a lot.
i had tried playing with the MTU's of various interfaces which i thought would of effected this before posting the issue but anyway... 
i will leave this open for a short wile in case anyone else wants to comment.

TMRh20, 2015-02-05:
Excellent, if you don't mind just leaving it open as a reminder for me to update the documentation. I can close it once that is done.

uggima, 2015-02-05:
fine fine :D

TMRh20, 2015-02-14:
Troubleshoting section added at http://tmrh20.github.io/RF24Ethernet/Troubleshooting.html

closed 2015-02-14 by TMRh20