nRF24/RF24 #77: Power on Reset delay for Arduino

jorihardman, 2015-02-24:
I had a question about the power on reset (T<sub>POR</sub>) delay detailed in Table 11 of the v2.0 product specification. I can see that for linux, there is a 100ms delay for the power ramp up time (T<sub>PUP</sub>), but I don't see an analogous 10ms delay for Arduino. It seems that the board is not officially in "Power Down" until this 10ms phase has passed. Is this a purposeful omission, or should it be added to the #if statements in the begin() function?

TMRh20, 2015-02-25:
Well, as far as I am aware, a need for it has never really been established, so it could be called a 'purposeful omission'.

jorihardman, 2015-02-25:
Fair enough. It doesn't seem to be necessary, but the specification does mention a very specific maximum time for it (although with no explanation) of 10.3ms. It just seemed odd that it was missing considering the Linux sections implement the maximum delay for T<sub>PUP</sub>.

jorihardman, 2015-02-25:
Fair enough. It doesn't seem necessary. I just found it odd that it was missing, considering the Linux sections do follow the spec for T<sub>PUP</sub>.

closed 2015-02-25 by jorihardman