nRF24/RF24 #79: Reliable transmission always fails

eskildsf, 2015-03-02:
Hello.

I have set up the RF24 library and installed the GettingStarted sketch where I set "RF24 radio(9,10);" instead of "RF24 radio(7,8);" on my Arduino Uno so that it corresponds to the connections given below:
NRF | Arduino
 1     | GND
 2     | 3v3
 3     | D9
 4     | D10
 5     | D13
 6     | D11
 7     | D12
 8     |

On my Raspberry Pi (Model B, rev 2) I have connected it as follows (in terms of https://superpiboy.files.wordpress.com/2014/07/raspberry-pi-rev2-gpio-pinout.jpg):
NRF | Raspberry Pi B rev 2
 1     | 6
 2     | 1
 3     | 22
 4     | 8
 5     | 23
 6     | 19
 7     | 21
 8     |
I installed the libraries in the way that you showed in the video https://www.youtube.com/watch?v=rBAIqAaRu0g.

Furthermore I have connected 10µF electrolytic capacitors on the GND/VCC connectors on the NRF24L01's.

On the Arduino I can send packages and I can receive them on the Raspberry Pi with the gettingstarted program (rf24libs/RF24/examples_RPi/gettingstarted). However, I always get "Now sending... failed Failed, response timed out." in the Arduino console.

When I put the Raspberry Pi into send mode I also get "Now sending... failed. Failed, response timed out." but I don't see anything on the Arduino console.

The output from the printDetails() function on the Raspberry Pi is:
```
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x65646f4e32
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
```

How can I debug this? I really want it to work :-). 

Thank you!

spaniakos, 2015-03-02:
Have you enables spi usinh rpi-config in the pri ?

>
> Hello.
>
> I have set up the RF24 library and installed the GettingStarted sketch
where I set "RF24 radio(9,10);" instead of "RF24 radio(7,8);" on my Arduino
Uno so that it corresponds to the connections given below:
> NRF | Arduino
> 1 | GND
> 2 | 3v3
> 3 | D9
> 4 | D10
> 5 | D13
> 6 | D11
> 7 | D12
> 8 |
>
> On my Raspberry Pi (Model B, rev 2) I have connected it as follows (in
terms of
https://superpiboy.files.wordpress.com/2014/07/raspberry-pi-rev2-gpio-pinout.jpg
):
> NRF | Raspberry Pi B rev 2
> 1 | 6
> 2 | 1
> 3 | 22
> 4 | 8
> 5 | 23
> 6 | 19
> 7 | 21
> 8 |
> I installed the libraries in the way that you showed in the video
https://www.youtube.com/watch?v=rBAIqAaRu0g.
>
> Furthermore I have connected 10µF electrolytic capacitors on the GND/VCC
connectors on the NRF24L01's.
>
> On the Arduino I can send packages and I can receive them on the
Raspberry Pi with the gettingstarted program
(rf24libs/RF24/examples_RPi/gettingstarted). However, I always get "Now
sending... failed Failed, response timed out." in the Arduino console.
>
> When I put the Raspberry Pi into send mode I also get "Now sending...
failed. Failed, response timed out." but I don't see anything on the
Arduino console.

eskildsf, 2015-03-02:
Yes. I have enabled it with raspi-config. Also, I should tell you that printDetails() yields no output in the Arduino console.

spaniakos, 2015-03-02:
It seems weird that you have no connection.
Have you tried to change the pins in the rpi example ? Or have yoy tried
with out the μf capacitors ?
It seems that the cobbection is working onw way,
Arduino > RPi
When i had this problem, it was just a misconfiguration of the miso/mosi
pins on the RPi

eskildsf, 2015-03-02:
Yeah, I tried without the capacitors at first. Then I saw some stuff online where people recommend it because the Arduino 3v3 might not be that stable.
I have not tried to change the pins in the rpi example.

A misconfiguration? How? I might want to look into that :-).

spaniakos, 2015-03-02:
Its either the physical pins , or the pins in the rpi program.
Check the pins there :-).
Its tha same like using rf24(7,8) rf24(9,10) but with some string
definitions.
Check all the examples and you will get what i am talking about.

eskildsf, 2015-03-02:
I just tried placing the MISO/MOSI on some other pins but then I get
```
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1	 = 0x0000000000 0x0000000000
RX_ADDR_P2-5	 = 0x00 0x00 0x00 0x00
TX_ADDR		 = 0x0000000000
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x00
RF_CH		 = 0x00
RF_SETUP	 = 0x00
CONFIG		 = 0x00
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = Disabled
PA Power	 = PA_MIN
```
which doesn't look good so I think the placement is right.

Which pins did you connect together to make it work for you?

spaniakos, 2015-03-02:
I am not near my boad until later .so i canot tell you the exact
configurations i use. But i could do that in some hours.

eskildsf, 2015-03-02:
That would be great!! Do you know if  Iam supposed to get anything out of radio.printDetails() on the arduino? There is nothing in the console.

spaniakos, 2015-03-02:
Without the printf.begin() and the include "printf.h" it wont work

eskildsf, 2015-03-02:
This is amazing... I must have rewired it 10000000 times by now and suddenly, it works when I switched to my Arduino Nano.
Well, thank you for your assistance. If I figure out why I didn't work before then I'll put it here for others to make use of.

Thank you :-).

closed 2015-03-02 by eskildsf

eskildsf, 2015-03-02:
It turns out that the issue was with the Arduino Uno. I have tested extensively and it will detect the NRF and print the radio.printDetails(); just fine, but it won't receive anything.
My solution was to use an Arduino Nano I had lying around instead.

I just think people should be aware that this can be an issue too. I have't had other issues with the board though so it was quite a surprise. I wonder if I would be able to measure a low voltage on the 3v3 output of the Uno but I don't have any multimeter so I cannot experiment with this.

Regards

ElectricRCAircraftGuy, 2015-03-02:
In my limited experience with these thus far, it seems to me that they are not reliable unless I use a good 3.3V external regulator...separate from the Arduino. I've had connection problems too. On a couple quick tests the problems disappeared once I used a separate, dedicated 3.3V 100mA or greater voltage regulator.

TMRh20, 2015-03-02:
Just a couple quick notes that might help somebody:

1. nrf24l01 modules are much simpler to get going, and work much more consistently when using smal PCBs with direct, soldered connections. When using breadboards, keep leads as short as possible. A 5us delay was added in December, (after toggling CSN pin), which appears to have greatly reduced the number of hardware issues encountered using breadboards.
 
2. PA enabled modules: These are the large modules, usually with external antenna, and either require a separate power supply, or need to be driven at a low PA level, since the Arduno power supplies cannot handle it.

3. Pro Tip: Use a 2v LED just like in the [3-pin ATTiny hack](http://nerdralph.blogspot.com/2014/01/nrf24l01-control-with-3-attiny85-pins.html) with a large (100uf etc) capacitor when using nrf24l01 modules with a 5v power supply. This has been very consistent and reliable for me via Arduino Pro Minis over a large period of testing.

4. The closest I have to an Uno is an Arduino Duemilanove, and it works nicely with non-pa nrf24l01 modules, running off the 3.3v output. The LED hack in step 3 can be used as an alternative.

ElectricRCAircraftGuy, 2015-03-03:
Good tips. Thanks.