nRF24/RF24 #84: RPi2?

WERAQS, 2015-03-05:
hello there,

downloaded the code to new RPi2, compiled it, but it seems to hang when running the codes, no Rx or Tx available, and it can't communicate with SPI devices, any updated about that?

-is it a multicore issue?
-or a new SPI layout?

also, is it possible to use same nRF24 Transreceiver and quadcore to gather node from different sensor layers simultaneously or bit of a delay?

eskildsf, 2015-03-05:
Hello. Did you remember to enable SPI with the rpi-config tool? :-)
It is not enabled by default so that may be the source of your problem on a new Raspbian installation.

Regards

WERAQS, 2015-03-05:
thanks for comment but this is my 4th raspberry pi (B, A+, B+, 2B) and others working fine..

TMRh20, 2015-03-05:
Per https://github.com/TMRh20/RF24/pull/72 the [Updates](https://github.com/TMRh20/RF24/tree/Updates) branch is tested and working on RPi2 . I would assume any issues to be related to configuration, wiring, etc.

``` is it possible to use same nRF24 Transreceiver and quadcore to gather node data from different sensor layers simultaneously or bit of a delay? ```

Not sure exactly what you mean by 'different sensor layers' , but any of the RF24 layers (RF24,RF24Network,RF24Mesh,RF24Ethernet) can be utilized independently of the others, providing the configuration is correct.

WERAQS, 2015-03-05:
maybe bad nRF24 batch, I'll test with old ones, thanks!

closed 2015-03-13 by TMRh20