nRF24/RF24 #85: IRQ management

Slartitorto, 2015-03-05:
On the RPI, the radio.available() in loop statement get up the cpu at 98%
There is a way to use irq or similar method to put the system in wait for receiveing ?

TMRh20, 2015-03-05:
Interrupts are a function of the system, not necessarily the radio driver, and I've not done much on RPi with interrupts.

The BCM2835 driver seems to have some event handling functions ( http://www.airspayce.com/mikem/bcm2835/group__gpio.html ) but I've not done much with them myself.

Python seems to support interrupts also: http://raspi.tv/tag/python-raspberry-pi-gpio-interrupts

One can also utilize time-wait delays etc. to reduce CPU load between polling for available(); since the device will generally only receive up to 1500payloads/second in 1MBPS mode, and has 3 FIFO buffers.

Slartitorto, 2015-03-05:
Do you think a sleep(50) between polling (I mean 50 ms) does not cause packet lost at 250KBPS ?

TMRh20, 2015-03-05:
That depends on many variables like how often you are sending, the length of the payload, use of dynamic payloads, etc.

closed 2015-03-13 by TMRh20

farconada, 2015-08-09:
I saw that example, and could be added to the RF24
http://balazs.kutilovi.cz/nrf24-on-raspberry-pi/