nRF24/RF24 #86: Pi Bit-Banging Spi

Hoche96, 2015-03-06:
Is there a way to use the NRF24L01 with another Pin as CE0 or CE1? Would be great,  because my display already use both.

TMRh20, 2015-03-07:
Technically yes, there is a way to do it with another pin as CE0 or CE1, but it isn't supported in the current driver.

With the new RF24/arch layout in the Updates branch, it is possible create another template for RPi and/or Linux that uses GPIO to drive the radio CSN pin. It likely won't perform quite as well, but it could be done. Of course, if your display could use another pin, that would be the easiest solution.

Hoche96, 2015-03-07:
Thanks for the reply.  Would be great when someone implent it.

Hoche96, 2015-03-07:
I found another solution, any chance to get the lib working with i2c? There is a spi to i2c interface for the nrf24l01.

TMRh20, 2015-03-07:
If you are talking about something like [this](http://www.elecrow.com/nrf24l01-wireless-shield-spi-to-i2c-interface-for-arduino-p-737.html) it doesn't require a library beyond the i2c lib, but is very limited in its capabilities from the looks of it. RF24 modules themselves only use SPI.

There is an [RF24 Remote]( https://github.com/mz-fuzzy/RF24Remote ) library that lets you make RF24 calls over serial etc. but again, not quite as fast as direct SPI.

Hoche96, 2015-03-07:
Ah thanks the RF24 Remote should do the work,  so I use an arduino via USB.

closed 2015-03-11 by Hoche96