nRF24/RF24 #90: virtual method handling in RF24 class

mz-fuzzy, 2015-03-18:
(1) as a consequence of redesign/new platforms support in 'Updates' branch, there was a virtual destructor added to RF24 class. This implies vtable creating in runtime. In case of avr targets, it's in RAM and unnecessarily consumes precious space there while it's not needed.

(2) Beside this, for overriding in derived classes, it would be good to have virtual methods in RF24.

My proposal for solving both:
- drive RF24's 'virtual' usage in library's xxx_config.h files by defining 'VIRTUAL' to nothing or to 'virtual'
- in RF24.h declare all public methods with 'VIRTUAL', for example
```
VIRTUAL void begin(void);
```

TMRh20, 2015-03-21:
1. I cannot seem to detect any change in memory or program space on Arduino with the virtual destructor added.
2. I don't understand why it would be a good idea to make every public function virtual in weighing simplicity vs the benefits. Can you demonstrate a use-case where this would be practical?

mz-fuzzy, 2015-03-21:
(1) GettingStarted sketch, Updates branch as it is:
```
Sketch uses 6,858 bytes (21%) of program storage space. Maximum is 32,256 bytes.
Global variables use 232 bytes (11%) of dynamic memory, leaving 1,816 bytes for local variables. 
```

Same sketch and branch, removed 'virtual' from RF24's destructor:
```
Sketch uses 6,178 bytes (19%) of program storage space. Maximum is 32,256 bytes.
Global variables use 212 bytes (10%) of dynamic memory, leaving 1,836 bytes for local variables. 
```

(2)
a. The header file then can serve as interface and there could be multiple implementation, like for example I have in RF24Remote
b. Anyone could derive the class and override any method's behaviour (like printDetails or any other) while still keeping possibility to use such derived class in other projects/libs where RF24 is expected.

TMRh20, 2015-03-21:
Hmm, I must have been editing the wrong file or something, because I am now seeing similar results with the destructor removed.

I still don't understand why it would be a good idea to make every public function virtual in weighing simplicity vs the benefits. One of my main goals was/is to provide a standardized interface and core RF24 driver that works roughly the same across various platforms. I've also spent quite a lot of time to identify the optimal processes and timing, for utilizing every available feature on the chip on various platforms, with similar behaviour. In the case of RF24Remote, you've introduced a large number of changes that introduce changes in timing and behaviour, which kind of flies directly in the face of using a standardized interface. As I've suggested before, I lean more towards the concept of improving the standard interface to cover as many scenarios as possible. IMHO RF24Remote could likely be built with faster response and higher throughput just using standardized RF24 functions.

mz-fuzzy, 2015-03-23:
ok...
*for RF24Remote*: RF24 startardized functions are used where possible. If you use Serial interface for remote driving, it's a 1:1 call of RF24's functions. For VUSB, it's different, because VUSB needs frequent calls of polling function and it's not feasible with RF24 *blocking* methods (and there are some), so 1:1 call is not possible in such case.

I have done some experimenting with speed of both Serial and VUSB, and the speed limit is on used remote interface. For VUSB I can't do much to improve, it's slow protocol (usb 1.1 + sw polling in vusb etc...).

*for virtual methods*: I'm tageting this scenario: Imagine that user has the nrf24 module connected on  RPi directly via SPI, and additionally via RF24Remote. RF24Remote's class RF24Frontend implements remote calls via Serial/VUSB by deriving and overriding RF24 class, so both RF24/SPI and RF24Remote can be used in parallel whereever RF24 class is expected. If you have any better suggestion how to do that, feel free to share it.

In my opinion having opiton to have virtual methods does not influence interface simplicity/timing. Anyway it's your lib, works perfectly, I definitely don't what to disregard that. Well, if I did not convince you (that seems to be the case) feel free to close this ticket, I can live also without such adaptations :-)

closed 2015-10-20 by Avamander