nRF24/RF24 #92: Teensy 3.1 eclipse C/C++ RF24 compile error

epa64, 2015-03-22:
Hello,

Thank's for this lib.
I work on a project with node.js and Teensy MESH Senssor with nRF24L01+ module.

I am sample compilation problem with the new RF24 lib :"log":

#############################################
 In file included from /home/epa/Documents/DEVELOPPEMENT/ARDUINO/TEENSY_3_1/libraries/RF24/arch/MRAA/gpio.cpp:6:0:
/home/epa/Documents/DEVELOPPEMENT/ARDUINO/TEENSY_3_1/libraries/RF24/arch/MRAA/gpio.h:15:20: fatal error: mraa.hpp: No such file or directory
compilation terminated.
make: *** [Libraries/RF24/arch/MRAA/gpio.cpp.o] Erreur 1
###############################################

My arch is teensys ?

With the old version (no arch directory), the sample compile with no problem.

PS: sorry for my bad inglish.

epa
best regard's

TMRh20, 2015-03-23:
This should only happen if *MRAA* is #defined somewhere.

The line 31 in RF24_config.h should probably be reverted to use RF24_MRAA
instead:

    #if ( defined (__linux) || defined (LINUX) ) && defined( __arm__ ) ||
defined(RF24_MRAA)

Changing that line to the above should resolve the issue

epa64, 2015-03-27:
I solved my problem.
I have to delete directories in  "arch", except tennsy.

Thank
Eric

closed 2015-03-27 by epa64

prophet-roshak, 2015-03-31:
With Eclipse you can also exculde files from build (via context menu in Project Explorer). Point is that Eclipse itself auto-generates makefile based on your personal project settings.