nRF24/RF24 #93: Attiny84

tguillaume02, 2015-03-24:
Hi,

I would like send data with attiny84 and RF24 and RF24Network.
I have try the sketch rf24pin85 on attiny84 and GettingStarted on UNO but I don't receive maybe wrong pin, I don't know.
I have try different pin without succes.
Could you help me please.

Thank you

TMRh20, 2015-03-26:
As far as I am aware there are no issues with ATTiny84, so it would likely be a configuration, hardware or wiring issue, etc..

tguillaume02, 2015-03-26:
Ok, thanks for your reply.
It's ok with rf24pin85 and GettingStarted.
Solution use 
http://code.google.com/p/arduino-tiny  and not  https://github.com/damellis/attiny/

Thanks

TMRh20, 2015-03-26:
Thanks for the info. This is mentioned in the example, but I guess it should be added to the ATTiny documentation too. The https://github.com/damellis/attiny/ core seems to be fairly popular and referenced in a lot of places, so it might be worth looking at updating to that...

joshuajnoble, 2015-03-31:
I can also confirm that https://github.com/damellis/attiny/ doesn't work with an ATTiny84 

@tguillaume02 could you share which version of the Arduino IDE you're using that you've had work for you?

tguillaume02, 2015-03-31:
@joshuajnoble IDE 1.0.6 work for me

closed 2015-12-14 by Avamander