nRF24/RF24 #94: Dropped packets

tony-gosu, 2015-03-26:
Thanks for the great library! I was wondering, in your test, did you see many dropped packets? I have read all over the web that these radios sometimes needs a capacitor between vcc and gnd. 

(1) Have you found the capacitor to help?
(2) Does having a smaller packet size reduce packet loss?

Thanks in advance!

TMRh20, 2015-03-27:
> I was wondering, in your test, did you see many dropped packets?

No?
> (1) Have you found the capacitor to help?

Yes, [in some cases](https://github.com/TMRh20/RF24/wiki/GettingStarted-sketch-failing-to-send:-how-to-diagnose%3F), usually with breadboards.
> (2) Does having a smaller packet size reduce packet loss?

Possibly

There is a lot of info about general nrf24l01 usage available via google, forums, included examples, etc.

gritschu, 2015-04-01:
I had a lot of dropped packets by sending between an Arduino (Nano clone) and a Raspberry Pi B+. Adding capacitors solved the problem completely in my case. (Had had the most trouble with the module on the Raspi side.) You should give it a try.

tony-gosu, 2015-04-01:
Hi gritschu. Thats for the response. What kind of capacitor and what size
wotked best for you?

Thanks. Tony

gritschu, 2015-04-01:
Hey tony-gosu,
I had the hint from this site:
http://hack.lenotta.com/arduino-raspberry-pi-switching-light-with-nrf24l01/
They suggested a 10 μF capacitor. (They also have a photograph there, how to solder it.)
I happened to have one 10 μF and one 100 μF at hand and just used them, which worked perfectly fine.

closed 2015-06-17 by TMRh20