nRF24/RF24 #95: RF24MESH with teensys 3.1 Problem  (Master down after 4 recieved)

epa64, 2015-03-27:
Hello,

I made test with a teensys 3.1 and a RF24L01+ module .
I meet a problem with the RF24MESH lib.
The examples with RF24 er RF24Network ( developments) work well.
With the examples " RF24Mesh_Example " and "RF24Mesh_Master", the " node " sending 4 requests and ask for one renew address, the Master's  does not (show) what it receives and seems not to answer any more. When i restart the Master, the node send, again,  4 request and the master never show the recieves data.

/****************************************/
Send OK: 37082
Send OK: 38082
Send OK: 39082
Send OK: 40082
 Write error Check Connection 
Renewing Address
/*************************************/

The tennsy asks a particular configuration in RF24Mesh_config.h ?

For my project (GPL3) I have a network of teensy (sensors) which communicates by RF24 with the Base. 
The Base receives the data of sensors and write them on the serial of the teensy (usb).

Best's regard
Eric

TMRh20, 2015-03-28:
This was a bug, same issue affected the Due exactly the same.
You will likely want to update RF24Network_Dev and RF24Mesh. 

Thanks for the info, please let me know if this fixes it for you as well.

epa64, 2015-03-29:
Great, I just try with two Teensys 3.1 MESH's work fine.

I try  with 5 Teensys, the next week.

Best's regards

Eric

slax32, 2015-03-29:
😃

closed 2015-06-17 by TMRh20

epa64, 2015-06-19:
Hello,

I made test with a teensys 3.1 and a RF24L01+ module .
I meet a problem with the new RF24Network_Developpment RF24MESH_Master lib.
The examples with RF24 er RF24Network_Developpement work well.
The write method of RF24Network_Developpment return always false (0), is not the same with RF24Network_Master.
The Example RF24Mesh_Example_Master and RF24Mesh_Example doesn't work fine.
The Node seems to lock up : 
11770 MSH: No poll response from level 3
13906 MSH: Poll response > -64dbm
13906 MSH: Got poll from level 0
 RF24Mesh --> HEADER after Write type 195, reserved 1,to_node 0 
13908  MSH: Request address this node: 04
Set address 04 rcvd 04
 15631 MSH: No poll response from level 1
15872 MSH: No poll response from level 2
16049 MSH: No poll response from level 3
17615 MSH: Poll response > -64dbm
17615 MSH: Got poll from level 0
 RF24Mesh --> HEADER after Write type 195, reserved 1,to_node 0 
17616  MSH: Request address this node: 04
Set address 04 rcvd 04
 18315 MSH: No poll response from level 1
19580 MSH: No poll response from level 2
20717 MSH: No poll response from level 3

Thank's for your help.

Best's regards
Eric Papet