nRF24/RF24 #96: Setting pins, channels, pipes, address, and other required settings

chisight, 2015-04-01:
I'm starting from https://www.youtube.com/watch?v=rBAIqAaRu0g RF24Ethernet - TCP/IP for nrf24l01+ radios over RF24Network.

The interface to the various modules seems to require setting various identical or paired settings on both ends, but there seems to be little documentation on exactly what must match or how to set it so that install.sh will make use of it.  A quick grep finds around 60 possible places for setting the CE and CSN pins alone.

install.sh also compiles many things but there is no clear instructions on how to recompile once the settings are adjusted.  I'm currently just re-running the installer, ignoring the git errors, to recompile but have no clue if this is appropriate.

Once set, diagnostics procedures are needed to work up from hardware to layer 7.  On several attempts, I discovered I'd miswired CSN and there was no indication that was a problem.  Right now I'm stuck with a generic "Destination Host Unreachable" and have no clue which layer the problem is in or how to find out.

I'm working with both a RPi and an Arduino Nano and have had unpredictable results in my various attempts at trying to use RF24Ethernet and InteractiveServer.ino with RF24toTUN and I'm not even sure if that is an appropriate way to accomplish my goals.  I need a solution that allows adding radios/projects without big rewrites, works RPi to Arduino or Arduino to Arduino, and offers reliable communication at a range of 10 meters that is close to a drop in replacement for the USB interface.  It does not need to support the IDE's serial (USB) monitor or programming over this link. 

Steps and diagnostics for "Destination Host Unreachable" problem:

1) close Arduino IDE wipe everything on RPi and desktop.  
On desktop:
rm -rf ~/Arduino/libraries/RF24*; rm -rf ~/RF24*; rm -rf ~/Downloads/RF*
on RPi: 
sudo rm -f /usr/local/bin/gettingstarted /usr/local/bin/gettingstarted_call_response /usr/local/bin/pingpair_dyn /usr/local/bin/rf24totun /usr/local/bin/transfer; sudo rm -rf ~/RF24; sudo rm -rf /usr/local/lib/*rf24; sudo rm -rf /usr/local/include/RF24*

2) download:
desktop:
wget -O RF24Ethernet.zip https://github.com/TMRh20/RF24Ethernet/archive/master.zip && \
wget -O RF24Mesh.zip https://github.com/TMRh20/RF24Mesh/archive/master.zip && \
wget -O RF24.zip https://github.com/TMRh20/RF24/archive/master.zip && \
wget -O RF24Network.zip https://github.com/TMRh20/RF24Network/archive/Development.zip
unzip, rename folders removing -*, move to ~/Arduino/libraries/
RPi:
wget http://tmrh20.github.io/RF24Installer/RPi/install.sh
chmod +x install.sh 
sudo ./install.sh 
(yes to everything)

3) edit for local hardware and run:
Arduino:
open IDE and File/Examples/RF24Ethernet/InteractiveServer
edit to RF24 radio(A5,10);
compile, upload, open serial monitor
RPi:
reboot RPi
edit ~/RF24/rf24libs/RF24toTUN/rf24totun.h
  RF24 radio(RPI_V2_GPIO_P1_22, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ);
re-run install.sh, again saying yes to all.
edit  ~/RF24/rf24libs/RF24toTUN/rf24totun_configAndPing.sh 
  MYIP="10.10.2.${1}/24"
  PINGIP="10.10.2.${2}"
sudo ~/RF24/rf24libs/RF24toTUN/rf24totun_configAndPing.sh 2 4

4) results:
sudo ~/RF24/rf24libs/RF24toTUN/rf24totun_configAndPing.sh 2 4
Successfully attached to tun/tap device tun_nrf24
PING 10.10.2.4 (10.10.2.4) from 10.10.2.2 tun_nrf24: 56(84) bytes of data.
From 10.10.2.2 icmp_seq=1 Destination Host Unreachable
From 10.10.2.2 icmp_seq=2 Destination Host Unreachable
From 10.10.2.2 icmp_seq=3 Destination Host Unreachable

--- 10.10.2.4 ping statistics ---
3 packets transmitted, 0 received, +3 errors, 100% packet loss, time 2000ms
pipe 3

5) diagnostics:
add radio.printDetails(); and uncomment printf on Ardunio
set PRINT_DEBUG 3 in rf24libs/RF24toTUN/rf24totun.h and running install again
RPi:
sudo ~/RF24/rf24libs/RF24toTUN/rf24totun_configAndPing.sh 2 4
Successfully attached to tun/tap device tun_nrf24
================ SPI Configuration ================
CSN Pin          = CE0 (PI Hardware Driven)
CE Pin           = Custom GPIO25
Clock Speed      = 8 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5     = 0x33 0xce 0x3e 0xe3
TX_ADDR          = 0xcccccc3ccc
RX_PW_P0-6       = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA            = 0x00
EN_RXADDR        = 0x3f
RF_CH            = 0x61
RF_SETUP         = 0x07
CONFIG           = 0x0f
DYNPD/FEATURE    = 0x3f 0x05
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX
PING 10.10.2.4 (10.10.2.4) from 10.10.2.2 tun_nrf24: 56(84) bytes of data.
Tun: Successfully read 42 bytes from tun device
Radio: Sending 42 bytes ... 
********************************************************************************
radio TX
Buffer size: 42 bytes
������RFRF24

********************************************************************************
Tun: Successfully read 2a bytes from tun device
Radio: Sending 2a bytes ... 
********************************************************************************
radio TX
Buffer size: 2a bytes
������RFRF24

********************************************************************************
Tun: Successfully read 2a bytes from tun device
Radio: Sending 2a bytes ... 
********************************************************************************
radio TX
Buffer size: 2a bytes
������RFRF24

********************************************************************************
From 10.10.2.2 icmp_seq=1 Destination Host Unreachable
From 10.10.2.2 icmp_seq=2 Destination Host Unreachable
From 10.10.2.2 icmp_seq=3 Destination Host Unreachable

--- 10.10.2.4 ping statistics ---
3 packets transmitted, 0 received, +3 errors, 100% packet loss, time 2000ms
pipe 3

Arduino:
start
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xcccccc3ccc 0xcccccc3c3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xcccccccc3c
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3f
EN_RXADDR	 = 0x3f
RF_CH		 = 0x61
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x05
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

Pipes clearly don't match but I'm unable to locate those strings in the source.  AA also appears to be mismatched.

I'm sometimes able to get ping to work by switching to /usr/local/bin/rf24totun -m but it's unreliable and I'm unable to identify a pattern for when it works and when it doesn't.

Going back to basics with GettingStarted.cpp/.ino results in mismatched EN_AA.  Adding   radio.setAutoAck(true); to both ends resolves that.  Adding a delay(50); to the Arduino gives the RPi time to switch the radio from send back to receive and lets most packets succeed.  At least tells me hardware is working.  Intermittently, even this fails and will not work until the RPi has been rebooted.  The Pi is Type: Model B, Revision: 1, Memory: 256MB, Maker: Egoman running Raspbian and is up to date.

Side note, strncpy_P(bf, PSTR(... and strncpy_P(buffer, i,... is much smaller than sprintf().

TMRh20, 2015-04-01:
@chisight Thanks for your input/feedback. RF24Ethernet is an experimental open-source library in development and beta testing, so users are generally expected to be 'advanced users' with a basic knowledge of RF24, building programs in C/C++ and IP/Networking. Your comments seem to indicate that you are likely making this much more complicated than it needs to be.

> The interface to the various modules seems to require setting various identical or paired settings on both ends, but there seems to be little documentation on exactly what must match or how to set [...]

Have you noticed the included examples and/or the extensive documentation at http://tmrh20.github.io ? I am completely baffled at what you are doing regarding grep and CE/CSN.

> install.sh also compiles many things but there is no clear instructions on how to recompile once the  settings are adjusted. I'm currently just re-running the installer, ignoring the git errors, to recompile but have no clue if this is appropriate.

You mean like the instructions [here](http://tmrh20.github.io/RF24/RPi.html) ? Since you are using a beta version of an experimental library in development, it is simply assumed that you would have many of the required troubleshooting skills. Also, if you read the instructions the installer provides, it literally tells you exactly how to manage upgrading/reinstalling.

> Once set, diagnostics procedures are needed to work up from hardware to layer 7. On several attempts, I discovered I'd miswired CSN and there was no indication that was a problem. Right now I'm [...]

You mean simple troubleshooting steps like [these](http://tmrh20.github.io/RF24Ethernet/Troubleshooting.html) ? IMHO, once the physical hardware is connected properly, tested and working, there is little need to 'test' other individual layers with 'detailed diagnostic procedures' beyond simply configuring (set correct CE/CSN pins) and running the included examples . Again, no need to overcomplicate things.

> Pipes clearly don't match but I'm unable to locate those strings in the source. AA also appears to be mismatched.

Pipes look about right to me, don't worry too much about trying to figure out the addressing format at this point. AA should be matching, so something is wrong there, what the cause is, I'm not sure. Likely hardware/config/wiring.

> Adding a delay(50); to the Arduino gives the RPi time to switch the radio from send back to receive and lets most packets succeed. At least tells me hardware is working. Intermittently, even this fails and will not work until the RPi has been rebooted.

This is not needed, and seems to be a good indication that you have problems with your wiring or modules. Any chance these are not true 'nordic' nrf24l01 chips (clones) ? If you have problems with the gettingstarted examples, don't move to the higher layers, see [here](https://github.com/TMRh20/RF24/wiki/GettingStarted-sketch-failing-to-send:-how-to-diagnose%3F)

chisight, 2015-04-02:
Thank you very much for the reply.

> You mean like the instructions here ? Since you are using a beta version of an experimental library in development, it is simply assumed that you would have many of the required troubleshooting skills. Also, if you read the instructions the installer provides, it literally tells you exactly how to manage upgrading/reinstalling.
> Have you noticed the included examples and/or the extensive documentation at http://tmrh20.github.io ? I am completely baffled at what you are doing regarding grep and CE/CSN.

Yes, exactly like that link but for RF24Ethernet with mesh.  I followed that page, RF24 was working well with the Arduino as ping out and either Arduino or RPi as pong back.

Not upgrading/reinstalling, I saw the instructions for that.  The default pins are not matched to the ribon cable I made before finding your fork so I have to locate where the radio object is defined for the particular program that I'm trying to use, edit the pins, and make.  The build instructions are not in the installer, though it is possible to cut and paste the commands that are there.  Most are simple makes, but the RF24toTUN uses MESH=1 with the settings shown in the video.  (a key issue here.)
It would be nice if these libraries used a common header somewhere to declare the pins so that once the pins are set and then everything built, all programs would have them set rather than having to dig into this header or that c program to find the creation of the radio object and edit it.  The install.sh has no need to build things if it's just going to build with the wrong pins defined.  For me, I was unsure if the radio pins were defined at layer 1, once per layer, or at layer 7, though I've since identified that it's once per application.  Defining layer 1 things like pins at layer 7 just made no sense to me so it took me awhile to catch on.  I'm from the days when you used to load your stack layer by layer and I guess it stuck that the program containing layer 1 got the card hardware configuration so I incorrectly concluded that would be in the shared code in /usr/local/lib/ and built by install.sh

I have read a lot of the http://tmrh20.github.io/ site.  Pages like http://tmrh20.github.io/RF24Ethernet/classRF24Server.html#aed5308faf0652ddc150589956d35d7ed are clearly still being developed, but the examples helped to understand better for the most part and reading the library code filled in some gaps.

> You mean simple troubleshooting steps like these ? IMHO, once the physical hardware is connected properly, tested and working, there is little need to 'test' other individual layers with 'detailed [...]

I followed that page's instructions and even identified why ping out worked from the Arduino reliably and only intermittently from the RPi using that very simplistic low level test program.  I also tested Arduino to Arduino with very high reliability.

In my case, diagnosing the layers was exactly what was needed.  I followed the youtube video referenced and the instructions there were to enable mesh for RF24toTUN and then to use RF24Ethernet Interactive Server, which is a non mesh server.  Even without working through the layers, I was able to identify that the mesh calls were missing from the shown demo program and once added, and after correcting the AA mismatch, everything is now reliable.  I would still suggest using strncpy() rather than sprintf() to save 1.3K in that server as demos tend to get expanded into user's real programs.

The video should either show not building mesh in RF24toTUN, or show using SimpleServer_Mesh.ino.  I now think it likely that I miskeyed my answer to "Do you want to compile RF24toTUN with RF24Mesh support?" in the times it was working.

>> Adding a delay(50); to the Arduino gives the RPi time to switch the radio from send back to receive ...
> This is not needed, and seems to be a good indication that you have problems with your wiring or modules. Any chance these are not true 'nordic' nrf24l01 chips (clones) ? If you have problems with the gettingstarted examples, don't move to the higher layers, see here

That page is nice, and I had not found that one.  I checked the spec sheet, added a load resistor to simulate max load and noted no dip in power rather than playing with capacitors, I may add a small decoupling cap anyway.  I had also missed that Arduino SPI is running at SPI_CLOCK_DIV2 which is 8Mhz and fast for a breadboard.  I did experiment with lower RPi SPI speeds with no improvements.

Using the RPi with either radio, it works very reliably as pong back, but loses most packets on ping out.  Adding the delay to the Ardunio's pong back allows the RPi's ping out to work much better, though less than 100%, probably because the scheduler is not consistent in when it gives this program it's slice.

In the very simplistic GettingStarted.cpp/.ino, the RPi is just too slow.  It has to transmit the spi over system calls which I assume block and end the time slice, then it has to flip the radio back to receive on it's next time slice.  The Arduino has no threading so when it switches from receive to transmit, it does so very quickly.  I would guess that accessing SPI without system calls would also resolve this though I've not tried anything except the BCM2835 driver so I don't know.  I did play with the RPi scheduler and improve things but restored it to default once I realized that this wouldn't happen in less simplistic examples.

[... cut ...] andled in module with that demo.

Yes the modules are clones but are now very reliable.  I still don't know why the RPi was sometimes having trouble and requiring a reboot, it has not happened since I added the mesh calls to my custom version of InteractiveServer.ino.

I'd like to say thank you for all this wonderful work, yours seems to be the best of all the RF24 forks and your work on implementing all network layers plus mesh is impressive.  I look forward to when you add channel hopping too.

TMRh20, 2015-04-02:
> ...but the RF24toTUN uses MESH=1 with the settings shown in the video. (a key issue here.)

Technically that is a bit of a design flaw. I chose to make *building with RF24Mesh optional* and the MESH=1 option only includes RF24Mesh in the build. You still need to set the correct options to actually use RF24Mesh. This will likely be changed to include RF24Mesh by default, to prevent confusion.

I am also building out a new lib, [RF24Gateway](http://tmrh20.github.io/RF24Gateway/index.html) , to replace RF24toTUN, and it is nearing a usable state (only tested with RF24Mesh). 

> Defining layer 1 things like pins at layer 7 just made no sense to me so it took me awhile to catch on. I'm from the days when you used to load your stack layer by layer and I guess it stuck that [...]

The system is designed so users can load and access every layer, regardless of what layer they are working at. Keeping the pins defined at layer 1 only would mean users starting at higher layers need to dig through code and lower layer libs/docs to configure their system as you had assumed, and manually edit lower-layer config files, instead of being presented with the 2 main configuration options (CE,CSN pins) in every example. There are pros/cons to both methods, I would say.

> I followed the youtube video referenced and the instructions there were to enable mesh for RF24toTUN and then to use RF24Ethernet Interactive Server, which is a non mesh server. Even without working [...]

This is incorrect. RF24Ethernet default is to build with mesh, but leave it disabled and use TAP/ARP. This again tells me that you have something odd going on, and are modifying things that shouldn't need to be modified, instead of addressing the root cause.

> I would still suggest using strncpy() rather than sprintf() to save 1.3K in that server as demos tend to get expanded into user's real programs.

Yes, that sounds reasonable, but optimization is super low priority right now. That is part of what I would consider 'cleanup' tasks, which I usually leave until I've achieved the kind of functionality, reliability and stability I am looking for. 1.3K does not matter at all if the library is crashing/hanging. User contributions are always welcome. :p

> Using the RPi with either radio, it works very reliably as pong back, but loses most packets on ping out. Adding the delay to the Ardunio's pong back allows the RPi's ping out to work much better, [...]

This again seems to be an indication that you have other problems. The BCM2835 driver provides the fastest, most consistent and reliable interaction on RPi when compared to SPIDEV or MRAA in all of my testing.

> Even without working through the layers, I was able to identify that the mesh calls were missing from the shown demo program and once added, and after correcting the AA mismatch, everything is now reliable.

The demo video and default config do not use RF24Mesh at all. Something odd is happening I would guess. Every example is designed to work out of the box, using the default configuration.

> In the very simplistic GettingStarted.cpp/.ino, the RPi is just too slow. It has to transmit the spi over system calls which I assume block and end the time slice, then it has to flip the radio back to receive on it's next time slice. 

Again, this has been thouroughly tested on my RPis running raspbmc, raspbian, and osmc and it performs very consistently on all my platforms. I would again suggest hardware issues or some other odd problems.

> I'd like to say thank you for all this wonderful work, yours seems to be the best of all the RF24 forks and your work on implementing all network layers plus mesh is impressive.

Thanks. If I had more resources at my disposal, it would be easier to address some of the odd issues that come up when using cloned modules or systems with non-standard configurations etc.  

I am really thinking your issues are hardware related. I've been thinking of ways to address this, but in reality, the best would be to get ahold of some of these clones myself. Since development, design, testing etc is all supported and funded solely by me + contributors, with zero external funding, that may not happen.

eskildsf, 2015-04-02:
Sorry to hijack the issue :-).

I think you should make an eBay/Amazon wish list or something. I would be more than happy to buy some clones for you if it would help your development process!

TMRh20, 2015-04-02:
@eskildsf Haha, well technically this makes the development process more difficult, but makes it easier for users and support. Thanks for the good intentions though.

Everything above RF24 is a totally personal project, so keeping future updates closed-source and proprietary is probably in my personal best interest to be perfectly honest.  No need for any worries about documentation, examples, fake hardware, etc. etc., and development would move faster. 

This is part of what confuses me so much about expectations for things like 'detailed troubleshooting procedures' etc, because things like that will simply never happen if I am the only person working on this in my spare time.  Thats why I'm almost thinking up doing a kickstarter or something to generate funding, and let that determine the future of open vs closed source in this scenario. I don't want to sell anything, or provide free technical support services, and think it might also be a good indication of whether further open development projects are worth pursuing.

eskildsf, 2015-04-02:
To be honest I think this project has some of the best documentation around. There is only so many problems that documentation can solve. You can never eliminate all possible issues no matter how much you write.
I especially like the short videos you made that showed the set up procedure in action.

I hope that you keep it open source but I understand the burden of this ambitious project.

chisight, 2015-04-03:
the "determine the future of open vs closed source" unfortunately voids my ability to use this library.  Thank you for the support and i wish you luck going forward.

closed 2015-04-03 by chisight

TMRh20, 2015-04-03:
Lol, excellent. If it keeps the trolls away, maybe we can move forward.

chisight, 2015-04-03:
You should totally close source it.  Open source is not your thing.

TMRh20, 2015-04-03:
Simple facts: 

1. @chisight  joined Github March 31, 2015
2. @chisight only activity has been on ths repository.
3. @chisight Has made many complaints and suggestions, and has time to do the back and forth, but not any time to make any kind of contribution
4. @chisight  Talks as though he/she is somehow doing me a favor by using the library. 

PS. To anybody else enjoying the show, :beers:  (Its not likely to ever really be made closed source)

chisight, 2015-04-03:
yes i joined this site to interact with you.  that was the point.  done.

Chisight, 2015-04-03:
void sendPage(EthernetClient& _client, const char* _pointer, size_t size ){
  
  size_t bufSize = UIP_TCP_MSS;
  
  char buffer[bufSize+1]; // 91 bytes by default
  // Create a pointer for iterating through the array
  const char *i;
  // Increment the iterator (i) in increments of 45-3 (OUTPUT_BUFFER_SIZE-3) and send the data to the client
  for(i=_pointer; i<_pointer+(size-(bufSize));i+=bufSize){
    strncpy_P(buffer, i, bufSize);
    _client.write( buffer );
  }
  // For the last bit, send only the data that remains
  strncpy_P(buffer, i, ((_pointer+size)-i)+1);
  _client.write( buffer );
}

Chisight, 2015-04-03:
// Include some variables, print them into the page manually
{  
  char bf[OUTPUT_BUFFER_SIZE];
  
  if(!led_state){
    strncpy_P(bf, PSTR("<tr><td> </td><td bgcolor=lightpink>LED is OFF</td></tr>\n"),OUTPUT_BUFFER_SIZE-1);
  }else{
    strncpy_P(bf, PSTR("<tr><td> </td><td bgcolor=darkseagreen 1>LED is ON</td></tr>\n"),OUTPUT_BUFFER_SIZE-1);
  }
  _client.write(bf);
}