nRF24/RF24 #97: Building the library for Python Wrapper failed

gritschu, 2015-04-01:
Hello,
I just tried to build the library following these steps on a Raspberry Pi B+:
https://tmrh20.github.io/RF24/Python.html

I tried to build the library by calling:
./setup.py build 
inside the folder:
/RF24/RPi/pyRF24

The building failed and I got these errors:

running build
running build_ext
building 'RF24' extension
creating build
creating build/temp.linux-armv6l-2.7
gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c pyRF24.cpp -o build/temp.linux-armv6l-2.7/pyRF24.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for Ada/C/ObjC but not for C++ [enabled by default]
pyRF24.cpp: In function ‘void startFastWrite_wrap2(RF24&, std::string, bool, bool)’:
pyRF24.cpp:57:69: error: no matching function for call to ‘RF24::startFastWrite(const char*, std::basic_string<char>::size_type, const bool&, bool&)’
pyRF24.cpp:57:69: note: candidate is:
/usr/local/include/RF24/RF24.h:822:8: note: void RF24::startFastWrite(const void*, uint8_t, bool)
/usr/local/include/RF24/RF24.h:822:8: note:   candidate expects 3 arguments, 4 provided
pyRF24.cpp: In function ‘void init_module_RF24()’:
pyRF24.cpp:142:40: error: ‘RPI_BPLUS_GPIO_J8_03’ was not declared in this scope
pyRF24.cpp:143:40: error: ‘RPI_BPLUS_GPIO_J8_05’ was not declared in this scope
pyRF24.cpp:144:40: error: ‘RPI_BPLUS_GPIO_J8_07’ was not declared in this scope
pyRF24.cpp:145:40: error: ‘RPI_BPLUS_GPIO_J8_08’ was not declared in this scope
pyRF24.cpp:146:40: error: ‘RPI_BPLUS_GPIO_J8_10’ was not declared in this scope
pyRF24.cpp:147:40: error: ‘RPI_BPLUS_GPIO_J8_11’ was not declared in this scope
pyRF24.cpp:148:40: error: ‘RPI_BPLUS_GPIO_J8_12’ was not declared in this scope
pyRF24.cpp:149:40: error: ‘RPI_BPLUS_GPIO_J8_13’ was not declared in this scope
pyRF24.cpp:150:40: error: ‘RPI_BPLUS_GPIO_J8_15’ was not declared in this scope
pyRF24.cpp:151:40: error: ‘RPI_BPLUS_GPIO_J8_16’ was not declared in this scope
pyRF24.cpp:152:40: error: ‘RPI_BPLUS_GPIO_J8_18’ was not declared in this scope
pyRF24.cpp:153:40: error: ‘RPI_BPLUS_GPIO_J8_19’ was not declared in this scope
pyRF24.cpp:154:40: error: ‘RPI_BPLUS_GPIO_J8_21’ was not declared in this scope
pyRF24.cpp:155:40: error: ‘RPI_BPLUS_GPIO_J8_22’ was not declared in this scope
pyRF24.cpp:156:40: error: ‘RPI_BPLUS_GPIO_J8_23’ was not declared in this scope
pyRF24.cpp:157:40: error: ‘RPI_BPLUS_GPIO_J8_24’ was not declared in this scope
pyRF24.cpp:158:40: error: ‘RPI_BPLUS_GPIO_J8_26’ was not declared in this scope
pyRF24.cpp:159:40: error: ‘RPI_BPLUS_GPIO_J8_29’ was not declared in this scope
pyRF24.cpp:160:40: error: ‘RPI_BPLUS_GPIO_J8_31’ was not declared in this scope
pyRF24.cpp:161:40: error: ‘RPI_BPLUS_GPIO_J8_32’ was not declared in this scope
pyRF24.cpp:162:40: error: ‘RPI_BPLUS_GPIO_J8_33’ was not declared in this scope
pyRF24.cpp:163:40: error: ‘RPI_BPLUS_GPIO_J8_35’ was not declared in this scope
pyRF24.cpp:164:40: error: ‘RPI_BPLUS_GPIO_J8_36’ was not declared in this scope
pyRF24.cpp:165:40: error: ‘RPI_BPLUS_GPIO_J8_37’ was not declared in this scope
pyRF24.cpp:166:40: error: ‘RPI_BPLUS_GPIO_J8_38’ was not declared in this scope
pyRF24.cpp:167:40: error: ‘RPI_BPLUS_GPIO_J8_40’ was not declared in this scope
pyRF24.cpp:245:35: error: ‘closeReadingPipe’ is not a member of ‘RF24’
pyRF24.cpp:268:29: error: ‘rxFifoFull’ is not a member of ‘RF24’
pyRF24.cpp:284:82: error: no matches converting function ‘txStandBy’ to type ‘bool (class RF24::*)(uint32_t, bool) {aka bool (class RF24::*)(unsigned int, bool)}’
/usr/local/include/RF24/RF24.h:765:9: error: candidates are: bool RF24::txStandBy()
/usr/local/include/RF24/RF24.h:785:9: error:                 bool RF24::txStandBy(uint32_t)
pyRF24.cpp:293:45: error: ‘failureDetected’ is not a member of ‘RF24’
error: command 'gcc' failed with exit status 1

Did I miss something?
Thank you in advance.

amboleos, 2015-04-01:
Hello gritschu

I got same problem two days ago. I dont know why but this is because of lack of some drivers. I will tell the answer after I check my RPi.

gritschu, 2015-04-01:
Thank you in advance, amboleos.

TMRh20, 2015-04-01:
I'm guessing you may need to re-install the RF24 driver before trying to build the python wrapper.

See the manual install info [here](http://tmrh20.github.io/RF24/RPi.html)
Then try the same steps as before to build the python wrapper.

note: You likely want to delete the RF24 folder ( ``` sudo rm -r RF24 ``` ) before reinstalling and start with step 2 of the manual install.

amboleos, 2015-04-01:
I think you follow this http://tmrh20.github.io/RF24/Python.html
I find the absent defination in some files so copy that file into pyRF24.py
In my pyRF24 folder, I put some extra files which are found in RF24 folder (You need search them). 

bcm2835.c  
build       
pyRF24.cpp       
readme.md           
setup.py  
spi.h
bcm2835.h  
includes.h  
RF24_arch_config.h  
spi.cpp

Then I edit pyRF24.cpp file as

#include <RF24/RF24.h>
#include <boost/python.hpp>
#include "bcm2835.h"
#include "includes.h"

...

Then ./setup.py build

gritschu, 2015-04-01:
Thank you TMRh20 and amboleos for your help,
I followed TMRh20's suggestion to reinstall the driver. The build process worked now.
I had to fiddle about a little then, because the GettingStarted-example (.ino and .cpp) worked, but the pingpair_dyn did not. In the end it was the other setting of the SPI bus pins in the .ino file at the Ardiuno side, I had to adjust. Then the .cpp worked on the Pi – and the .pingpair_dyn.py as well.
Thanks a lot for your effort and the great library. ^_^
Yours.

WERAQS, 2015-04-06:
did the exact same thing but now getting;

root@RPi2B01:~/rf24libs/RF24/RPi/pyRF24# ./setup.py build  
running build
running build_ext
building 'RF24' extension
creating build
creating build/temp.linux-armv7l-2.7
gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c pyRF24.cpp -o build/temp.linux-armv7l-2.7/pyRF24.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for Ada/C/ObjC but not for C++ [enabled by default]
--freezes for a while on RPi2B, but totally stuck on RPiA+ (requires hard reset)--
^Cinterrupted

root@RPi2B01:~/rf24libs/RF24/RPi/pyRF24# sudo ./setup.py install  
running install
running build
running build_ext
building 'RF24' extension
gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c pyRF24.cpp -o build/temp.linux-armv7l-2.7/pyRF24.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for Ada/C/ObjC but not for C++ [enabled by default]
--freezes for a while on RPi2B, but totally stuck on RPiA+ (requires hard reset)--
^Cinterrupted

TMRh20, 2015-04-06:
@WERAQS It normally gets to that point an then takes a really long time to compile. Have you tried just waiting for like 10-15 mins?

WERAQS, 2015-04-06:
never did :) I'll try to wait, this time I'll hook up all the raspberry pi's.. because 10-15minutes x 9.. LONG time.

TMRh20, 2015-04-07:
Excellent. I usually do other things for about 10-15 mins, so wasn't really sure how long it actually took.

closed 2015-04-07 by TMRh20