nRF24/RF24 #98: no payload > 8 byte with 250KBPS

c128, 2015-04-06:
when set transfer to low speed  250kbps i can transfer only 8 byte with ackpayload

you can check this program with 12 byte 

/* TX
  1 - GND
   2 - VCC 3.3V !!! NOT 5V
   3 - CE to Arduino pin 9
   4 - CSN to Arduino pin 10
   5 - SCK to Arduino pin 13
   6 - MOSI to Arduino pin 11
   7 - MISO to Arduino pin 12
   8 - UNUSED
*/

#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"
#include "printf.h"

// The sizeof this struct should not exceed 32 bytes
struct RF24Data {
  byte dato1;
  byte dato2;
  byte dato3;
};

struct RF24AckPayload {
  float back1;
  float back2;
  float back3;
};

RF24Data nrf24Data;
RF24AckPayload nrf24AckPayload;

/*************  USER Configuration *****************************/

RF24 radio(9,10);     // (CE_PIN, CS_PIN);Set up nRF24L01 radio on SPI bus plus pins 7 & 8

/***************************************************************/

const uint64_t pipes[2] = { 0xABCDABCD71LL, 0x544d52687CLL };

void setup(void) {

  Serial.begin(9600);
  delay(1000);
  printf_begin();
  radio.begin();                           // Setup and configure rf radio
  radio.setChannel(1);                     // Set the channel
  radio.setPALevel(RF24_PA_LOW);           // Set PA LOW for this demonstration. We want the radio to be as lossy as possible for this example.
  radio.setDataRate(RF24_1MBPS);           // Raise the data rate to reduce transmission distance and increase lossiness
  radio.setAutoAck(1);                     // Ensure autoACK is enabled
  radio.enableAckPayload();
  radio.enableDynamicPayloads();
  radio.setRetries(2,15);                  // Optionally, increase the delay between retries. Want the number of auto-retries as high as possible (15)
  radio.setCRCLength(RF24_CRC_8);         // Set CRC length to 16-bit to ensure quality of data
  radio.openWritingPipe(pipes[0]);
  radio.openReadingPipe(1,pipes[1]);
  
  radio.printDetails();                   // Dump the configuration of the rf unit for debugging
 
  
  radio.stopListening();
}

void loop(void){

  nrf24Data.dato1 = 10;
  nrf24Data.dato2 = 13;
  nrf24Data.dato3 = 20;
  
 
     
    if(radio.isAckPayloadAvailable())
    {
        radio.read(&nrf24AckPayload, sizeof(nrf24AckPayload));
        Serial.print("//size of data   ");
        Serial.print(sizeof(nrf24AckPayload));
        Serial.print("///1' received ack payload is : ");
        Serial.print(nrf24AckPayload.back1);
        Serial.print("///2' received ack payload is : ");
        Serial.print(nrf24AckPayload.back2);
        Serial.print("///3' received ack payload is : ");
        Serial.println(nrf24AckPayload.back3);
      }
      
      radio.write(&nrf24Data, sizeof(RF24Data),0);

  
}

----------------------------*********************************************************************888

/// RX 

/* ****** RICEZIONE
  1 - GND
   2 - VCC 3.3V !!! NOT 5V
   3 - CE to Arduino pin 9
   4 - CSN to Arduino pin 10
   5 - SCK to Arduino pin 13
   6 - MOSI to Arduino pin 11
   7 - MISO to Arduino pin 12
   8 - UNUSED
*/

#include "Arduino.h"
#include <SPI.h>
#include "RF24.h"         //https://github.com/lnxbil/RF24 OR https://github.com/TMRh20/RF24
#include "printf.h"       

// The sizeof this struct should not exceed 32 bytes
struct RF24Data {
  byte dato1;
  byte dato2;
  byte dato3;
};

struct RF24AckPayload {
  float back1;
  float back2;
  float back3;
};

RF24Data nrf24Data;
RF24AckPayload nrf24AckPayload;

/*************  USER Configuration *****************************/

RF24 radio(9,10);      // (CE_PIN, CS_PIN);Set up nRF24L01 radio on SPI bus plus pins 7 & 8

/***************************************************************/

const uint64_t pipes[2] = { 0xABCDABCD71LL, 0x544d52687CLL };

void resetRF24Data() 
{
  nrf24Data.dato1 = 0;
  nrf24Data.dato2 = 0;
  nrf24Data.dato3 = 0;
}

void resetRF24AckPayload() 
{
  nrf24AckPayload.back1 = 0;
  nrf24AckPayload.back2 = 0;
  nrf24AckPayload.back3 = 0;
}

void setup(void) {
  
  resetRF24Data();
  resetRF24AckPayload();
  
  Serial.begin(9600);
  delay(1000); 
  printf_begin();
  radio.begin();                           // Setup and configure rf radio
  radio.setChannel(1);                     // Set the channel
  radio.setPALevel(RF24_PA_LOW);           // Set PA LOW for this demonstration. We want the radio to be as lossy as possible for this example.
  radio.setDataRate(RF24_1MBPS);           // Raise the data rate to reduce transmission distance and increase lossiness
  radio.setAutoAck(1);                     // Ensure autoACK is enabled
  radio.enableAckPayload();
  radio.enableDynamicPayloads();
  radio.setRetries(2,15);                  // Optionally, increase the delay between retries. Want the number of auto-retries as high as possible (15)
  radio.setCRCLength(RF24_CRC_8);         // Set CRC length to 16-bit to ensure quality of data
  radio.openWritingPipe(pipes[1]);        // Both radios listen on the same pipes by default, and switch when writing
  radio.openReadingPipe(1,pipes[0]);
  
  radio.printDetails();                   // Dump the configuration of the rf unit for debugging

  
  
  radio.startListening();

}

void loop(void){
  
  nrf24AckPayload.back1 = 101.35;
  nrf24AckPayload.back2 = 13.55;
  nrf24AckPayload.back3 = 21.233;
  
  while(radio.available())
  {
    radio.read(&nrf24Data, sizeof(nrf24Data));
    radio.writeAckPayload(1,&nrf24AckPayload,sizeof(nrf24AckPayload));
  Serial.print("//size of payload   ");
  Serial.print(sizeof(nrf24AckPayload));
  Serial.print("-- ricevuto   //dato1   ");
  Serial.print(nrf24Data.dato1);
  Serial.print("// dato2 ");    
  Serial.print(nrf24Data.dato2);
  Serial.print("// dato3 ");
  Serial.println(nrf24Data.dato3);
  }
    
}

TMRh20, 2015-04-06:
> There is a restriction on the length of ARD when using ACK packets with payload. The ARD time must
never be shorter than the sum of the startup time and the time on-air for the ACK packet:
> • For 2Mbps data rate and 5 byte address; 15 byte is maximum ACK packet payload length for
ARD=250µs (reset value).
> • For 1Mbps data rate and 5 byte address; 5 byte is maximum ACK packet payload length for
ARD=250µs (reset value).
> ARD=500µs is long enough for any ACK payload length in 1 or 2Mbps mode.
> • For 250kbps data rate and 5byte address the following values apply:

> 1500us ARD  ALL sizes
> 1250us ARD <= 24
> 1000us ARD <= 16
> 750us ARD <= 8
> 500us ARD <= EMPTY/No ACK payload
Table 18. Maximum ACK payload length for different retransmit delays at 250kbps

**tl;dr:**  ``` radio.setRetries(4,15);   (1000us ARD) ```

TMRh20, 2015-04-06:
Please leave this issue open as a reminder to add this info to the documentation.

c128, 2015-04-06:
radio.setRetries(10,15); is ok for me at 250kbps

only one question:
is correct read before write or is indifferent?

radio.read(&nrf24Data, sizeof(nrf24Data));
radio.writeAckPayload(1,&nrf24AckPayload,sizeof(nrf24AckPayload));

thank you

prophet-roshak, 2015-04-06:
AFAIK ACK payload is being sent with ack packet, along with acknolwedge info of recieveing a transmission. So, ACK payload you set *after* recieving a packet will be sent back on recieving *next* packet anyway. In this case "radio.read" before "radio.write" is indifferent. 
If you want to send data back with first recieved transmission, in my opinion, you should set ACK payload before startListening.

If data you are sending back, depends on recieved data, it will be sent only with next transaction, or you should manually switch device roles, and send packet back.

TMRh20, 2015-04-07:
@c128 " *Listen to this dude ~~Rufus~~ Roshak, he knows what he's talking about.* "

closed 2015-04-07 by TMRh20

op2op2op2, 2015-04-07:
c128, I haven't already check it as I'm involved in other Project, but if you want to have the "most updated data" in your ack payload, I think you can also flush TX buffer )otherwise you can only write 3 ackPayloads) and writeAckPayload in every loop of your sketch, when a packet is recieved, your RF24 will send the last ackPayload that you have written. But, as prophet-roshak says, YOU CAN'T READ a packet and afterwards WRITE an ackPayload and expect that this last ackPayload will be recieved by the transmitter node. 
Sorry for my english, I hope you`ve understand me.