nRF24/RF24 #100: nRF24 module Teensy3.1 coolness breakboard

epa64, 2015-04-10:
Hello,
I made test with a teensys 3.1 , RF24L01+ (DEV Branch) module and rfx-teensy3-x-nrf24l01-carrier-board-w-prototyping fromm embeddedcoolnes.
http://embeddedcoolness.com/shop/rfx-teensy3-x-nrf24l01-carrier-board-w-prototyping-area-kit/
The IRQ pin is mapped in pin 2. CE = 2 and CS=7

I try  RF24/examples/GettingStarted  and the sender role block just after [Serial.println(F("Now sending"));]
/********************************LOG****************************************/
Role_0 : *** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
Begin Loop ...
 Role_1 : Now sending
/*****************************************************************************/

[radio.printDetails() ]
teensy + breakBoard coolness CE = 2 and CS=7 IRQ mapped in pin 2
-----------------------------------------------------------------------------------
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x65646f4e32
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x02
RF_SETUP	 = 0x0f
CONFIG		 = 0x08
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 2MBPS
Model		 = nRF24L01+
CRC Length	 = 8 bits
PA Power	 = PA_MAX

 tennsy without breakboard CE = 9 and CS=10 No IRQ mapped 
-----------------------------------------------------------------------------
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x65646f4e32
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x03
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_LOW

They work's fine without the breakboard.

Best's regard
Eric

TMRh20, 2015-04-10:
IRQ and CE should not use the same pin. Rfx docs show pin 3 for CE.

http://embeddedcoolness.com/shop/rfx-teensy3-x-nrf24l01-carrier-board-w-prototyping-area-kit/

epa64, 2015-04-10:
excuse me CE = 3 and CS =7

epa64, 2015-04-10:
and they dont't work

epa64, 2015-04-10:
It was an error in my document.

epa64, 2015-04-10:
teensy + breakBoard coolness CE = 3 and CS=7 IRQ mapped in pin 2

STATUS = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1 = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR = 0x65646f4e32
RX_PW_P0-6 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA = 0x3f
EN_RXADDR = 0x03
RF_CH = 0x02
RF_SETUP = 0x0f
CONFIG = 0x08
DYNPD/FEATURE = 0x00 0x00
Data Rate = 2MBPS
Model = nRF24L01+
CRC Length = 8 bits
PA Power = PA_MAX

I try RF24/examples/GettingStarted and the sender role block just after [Serial.println(F("Now sending"));]
/*****************************LOG************************************/
Role_0 : ** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
Begin Loop ...
Role_1 : Now sending
/*****************************************************************************/

tennsy without breakboard CE = 9 and CS=10 No IRQ mapped

STATUS = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1 = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR = 0x65646f4e32
RX_PW_P0-6 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA = 0x3f
EN_RXADDR = 0x03
RF_CH = 0x4c
RF_SETUP = 0x03
CONFIG = 0x0e
DYNPD/FEATURE = 0x00 0x00
Data Rate = 1MBPS
Model = nRF24L01+
CRC Length = 16 bits
PA Power = PA_LOW

They work's fine without the breakboard.

TMRh20, 2015-04-11:
I noticed this in device details:

> Carrier board also has two 8 pin (4×2) headers for plugging in a nRF24L01+ module, one header providing connection to the hardware SPI pins D11-D13, and the other to an alternative set of pins, D4-D6, for a software SPI (“bit banging”) implementation if preferred. IRQ pin is set at D2, CE is D3 and CSN is D7.

Are you sure you are using the hardware SPI header?

You also mention you are using RF24L01+ (DEV Branch) : is that the [Updates](https://github.com/TMRh20/RF24/tree/Updates) branch you are using? It incorporates the new SPI transaction API, so you may have slightly different results between that and the master branch.

The only reason there should be any difference with using the shield is incorrect or faulty connections really.

epa64, 2015-04-11:
Thank's for the reply

The nrf24 is in spi hardware pin.

I work with master nrf24 branche and nrf24Network Dev branch. I use your nrf24Mesh librari.
I try with the new SPI lib for the test.

Best s regard
Eric

<div>-------- Message d'origine --------</div><div>De : TMRh20 <notifications@github.com> </div><div>Date :11/04/2015  06:38  (GMT+01:00) </div><div>A : TMRh20/RF24 <RF24@noreply.github.com> </div><div>Cc : Eric PAPET <e.papet@dev1-0.com> </div><div>Objet : Re: [RF24] nRF24 module Teensy3.1 coolness breakboard (#100) </div><div>
</div>I noticed this in device details:

Carrier board also has two 8 pin (4×2) headers for plugging in a nRF24L01+ module, one header providing connection to the hardware SPI pins D11-D13, and the other to an alternative set of pins, D4-D6, for a software SPI (“bit banging”) implementation if preferred. IRQ pin is set at D2, CE is D3 and CSN is D7.

Are you sure you are using the hardware SPI header?

You also mention you are using RF24L01+ (DEV Branch) : is that the Updates branch you are using? It incorporates the new SPI transaction API, so you may have slightly different results between that and the master branch.

The only reason there should be any difference with using the shield is incorrect or faulty connections really.

epa64, 2015-04-17:
hi,

I try next with debug and printdetails.
https://github.com/TMRh20/RF24/blob/master/examples/GettingStarted_HandlingData/GettingStarted_HandlingData.ino

I suprise the assembyl embeddedCoolnes board does'nt take the setup configuration and block when write.
>>Send to /dev/ttyACM0: "T"<<
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe7e7e7e7e7 0xc2c2c2c2c2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x02
RF_SETUP	 = 0x0f
CONFIG		 = 0x08
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 2MBPS
Model		 = nRF24L01+
CRC Length	 = 8 bits
PA Power	 = PA_MAX
write_register(00,08)
write_register(02,03)
Now sending
[Writing 8 bytes 24 blanks]

I just modify the sample and place this 
if (role == 1)  {
	radio.setDataRate(RF24_1MBPS);
	radio.setPALevel(RF24_PA_LOW);
	radio.setCRCLength(RF24_CRC_16);
	radio.setAutoAck(false);
radio.printDetails();
	//
    radio.stopListening();                                    // First, stop listening so we can talk.
    Serial.println(F("Now sending"));

..........
...........
}

and i obtain this result

>>Send to /dev/ttyACM0: "T"<<
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
write_register(06,07)
write_register(06,03)
write_register(00,0c)
write_register(01,00)
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe7e7e7e7e7 0xc2c2c2c2c2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x02
RF_CH		 = 0x02
RF_SETUP	 = 0x03
CONFIG		 = 0x0c
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_LOW
write_register(00,0c)
write_register(02,03)
Now sending
[Writing 8 bytes 24 blanks]

When i try with my breadboard assembly they wok fine .
The write method does'nt block.
>>Send to /dev/ttyACM1: "T"<<
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x65646f4e32
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x03
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_LOW
write_register(00,0e)
write_register(02,03)
Now sending
[Writing 8 bytes 24 blanks]
write_register(07,70)
write_register(00,0f)
write_register(07,70)
write_register(02,02)
Failed, response timed out.

thank's for your time.

Best regards

Eric

TMRh20, 2015-04-18:
Well, lets break this down to the basic change involved: the breakout board.

The breakout board just connects to existing pins on the Arduino and runs those connections to its own header pins.

From there, the nrf42l01 module is plugged into one of those headers.

The only thing that really changed is the physical connections, so the problem must be there. If you have a multimeter, a quick test can check for faulty connections. Also, how are you powering the device? Maybe try using the external plug instead of USB or vice-versa. This seems like more of a hardware issue than anything that I can really help with.

epa64, 2015-04-18:
Your are right,

My project (GPLV3), soon on github, consists of a MESH sensors network 
(tennsy 3.1 + nrf24) and a router of message on Olimex A20 with node.js 
nd Redis.

This project (Proof of concept)  is realized for the elderly persons 
that must stay at home.

  We received a financing by the region Aquitaine France through a 
promoting association free software. AQUINETIC 
<http://www.pole-aquinetic.fr/>

If you want  I can send you a teensy and one board embeddedcoolness  to 
make test.

At the moment the assemblies which work are on a breadboard.

For reason of packaging I look for  assemblies to support the teensy 3.1 
+ nrf24 + sensors + food(supply).

In every case thank you for this library and the time which dedicates 
you it.

PS: The tension of nrf24 is 3,24 V.(Measured in the voltmeter.)

Best regards

Eric

<http://www.pole-aquinetic.fr/>

TMRh20, 2015-04-19:
Ok, for sometihng like this, I can probably help. If you really want to send me hardware to troubleshoot with, email me at tmrh20@gmail.com and we can maybe arrange something.

closed 2015-06-17 by TMRh20