nRF24/RF24 #102: Mbed porting

Pidjey, 2015-04-16:
Hi, I not so good with prgraming and I would like porting it to mbed. I have seen the portability template http://tmrh20.github.io/RF24/Portability.html but I have no idea how to achive it. For example this is the spi template:

    1 
   15 #include <string>
   16 #include <stdint.h>
   17 #include <unistd.h>
   18 #include <stdio.h>
   19 #include <stdlib.h>
   20 #include <getopt.h>
   21 #include <fcntl.h>
   22 #include <sys/ioctl.h>
   23 #include <inttypes.h>
   24 #include <linux/types.h>
   25 #include <linux/spi/spidev.h> 
   26 
   27 using namespace std;
   28 //class SPI {
   29 public:
   30 
   34     SPI();
   35     
   39     void begin(int busNo);
   40     
   46     uint8_t transfer(uint8_t tx_);
   47     
   54     void transfernb(char* tbuf, char* rbuf, uint32_t len);
   55 
   61     void transfern(char* buf, uint32_t len);
   62     
   63     virtual ~SPI();
   64 
   65 private:
   66 
   68     string device;
   70     uint8_t mode;
   72     uint8_t bits;
   74     uint32_t speed;
   75     int fd;
   76 
   77     void init();    
   78 };
   79 
   80 

And this is the mbed spi library:

/* mbed Microcontroller Library
 * Copyright (c) 2006-2013 ARM Limited
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */
#ifndef MBED_SPI_H
#define MBED_SPI_H
 
#include "platform.h"
 
#if DEVICE_SPI
 
#include "spi_api.h"
 
namespace mbed {
 
/** A SPI Master, used for communicating with SPI slave devices
 *
 * The default format is set to 8-bits, mode 0, and a clock frequency of 1MHz
 *
 * Most SPI devices will also require Chip Select and Reset signals. These
 * can be controlled using <DigitalOut> pins
 *
 * Example:
 * @code
 * // Send a byte to a SPI slave, and record the response
 *
 * #include "mbed.h"
 *
 * SPI device(p5, p6, p7); // mosi, miso, sclk
 *
 * int main() {
 *     int response = device.write(0xFF);
 * }
 * @endcode
 */
00047 class SPI {
 
public:
 
    /** Create a SPI master connected to the specified pins
     *
     * Pin OPTIONS:
     *  (5, 6, 7) or (11, 12, 13)
     *
     *  mosi or miso can be specfied as NC if not used
     *
     *  @param mosi SPI Master Out, Slave In pin
     *  @param miso SPI Master In, Slave Out pin
     *  @param sclk SPI Clock pin
     */
    SPI(PinName mosi, PinName miso, PinName sclk, PinName _unused=NC);
 
    /** Configure the data transmission format
     *
     *  @param bits Number of bits per SPI frame (4 - 16)
     *  @param mode Clock polarity and phase mode (0 - 3)
     *
     * @code
     * mode | POL PHA
     * -----+--------
     *   0  |  0   0
     *   1  |  0   1
     *   2  |  1   0
     *   3  |  1   1
     * @endcode
     */
    void format(int bits, int mode = 0);
 
    /** Set the spi bus clock frequency
     *
     *  @param hz SCLK frequency in hz (default = 1MHz)
     */
    void frequency(int hz = 1000000);
 
    /** Write to the SPI Slave and return the response
     *
     *  @param value Data to be sent to the SPI slave
     *
     *  @returns
     *    Response from the SPI slave
    */
    virtual int write(int value);
 
public:
    virtual ~SPI() {
    }
 
protected:
    spi_t _spi;
 
    void aquire(void);
    static SPI *_owner;
    int _bits;
    int _mode;
    int _hz;
};
 
} // namespace mbed
 
#endif
 
#endif

What should I do with them

After this, how can should how should I say to the library that theese are my config files and my compiler should use them

TMRh20, 2015-04-16:
Well, this can be a bit involved, but you basically need to align the existing SPI functionality with the Arduino API.

This is generally done by creating a kind of wrapper for the existing SPI driver. As an example, in the spi.begin() function of the wrapper, you could do something like the following:

```
void spiclass::begin(){

    SPI spi(p5, p6, p7); // mosi, miso, sclk  
}
```

In this case, the [Arduino SPI API](http://www.arduino.cc/en/Reference/SPI) is used to call the mbed SPI driver, so the core RF24 library can use the same functions for both.

This would generally need to be done for SPI, GPIO, Timing functions, as well as anything relating to program memory. As I said it is a bit involved, and also requires some knowledge of both Arduino and Mbed platforms, so we would probably need some assistance etc.. in order to make this happen.

Pidjey, 2015-04-17:
Thank you for answering. I don't have the skills to do it, but i will try to get some people on it, but it will be hard. I feel that Lpc812 is quite more interesting for remote resnsors than attiny85.

http://www.nxp.com/documents/user_manual/UM10601.pdf

Pidjey, 2015-04-20:
Just one more question, how can I define that my compiler has to use my new template

TMRh20, 2015-04-20:
You can probably just use a #define for detection like ATTiny, and specify the RF24_arch_config file to load.

1. Detection is done in RF24_config.h, and generally only loads the arch/<DEVICE>/RF24_arch_config.h file  (except for Linux devices use includes.h)
2. The RF24_arch_config.h file loads specific files (spi.h, gpio.h etc) and defines things as required

For example, with Arduino, the detection is done solely in the RF24_config.h file.

``` #if defined ARDUINO ```

If the MBED platform has a similar #define, you can use that, or if there is a microcontroller specific #define you can use that, else something would have to be manually defined somewhere.

```
#elif defined(__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__) || defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
  
  #define RF24_TINY
  #include "arch/ATTiny/RF24_arch_config.h"
```

Pidjey, 2015-04-21:
>    arch//RF24_arch_config.h file (except for Linux devices use includes.h)

Well I have tried in everyway porting it but I couldn't get it, I don't
have the skill, so I decided to use attiny85. What was my surprese? Attiny
is totally useless, I just wanted to attach just an ds18b20 temperature
sensor, but the RT24Mesh firmware is 81%, adding OneWire library exceeds
the maximum of 8Kb in 2Kb making it totally useless.

I haven't found anyone for porting to mbed, so want to ask you gently for
it. Main advantages of Mbed? Scalable hardware, plenty of different models
with different capabilities of MCUs, form ARM cortex M0+ to M4. And a nice
feature, Mbed official RTOS. It could be quite useful for developing an
interest stack.

I'm planing in using it in LPC812 for sensors, so take a look and if you
have ebough time, and you want.... give it a try because it could be a nice
improvement.

Avamander, 2015-10-20:
@Pidjey If you are still dealing with this you could possibly rewrite the ds18b20 library as bare minimum as possible.

closed 2015-12-14 by Avamander