nRF24/RF24 #106: Checking available pipes with python

amboleos, 2015-04-23:
Hello, I am not sure that my question is suitable for "Issues", if not, sorry.
I only need to find out which pipe is available. It would be better if you add usage example for python in http://tmrh20.github.io/RF24/classRF24.html
   
 for pipe in [0x00,0x01,0x02]:
        if radio.available(pipe):
            while radio.available(pipe):
                l = radio.getDynamicPayloadSize()
                receive_payload = radio.read(l)
                time.sleep(0.001)

File "gatebay.py", line 186, in radioCheck
    if radio.available(pipe):
ArgumentError: Python argument types in
    RF24.available(RF24, int)
did not match C++ signature:
    available(RF24 {lvalue})

mz-fuzzy, 2015-04-23:
yes, some doc would help on this. It's so:
for your case you can use use **available_pipe()** method. It returns tuple (result, pipe_nr).

comment:
It's not (easily) possible to return a value in method's parameter in Python and it's also not feasible to overload original **available()** method. Hence there is a new **available_pipe()** method which returns tuple - boolean result and pipe id. 

Call it this way:
```
is_available, pipe = radio.available_pipe()
```

amboleos, 2015-04-23:
Thank you very much mz-fuzzy
Is there any place where I can find all python functions for RF24 class?

mz-fuzzy, 2015-04-23:
at the moment you have to check [source code](https://github.com/TMRh20/RF24/blob/master/RPi/pyRF24/pyRF24.cpp), scroll to the bottom.

mz-fuzzy, 2015-04-23:
Also... looking closer to the example, **pipe** in only **output** parameter, so you have to redesign your code. Pipe is not an input param, only output. Even in the C++ interface's as I looked into RF24.h

closed 2015-09-24 by TMRh20