nRF24/RF24 #108: Same Code Doesn't Work with TMRH20 RF24 Library but Does with Stanleyseow

Ge0Dave, 2015-05-11:
This is the C++ code I am using on the Pi:

#include <cstdlib>
#include <iostream>
#include <RF24/RF24.h>

using namespace std;

// Setup for GPIO 22 CE and CE1 CSN with SPI Speed @ 8Mhz
RF24 radio(RPI_V2_GPIO_P1_15,BCM2835_SPI_CS0, BCM2835_SPI_SPEED_8MHZ); 

void setup(void)
{
    // init radio for reading
    radio.begin();
    //radio.enableDynamicPayloads();
    radio.setPayloadSize(32);
    radio.setAutoAck(1);
    radio.setRetries(15,15);
    radio.setDataRate(RF24_1MBPS);
    radio.setPALevel(RF24_PA_MAX);
    radio.setChannel(120);
    radio.setCRCLength(RF24_CRC_16);
    radio.openReadingPipe(0,0xDEADBEEF00LL);
    radio.startListening();
	printDetails(); 
}
 
void loop(void)
{
    // 32 byte character array is max payload
    char receivePayload[32];
 
    while (radio.available())
    {
        // read from radio until payload size is reached
        uint8_t len = radio.getDynamicPayloadSize();
        radio.read(receivePayload, len);
 
        // display payload
        cout << receivePayload << endl;
    }
}
 
int main(int argc, char** argv) 
{
    setup();
    while(1)
        loop();
 
    return 0;
}

When I build your library it doesn't work. When I build the Stanleyseow library my code works perfectly. I want to use the python wrapper so I would prefer to use your library. I am transmitting to the Pi via an MSP430G2553 using the Spirilis nrf24L01+ library for the MSP430 Launchpad...see http://forum.43oh.com/topic/4707-nrf24l01-raspberry-pi-and-msp430g2553/.

Any ideas?

TMRh20, 2015-05-12:
SSeow fork is now an older version of my RF24 fork I believe.

What is the actual problem/symptoms?

#include
#include

http://forum.43oh.com/topic/4707-nrf24l01-raspberry-pi-and-msp430g2553/.

Ge0Dave, 2015-05-12:
Thanks for the reply. I am not sure where to start. 

I build the SSeow (https://github.com/stanleyseow/RF24/tree/master/RPi/RF24) library using sudo make install and I am able to successfully run the code in my previous message and receive the payload from the transmitter. 

I build your library (https://github.com/TMRh20/RF24) using sudo make install and it will print the details but nothing is ever received.

Here is the details of the SSeow:
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xdeadbeef00 0xc2c2c2c2c2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x20 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x78
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
296/0
296/0
296/0
296/0

Your printDetails():
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xdeadbeef00 0xc2c2c2c2c2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x20 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x78
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX 

I know that isn't much to go on...is there anything else I could do to troubleshoot?

TMRh20, 2015-05-12:
Well, you could try enablig debugging by editing the config.h file. I would be interested in how the examples work, but not sure if you would have similar examples for the MSP430. One thing to try may be to try changing settings. Could you post the MSP430 code also?

DShaeffer, 2015-05-12:
I haven't looked at the examples closley...perhaps I will try that.

Here is the MSP430 code:

```
#include <msp430.h>
#include "msprf24.h"
#include "nrf_userconfig.h"
#include "stdint.h"
#include "stdio.h" //for sprintf
#include "math.h"

volatile unsigned int user;

unsigned int sdelay = 0; //each 2 delay = 25 secs

/*  Function Definitions  */
void ADC_init(void);

void main(){
	char addr[5];
	char txBuf[32];

	WDTCTL = WDTHOLD | WDTPW;
	DCOCTL = CALDCO_1MHZ;
	BCSCTL1 = CALBC1_1MHZ;
	BCSCTL2 = DIVS_2;  // SMCLK = DCOCLK/2
	// SPI (USCI) uses SMCLK, prefer SMCLK < 10MHz (SPI speed limit for nRF24 = 10MHz)

	ADC_init();

	BCSCTL3 |= LFXT1S_2;

	//Timer A
	TACCR0 = 12000;// Source Timer A from ACLK (TASSEL_1), up mode (MC_1). // Up mode counts up to TACCR0. SLAU144E p.12-20
	TACTL = TASSEL_1 + ID_3 + MC_1;    // ACLCK, 1/8 DIVIDER, upmode to TCCR0 value
	TACCTL0 = CCIE;// CCIE = Interrupt when timer counts to TACCR1
	TACCR1 = 0;// starting value for capture/control register

	//setup pins
	P1DIR |= BIT0;// Red used for status

	user = 0xFE;

	/* Initial values for nRF24L01+ library config variables */
	rf_crc = RF24_EN_CRC | RF24_CRCO; // CRC enabled, 16-bit
	rf_addr_width      = 5;
	rf_speed_power     = RF24_SPEED_1MBPS | RF24_POWER_MAX;
	rf_channel         = 120;

	msprf24_init();  // All RX pipes closed by default
	msprf24_set_pipe_packetsize(0, 32);
	msprf24_open_pipe(0, 1);  // Open pipe#0 with Enhanced ShockBurst enabled for receiving Auto-ACKs
	msprf24_set_retransmit_count(15);

	// Transmit to 'rad01' (0x72 0x61 0x64 0x30 0x31) F0F0F0F0E4
	msprf24_standby();
	user = msprf24_current_state();
	addr[0] = 0xDE;	addr[1] = 0xAD;	addr[2] = 0xBE;	addr[3] = 0xEF;	addr[4] = 0x00;
	w_tx_addr(addr);
	w_rx_addr(0, addr);  // Pipe 0 receives auto-ack's, autoacks are sent back to the TX addr so the PTX node
			             // needs to listen to the TX addr on pipe#0 to receive them.

	 while(1) {
	                // Handle RF acknowledgements
	                if (rf_irq & RF24_IRQ_FLAGGED) {
	                        msprf24_get_irq_reason();

	                        // Handle TX acknowledgements
	                        if (rf_irq & RF24_IRQ_TX) {
	                        	//blink the receive the light to indicate successful transaction
									P1OUT |= BIT0;  // Red LED on
									__delay_cycles(95000);
									P1OUT &= ~BIT0; // Red LED off
									__delay_cycles(95000);
									P1OUT |= BIT0;  // Red LED on
									__delay_cycles(95000);
									P1OUT &= ~BIT0; // Red LED off
	                                // Acknowledge
	                                msprf24_irq_clear(RF24_IRQ_TX);
	                                msprf24_powerdown();

	                        }

	                        if (rf_irq & RF24_IRQ_TXFAILED) {
	                                msprf24_irq_clear(RF24_IRQ_TXFAILED);
	                                flush_tx();
	                                msprf24_powerdown();

	                        }

	                        if (rf_irq & RF24_IRQ_RX) {
	                                // Really no reason we should be receiving anything.
	                                flush_rx();
	                                msprf24_irq_clear(RF24_IRQ_RX);
	                        } /* rf_irq & RF24_IRQ_RX */
	                }  /* rf_irq & RF24_IRQ_FLAGGED */

	                if(sdelay == 1){
	                			msprf24_standby();
	                			flush_tx();

	                			ADC10CTL0 |= ADC10SC;   // start a new conversion
	                			while ((ADC10CTL1 & ADC10BUSY) == 0x01);   // wait for conversion to end

	                			//sprintf(txBuf, "%d", ADC10MEM); //pass temp to tx buffer

	                			char end1 = '/';
	                			int end2 = 0;

	                			sprintf(txBuf, "%d%c%d", ADC10MEM, end1, end2); //pass temp to tx buffer

	                			//send txBuf to rx module
	                			w_tx_payload(32, txBuf);
	                			msprf24_activate_tx();

	                			sdelay=0;
	                			_BIS_SR(LPM3_bits + GIE);                 // Enter LPM3 w/ interrupt55
	                		}
	        }

}

void ADC_init(void) {
            // Use Vcc/Vss for Up/Low Refs, 16 x ADC10CLKs, turn on ADC
    ADC10CTL0 = SREF_0 + ADC10SHT_2 + ADC10ON;
            // A1 input, use ADC10CLK div 1, single channel mode
    ADC10CTL1 =  INCH_4 + SHS_0 + ADC10SSEL_0 + ADC10DIV_0 + CONSEQ_0;
    ADC10AE0 = BIT4;      // Enable ADC input on P1.4

    ADC10CTL0 |= ENC;     // Enable conversions.
} // ADC_init

// Timer A0 interrupt service routine
#pragma vector=TIMER0_A0_VECTOR
__interrupt void Timer_A (void)
{
	sdelay++;
}
```

Avamander, 2015-10-20:
Any status update on this?

closed 2015-10-23 by Avamander