nRF24/RF24 #109: Windows 10 .NET port?

Christilut, 2015-05-15:
It would be really great to see this library ported to .NET so we can run nRF's from Raspberry Pi 2's with Windows 10.

I'm not entirely sure what the current state of win10 on the pi is but there are [SPI functions](https://msdn.microsoft.com/en-us/library/windows.devices.spi.aspx) so hopefully it's a relatively simple port.

TMRh20, 2015-05-16:
I know windows is very common, can be easy to use, etc etc, but I'm hesitant to support the use of open-source software with their proprietary products.

I know it will probably come eventually, but if I can encourage people to use Linux and open-source simply by not supporting Windows, then maybe its a win-win situation for everybody in the end. (except MS)

Christilut, 2015-05-16:
Too bad you think of Microsoft that way (but understandable). I think they are on the way to creating some really great stuff with Windows 10. Don't forget that [.NET Core](http://blogs.msdn.com/b/dotnet/archive/2014/11/12/net-core-is-open-source.aspx) is open source now. So they are going the right way :)

Anyway I might try porting it in a few months, when Windows 10 is a bit more stable.

TMRh20, 2015-05-18:
> I think they are on the way to creating some really great stuff with Windows 10.

I know, that is part of the problem lol. 

If you do try porting, I'll definitely try to help out, I just have a different philosophy from MS, and am usually not an early adopter of any new system.

nkolban, 2015-06-18:
I too am interested in making this library available for Win 10 so if a project or sub project starts to port, please count me in.  kolban1@kolban.com

Avamander, 2015-10-20:
The issue can be reopened if someone decides to start porting.

closed 2015-10-20 by Avamander

Avamander, 2015-11-11:
Please open a new issue if you wish to begin porting the library.