nRF24/RF24 #110: ACK fails

OOTTEE, 2015-05-16:
Hello, 
I'm trying to connect a raspberry and Arduino, When I try to make a delivery from raspberry to Arduino Arduino packets arrive at, but do not receive the ACK in the raspberry.
otherwise, when sending packages from the Arduino to the raspberry everything works correctly.

* Raspberry code:

```c++

#include <cstdlib>
#include <iostream>
#include <sstream>
#include <string>
#include "./RF24.h"
#include <wiringPi.h>
#include <errno.h>

#define INTERRUPT 4

using namespace std; // CE Pin, CSN Pin, SPI Speed

// Setup for GPIO 22 CE and CE0 CSN with SPI Speed @ 8Mhz
RF24 radio(RPI_V2_GPIO_P1_15, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ);

// Radio pipe addresses for the 2 nodes to communicate.
const uint64_t pipes[2] = { 0xFFFFFFFFFFLL, 0x0000000002LL };

char receive_payload[33],output[32]; // +1 to allow room for a terminating NULL char

int count = 0;

bool tx,rx,fail;

float h,t,f;

void interrupt(){
    radio.whatHappened( tx,fail,rx);
    if(rx){
        printf("recibido: ");
        count ++;
        if(radio.available()){
            while(radio.available()){
                radio.read( receive_payload, 32 );
                memcpy(&h, (receive_payload)+10, 4);
                memcpy(&t, (receive_payload)+14, 4);
                memcpy(&f, (receive_payload)+18, 4);
                printf("H:%f C:%f F:%f\n", h,t,f);
            }
        }       
    }
    if(tx){
        printf("enviado\n");
        radio.startListening();
    }
    if(fail){
        printf("fallo\n");
        radio.startListening();
    }
}

int main(int argc, char** argv){

  radio.begin();
  radio.setRetries(5,15);
  radio.setPALevel(RF24_PA_HIGH);
  radio.setAutoAck(1);
  radio.setChannel(50);
  radio.setDataRate(RF24_1MBPS);

  radio.openWritingPipe(pipes[1]);
  radio.openReadingPipe(1,pipes[0]);

  radio.printDetails();

  radio.stopListening();

  if (wiringPiSetup () < 0) {
      fprintf (stderr, "Unable to setup wiringPi: %s\n", strerror (errno));
      return 1;
  }

  // set Pin 17/0 generate an interrupt on high-to-low transitions
  // and attach myInterrupt() to the interrupt
  if ( wiringPiISR (INTERRUPT, INT_EDGE_FALLING, &interrupt) < 0 ) {
      fprintf (stderr, "Unable to setup ISR: %s\n", strerror (errno));
      return 1;
  }

  radio.startListening();
  output[0] = 'G';
string txt; 
  while(1){
    cin>>txt;
    radio.stopListening();
    radio.startWrite( &output , 32, 0);
  }

}
```

* Arduino Code:

```c++
#include <Timer.h>
#include <Event.h>
#include <SPI.h>
#include <EEPROM.h>
#include "nRF24L01.h"
#include "RF24.h"
#include <TimerOne.h>
#include "DHT.h"
#include "printf.h"

#define DHTPIN 4
#define DHTTYPE DHT11   // DHT 11 

DHT dht(DHTPIN, DHTTYPE);

//MEMORY POSITIONS
#define CHANNEL_POSITION 10
#define POWER_POSITION   12

uint64_t address_server  = 0xFFFFFFFFFFLL;
uint64_t address_cliente = 0x0000000002LL;

uint8_t output[32];
String inputString = ""; 
String serialAction = "";
boolean stringComplete = false, readValue=false;

//ID identificador de tipo de dispositivo 
#define TYPE "0003" //INTERRUPTOR

// POSITION VALUES PAYLOAD PARAMS [ACTION]
#define ACTION_B 0

// ACTION VALUES
#define GET  'G'

#define OK 'O'
#define FAIL 'F'

// PIN RELES
#define LED 9
#define BUTTON 3

// Hardware configuration
RF24 radio(7,8);  
bool tx,fail,rx;
// PAQUETE RECIVIDO
uint8_t receive_payload[33];
uint8_t action;

/********************** Setup *********************/
void setup(){
  pinMode(BUTTON,INPUT_PULLUP);

  pinMode(LED, OUTPUT);
  digitalWrite(LED, LOW);

  delay(500);

  Serial.begin(115200);
  inputString.reserve(200);

  radio.begin(); 
  printf_begin();
  radio.setPALevel(RF24_HIGH);
  radio.setDataRate(RF24_1MBPS);
  radio.setAutoAck(true);
  radio.setRetries(5,15);
  radio.setChannel(50);           

  readAddr(&address_cliente);                                             
  radio.openWritingPipe(address_server);
  radio.openReadingPipe(1,address_cliente);

  radio.printDetails();

  radio.whatHappened(tx,fail,rx);   
  radio.startListening();

  delay(50);
  attachInterrupt(0, check_radio, LOW);

  delay(50);

  dht.begin();
  readTemp();

  Serial.println("INICIADO");
}

/********************** Main Loop *********************/
int boton = HIGH;
int pulsacion = 0;

void loop() {   
//Send packege
  boton = digitalRead(BUTTON);
  if( boton == LOW && pulsacion == 0){
      output[9] = GET;
      radio.stopListening();
      radio.startWrite( &output , 32, 0);
      pulsacion = 1;
  }
  if( boton == HIGH && pulsacion == 1){
      pulsacion = 0;
      digitalWrite(LED,!digitalRead(LED));
      delay(100);
  }
}

float h,t,f;
void readTemp(){
  int i;
  h = dht.readHumidity(); 
  t = dht.readTemperature(); //Celsius
  f = dht.readTemperature(true); //Fahrenheit
  for(i=0;i<4;i++){
     output[10+i]= ((uint8_t*)(&h))[i];
     output[14+i]= ((uint8_t*)(&t))[i];
     output[18+i]= ((uint8_t*)(&f))[i];
  }
}

/********************** Interrupt RADIO *********************/
void check_radio(void)                               
{
  radio.whatHappened(tx,fail,rx);                    

  if ( tx ) { //transmisin completada
      radio.startListening();
      printf("Ack Payload:Send\n\r");
  }

  if ( fail ) {
      delay(100);
      radio.stopListening();
      radio.startWrite( &output , 32, 0);
      printf("Ack Payload:Failed\n\r");
  }

  if ( rx ){
      radio.read( receive_payload, 32 );
      action = receive_payload[ACTION_B];
      if(action == GET){
        output[9] = GET;
      }else{ 
        output[9] = FAIL;
      } 
  }
}
```

Thank you very much, took a long time with this problem and I'm going crazy.

prophet-roshak, 2015-05-16:
There may be a timing issue, when RPi end is not enabled listening yet, while ardiuno is already sending ack. Have you tried adding a delay before sending ack from arduino? These things are really very fast sometimes :) 

Also there can be an issue with RPi always entering send mode, as main loops wants. Try not to send another packet, while you haven't recieved an ack or timeout passed...

OOTTEE, 2015-05-16:
thank you very much, but I do not know how to delay the  ack.

TMRh20, 2015-05-17:
@OOTTEE Can you also post the output from the RPi and Arduino so it is clearer what is happening?

OOTTEE, 2015-05-17:
@TMRh20 The output when sending a packet from the RPI to Arduino:

* RPI:

```shell
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x0000000002 0xffffffffff
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x0000000002
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x32
RF_SETUP	 = 0x05
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_HIGH
f     /* I'M INSERT A CHARACTER FOR SEND A PACKAGE */
fallo  /* INTERRUPT FAIL FLAG (whathappend(tx,fail,rx)) */

```
* arduino:
```shell
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xffffffffff 0x0000000002
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xffffffffff
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x32
RF_SETUP	 = 0x07
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
INICIADO
Recibido: GET /* PACKAGE RECEIVED */
```

Thank you very much.

TMRh20, 2015-05-17:
Try maybe:
``` const uint64_t pipes[2] = { 0xFFFFFFFFFFLL, 0xCCCCCCCCC2LL }; ```

OOTTEE, 2015-05-17:
XD I do not understand, but it works ¡¡¡

Thank you very much, I never would have imagined what

TMRh20, 2015-05-17:
See GCopelands [blog post on RF24 addressing](http://maniacalbits.blogspot.com/2013/04/rf24-addressing-nrf24l01-radios-require.html)

Also:

> Note: Addresses where the level shifts only one time (that is, 000FFFFFFF) can often be detected in noise and can give a false detection, which may give a raised Packet Error Rate. Addresses
as a continuation of the preamble (hi-low toggling) also raises the Packet Error Rate.

http://www.nordicsemi.com/eng/content/download/2726/34069/file/nRF24L01P_Product_Specification_1_0.pdf

OOTTEE, 2015-05-17:
Thank you very much, shut the issue.

closed 2015-05-17 by OOTTEE