nRF24/RF24 #111: Problem building on Arduino 1.6.x

brunnels, 2015-05-17:
The PROGMEM workaround is no longer needed in Arduino 1.6.x.  If you comment out the following in RF24_config.h everything builds.

```
// Avoid spurious warnings
// Arduino DUE is arm and uses traditional PROGMEM constructs
#if 1
#if ! defined( NATIVE ) && defined( ARDUINO ) && ! defined(__arm__)
#undef PROGMEM
#define PROGMEM __attribute__(( section(".progmem.data") ))
#undef PSTR
#define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
#endif
#endif
```

TMRh20, 2015-05-18:
Is there a problem building, or is it just not needed? A quick test with 1.0.6, 1.6.1 and 1.6.4 seems to show that it builds OK in all, with or without that code. It does look like it could be removed completely.

brunnels, 2015-05-18:
I don't think it is.  It was originally a workaround for a compiler issue.  I was getting an error with #include <avr/pgmspace.h> in my main sketch or in included libraries and removing the workaround resolved it.

TMRh20, 2015-06-17:
This should be fixed now, with sequentially received, duplicate fragmented payloads being dropped.

closed 2015-09-24 by TMRh20