nRF24/RF24 #112: Acks only being sent on pipes 0 and 1 from RPi

greeebs, 2015-05-22:
I have a number of Arduino's with NRF24L01+'s sending data to a RPi "hub" with an NRF24L01+ on it.
The RPi has 3 pipes open for listening and each of the Arduino's is configured with a different TX address corresponding to the addresses on the 3 RPi pipes.

Out of the three Arduino's, only the ones whose addresses correspond to the addresses on pipes 0 and 1 receive an ACK packet for anything they send. Packets on the other pipes are received correctly, but no Ack is sent (well, I can't be sure of that, but certainly no Ack is received!)

If I change the order of which addresses are listening on which pipe, without updating or even resetting any of the Arduino's, the Arduino that fails to get ack packets is always the one sending to the listener on pipe 2.

Here's the Config on the RPi:
```
================ SPI Configuration ================
CSN Pin          = CE0 (PI Hardware Driven)
CE Pin           = Custom GPIO25
Clock Speed      = 8 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0x7859b8dc54 0x7859b8dc58
RX_ADDR_P2-5     = 0x56 0xc4 0xc5 0xc6
TX_ADDR          = 0x50756d7031
RX_PW_P0-6       = 0x20 0x20 0x20 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x07
RF_CH            = 0x41
RF_SETUP         = 0x27
CONFIG           = 0x0f
DYNPD/FEATURE    = 0x03 0x06
Data Rate        = 250KBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX
```
My Arduino's are each sending a single packet at varying rates from every 2 seconds to every 10 seconds and with the config above, the ones with TX addresses of 0x7859b8dc54 and 0x7859b8dc58 receive ACK packets while the one sending with 0x7859b8dc56 does not.
If I switch address 0x7859b8dc58 onto pipe 2 and switch 0x7859b8dc56 onto pipe 1, the arduino using 0x7859b8dc58 stops getting ack's and the arduino using 0x7859b8dc56 starts getting them.

TMRh20, 2015-05-22:
http://en.m.wikipedia.org/wiki/Wikipedia:Reference_desk/How_to_ask_a_software_question

Need more info.

greeebs, 2015-05-22:
Hmm, I thought I'd covered everything in that link, I didn't think to post code because its as simple as it gets... here's the code, I'm hoping that's the "more info" you need?

**All 4 devices do this at the start:**
```c++
const uint64_t addresses[] = { 0x7859B8DC56LL, 0x7859B8DC58LL, 0x7859B8DC54LL };

radio.begin();
radio.setAutoAck(true);
radio.enableAckPayload();
radio.setChannel(65);
radio.setCRCLength(RF24_CRC_16);
radio.setDataRate(RF24_250KBPS);
radio.setPayloadSize(32);
```

**The Pi does this:**
```c++
RF24 radio(RPI_V2_GPIO_P1_22, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ);

radio.openReadingPipe(0, addresses[0]); // This one will ACK
radio.openReadingPipe(1, addresses[1]); // So will this one
radio.openReadingPipe(2, addresses[2]); // This one wont.
radio.startListening();
while (1)
	if (radio.available(&p)) {
		radio.read(buf, 32);
		printf("%s\n", buf);
	}
}
```

**Each of the Arduino's do this:**
```c++
#define RF24_PACKETGAP 10 // or 12 or 15 depending on the Arduino. 
uint8_t me = 0; // This is either 0, 1 or 2 depending on the Arduino.
uint8_t rf24errCnt = 0;
char xmitBuf[32];
RF24 radio(9, 10);
uint16_t pktCnt = 0;

setup() {
  radio.openWritingPipe(addresses[me]);
  radio.openReadingPipe(0, addresses[me]);
}

loop() {
  if ((millis() - lastSendTime) > RF24_PACKETGAP) {
	pktCnt++;
	sprintf(xmitBuf, "%d - %d:%d", me, pktCnt, rf24errCnt);
	// Send the RF24 packet.
	radio.powerUp();
	if (!radio.write(xmitBuf, payloadSize)) {
		rf24errCnt++;
	}
	radio.txStandBy();
	radio.powerDown();
	lastSendTime = millis();
  }
}
```
The Pi sees:
```
0 - 1:0
1 - 1:0
2 - 1:0
0 - 2:0
1 - 2:0
2 - 2:1
0 - 3:0
1 - 3:0
0 - 4:0
2 - 3:2
```
Clearly the Pi is getting all packets, but the Arduino sending to pipe 2 is not getting any ACKs.

If I change the PI to this:
```c++
radio.openReadingPipe(0, addresses[2]); // This one will ACK
radio.openReadingPipe(1, addresses[1]); // So will this one
radio.openReadingPipe(2, addresses[0]); // This one wont.
```
The Arduino with addresses[0] stops getting ACKs and the Arduino with addresses[2] starts getting them and this is what the Pi sees:
The Pi sees:
```
0 - 1:0
1 - 1:0
2 - 1:0
0 - 2:1
1 - 2:0
2 - 2:0
0 - 3:2
1 - 3:0
0 - 4:3
2 - 3:0
```
Once again, the Pi is getting all packets but the Arduino sending to pipe 2 is not getting any ACKs.

TMRh20, 2015-05-23:
Well it might seem a bit anal, and the code may seem trivial from a user perspective, but that is where 99% of technical problems are, (user based) and having all of the information removes a lot of assumptions that are required otherwise. 

In this case, I am guessing the problem is here:
```radio.openReadingPipe(0, addresses[me]);```

You have the RPi and Arduinos listening on the same address = :frowning: 

a: Each device must use a unique set of reading addresses for pipes 0-5
b: Pipes 2-5 will borrow the address from pipe1 except for 1 byte (as per your addressing scheme)

I generally will always open at least one uniqure reading pipe on every device, so in this case you would have:

```
RPi: 3 unique addresses (1 for each device that will write to it)
Arduinos: 1 unique address per device (1 for each device that will write to it)
Total: 6 unique reading addresses required
```
Feel free to take a look at RF24Network for a more complex example, which adds addressing, routing, subnetting etc. as well.

TMRh20, 2015-05-23:
I also noticed you have 
```radio.enableAckPayload();```

This will only work on pipes 0 and 1 unless you call
```radio.enableDynamicPayload```

greeebs, 2015-05-23:
The Arduino's don't actually ever receive anything except the ACKs. Putting that openReadingPipe in there was part of my efforts to try and make it all work and didn't actually change anything.

```
radio.enableDynamicPayloads()
```
has indeed fixed the issue.

Can I suggest the documentation for enableAckPayload() is updated to include the requirement for enableDynamicPayloads() if you want ACKs on pipes other than 0 and 1?

Thanks heaps for your efforts on the library and your assistance with this.

TMRh20, 2015-05-23:
Thanks for the info. Pretty sure I was supposed to update the docs a while
back so plz leave this issue open as a reminder.

closed 2015-06-17 by TMRh20