nRF24/RF24 #113: Weird issue with lost messages

henrikekblad, 2015-05-24:
Hi, 

Not sure were to turn with this rather annoying issue me and @bblacey experienced today when we had a ping-pong setup up and running in the MySensors project. 

The topology looked like this:

NodeA <-> GW <-> NodeB

NodeA sends a ping message via gateway to NodeB and NodeB sends a pong back to NodeA. And so on.

But somehow messages got lost on the way. The strange thing was that the sender got an hardware ack from the recipient radio. But still the RF24:available() would not return true as expected on recipient node.

The behaviour was reproducible on both our setups. 

So my question is if you gave you ever experienced this weirdness? A radio receiving a message and sending ack back without exposing message to the above layers? 

If it had a CRC error on the payload I hope it doesn't send any acks right?

TMRh20, 2015-05-24:
The only time I've seen something like that was with some custom code, where the data being sent was the same on multiple sends. See http://forum.arduino.cc/index.php?topic=273333.0

> The 2 bit PID field is used to detect if the received packet is new or retransmitted. PID prevents the PRX device from presenting the same payload more than once to the receiving host MCU. The PID field is incremented at the TX side for each new packet received through the SPI. The PID and CRC fields (see section 7.3.5 on page 30) are used by the PRX device to determine if a packet is retransmitted or new. When several data packets are lost on the link, the PID fields may become equal to the last received PID. If a packet has the same PID as the previous packet, nRF24L01+ compares the CRC sums from both packets. If the CRC sums are also equal,the last received packet is considered a copy of the previously received packet and discarded.

I'm not expertly familiar with all of the ins and outs of the mysensors code, but this can usually be resolved by ensuring each payload contains unique data so the CRC is unique between sequentially received payloads.

In playing around with a simple scanner, it seems that the radio uses old payloads as the 'auto-ack' packet.

henrikekblad, 2015-05-24:
Ok! Interesting. The content of the received payloads in this case is actually the same. 

But as the radio do a send between each received packets I imagine PID would increase and incoming data considered new and passed to MCU. 

The reuse of old payloads as auto-ack is also an _interesting_ fact. I have to think about how this can be the root cause tomorrow when my head is a bit clearer ;) I hope the content of ack never get passed down though.

bblacey, 2015-05-25:
Well, message uniqueness was in fact the problem for this ping-pong scenario.  I modified the ping pong test program to simply set the message payload to millis() and it now runs correctly.  This may explain a lot about why explicitly repeated messages are not reliably handled in the MySensors network - we aren't explicitly managing uniqueness.  @TMRh20, what is the best approach within the namespace required?  

@henrikekblad, note I think signed messages should always be unique because I believe we use a time salt but we need to ensure unsigned messages are unique as well and it should be handled transparently by the MySensors library.

henrikekblad, 2015-05-25:
Thanks for the help @TMRh20. I'll close this issue.

closed 2015-05-25 by henrikekblad

TMRh20, 2015-05-25:
Well a simple approach is to add a simple counter to each payload that
increments every send. For example, RF24Network uses a 16-bit header id.

See http://tmrh20.github.io/RF24Network/structRF24NetworkHeader.html

henrikekblad, 2015-05-25:
Yes, will do that in the next version of the OTA header. But that will have to wait until next major version.