nRF24/RF24 #116: so.. it is out there

WERAQS, 2015-06-05:
Powering the Next Billion Devices with Wi-Fi

http://arxiv.org/abs/1505.06815

any ideas for our "wireless" sensor network?

TMRh20, 2015-06-06:
Haha, wireless power transmission is something I've been interested in and experimenting with since I built my first electronics project (a self-powered radio) as a kid. If this stuff ends up working out, it looks like it would provide more than enough power for an RF24 sensor network. Very cool.

WERAQS, 2015-06-08:
yep! to calculate how much power my sensors are consuming I had to create 32bit Voltmeter out of Arduino Due to read 1mV change on the battery :) but now.. I think it is possible to create a true wireless-sensor-network out of WiFiPo, on the paper, they're mentioning sensors close to wireless devices such as modem not consuming power at all, I had the same impression but blamed on faulty chinese batteries.. maybe just maybe we're already using it?

eskildsf, 2015-06-08:
You might not want to get your hopes up...
http://www.eevblog.com/2014/08/07/ubeam-ultrasonic-wireless-charging-a-familiar-fish-smell/

WERAQS, 2015-06-08:
first, have you read the paper? 
second, eevblog criticizes EVERYTHING, again, EVERYTHING, but never seen anything useful from him, no offense if you related. 
third, tesla created this tech 100 years ago, fishy uh why are we following the developments or 
fourth and final, why am I using this for my sensors http://www.ebay.com/itm/111673149025 ? which works flawlessly and can charge 10 sensors at once.

TMRh20, 2015-06-08:
For the time being, observation is fine, but I would prefer debates to be kept to public forums etc. instead. That is unless we want to recreate the experiments (or similar) for ourselves, document it, and make the system free and open-source. :p

WERAQS, 2015-06-08:
always! I'm re-designing my boards now, I'll send to itead, get new boards and restart the test, if it succeeds I'll post everything ;) theoretically, I'm putting 2nd power supply on the board, some sort of antenna, transformer and a inductor set, but there must be primary power supply, the battery, CR2032 or CR2450. it'll be similar to iBeacon.

closed 2015-10-13 by Avamander