nRF24/RF24 #117: time.h: No such file or directory

trojanc, 2015-06-06:
Hi

I use platformio to build my arduino projects.
I can't seem to get around the issue of time.h being missing.

I moved from maniacbug's NF24 library to yours, because it seems to have all the features I required.
What I've done is replace the NF24 Library, fixed my code with the new API changes (startWrite() now has a 3rd param)

I've downloaded the Time library from arduino.cc and included it with my other libs, but still now luck...

When compiling I get the following error:

`In file included from .pioenvs/autogen_nanoatmega328/RF24/arch/BBB/compatibility.c:2:0:
.pioenvs/autogen_nanoatmega328/RF24/arch/BBB/compatibility.h:16:18: fatal error: time.h: No such file or directory
#include "time.h"
^
compilation terminated.
scons: *** [.pioenvs/autogen_nanoatmega328/RF24/arch/BBB/compatibility.o] Error 1`

I'm using a Arduino Nano, Linux, with platformio 2.1.0
Any help would be appreciated

trojanc, 2015-06-06:
I found a sort of work around....I just deleted the "arch" directory, because I noticed it's trying to compile all of them, and it didn't look like I needed anything in that directories....seems to work fine now.

TMRh20, 2015-06-06:
Sounds about right. It will detect the device when using arduino IDE or
compiling with gcc but otherwise is best to 'hide' or remove the folders
under /arch  for hardware not in use.

closed 2015-06-06 by trojanc