nRF24/RF24 #118: Some questions about the lib

ncolomer, 2015-06-16:
Hi there,

Thanks a lot for this fabulous well-documented library... it makes my [weather sensors](https://twitter.com/n_colomer/status/587881287748341760) run like a charm :)

Here are few questions, hope you can help:

- I don't see any explicit default value about `PALevel`, `DataRate` and `CRCLength`, `Retries`, `AddressWidth` in the [documentation](https://tmrh20.github.io/RF24/classRF24.html). Do we have to set them each time at `setup()` or are those settings durable? 
- Does the CRC bits reduce the max payload size?
- Is it possible to know what is the (3, 4 or 5-byte) address of the sender when receiving a message? I know the address (we set during init with `openWritingPipe` and `openReadingPipe`) is used under the hood, but I'd like to not include origin address in my message's payload to save some space (among the max 32B payload size)

Thanks for your help,
Nicolas

TMRh20, 2015-06-16:
Hmm, I guess the default configuration is not specifically documented...

1. Durable. Defaults should always be:  PALevel = MAX, dataRate = RF24_1MBPS, retries = (5,15), addressWidth = 5
2. Nope
3. Not only possible, but basically required.  
See the [datasheet](http://www.nordicsemi.com/jpn/content/download/2730/34105/file/nRF24L01_Product_Specification_v2_0.pdf)  - section 7.3 - enhanced shockburst packet format.  Addressing etc. is part of internal radio functionality and does'nt affect payload size. 
Also see section 7.7 - Multiceiver for addressing info.

Very difficult to answer this question, because there are many variables that will affect the actual answer. ie: In RF24Network, because of message routing, the sending node is a required field in the network header. In a direct data-link type communication, this info is known simply by what pipe the message arrived on, unless using multicast.

ncolomer, 2015-06-16:
Hi @TMRh20 and thanks a lot for your answer.

> See the datasheet - section 7.3 - enhanced shockburst packet format. Addressing etc. is part of internal radio functionality and does'nt affect payload size.

Ok so the emitter address is part of a frame. But I see nothing in the RF24 lib that allow to access it. Is there an accessor? Do we have to fetch nRF24L01+ registers ourself for that?

What I meant by "I'd like to not include origin address in my message's payload to save some space" is that I currently encode an id in the payload of each message sent to identify who send what. But I have the feeling that we could directly use the RF address for that (inducing maintaining kind of routing/mapping table)

> Also see section 7.7 - Multiceiver for addressing info.

Actually, I plan to achieve the following scheme: I have plenty of sensors (basically atmega battery-powered µCs) that emit various values at various rates to a central arduino-powered hub that acts as a proxy (ie. forwarding packets) to a more powerful server through an ethernet network. I'd like each RF24 links to be bidirectional so the hub can also send a message back to a particular sensor at any time. 

The upstream (sensors to hub) is pretty easy and I achieved to send messages without too much effort with the help of your lib, but that's the downstream (hub to sensors) part that's worrying me.

What I understand from the datasheet is that we can only achieve a 1-6 network (1 PRX and 6 PTX). Thus I imagine this is not limiting the upstream links (ie. any number of sensors can reach the hub) but only the downstream? Am I write?

I'm wondering if we could emulate kind of "floating" link (by opening and closing pipes) toward various sensors. The hub just have to switch pipes and destination address to send a message to a particular sensor. Do your think it is even possible?

TMRh20, 2015-06-16:
Np, but I don't really do tech support. The datasheet, class docs and additional libs contain about all the information I have. 

1. Pipe Number unless routed
2. No. 1 Master - 5 children - 25 grandchildren - 125 grand-grandchildren, etc all transceivers.
3. Yes/No?? Misunderstanding of functionality? You can only send on pipe0 using any address.

See http://tmrh20.github.io/RF24Network_Dev/Addressing.html
Also http://tmrh20.github.io/RF24Network_Dev/Tuning.html

Topology is dictated by the physical layer. If you want to work outside of that basic structure, you need to disable ESB and likely auto-ack, which will generally require the same type of functionality to be handled by software (inefficient).

ncolomer, 2015-06-17:
It's clearer now! Thanks for your answer, I'll do tests and dig into the doc.
(fyi my project is only a hobby for personal use, no commercial stuff here)
Again, thanks for your help!

closed 2015-06-17 by ncolomer