nRF24/RF24 #119: BBB /dev/spidev2.0 not work

velobiker, 2015-06-17:
Hi! I have beaglebone black rev.C with kernel Linux beaglebone 3.8.13-bone70. I want made radio network with my BBB and arduino UNO (arduino pro mini).
In BBB I have  /dev/spidev1.0 and /dev/spidev2.0  according to (http://elinux.org/BeagleBone_Black_Enable_SPIDEV).
I manual install RF24 (http://tmrh20.github.io/RF24/BBB.html). Then change in RF24/arch/BBB/spi.cpp (spidev2.0 instead spidev1.1). In example GettingStarted.cpp I using RF24 radio(115,1) according to https://github.com/TMRh20/RF24/issues/69. But my device nrf24l01+pa+lna not work.
Then I installed https://github.com/mlsorensen/RF24. In spi.cpp I change this->device = "/dev/spidev2.0";; In example ./pingtest (RF24 radio(115,117)) I see:
RF24/examples/pingpair/
ROLE: Ping out
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xf0f0f0f0e1 0xf0f0f0f0d2
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0xf0f0f0f0e1
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x03
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0f
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_HIGH
What I doing wrong with you library?

TMRh20, 2015-06-22:
Have you tried with the stock config? (using spidev 1.0) or is it not working with either?

velobiker, 2015-06-23:
I changed nrf24l01+pna+lna on usual nrf24l01+. (But lib mlsorensen/RF24 work with nrf24l01+pna+lna). Then I reinstall you lib and tried with the stock config, but when run ./gettingstarted (RF24 radio(115,0)) I see:
root@beaglebone:/usr/src/rf24libs/RF24/examples_RPi# ./gettingstarted
RF24/examples/GettingStarted/
STATUS           = 0xff RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1     = 0xffffffffff 0xffffffffff
RX_ADDR_P2-5     = 0xff 0xff 0xff 0xff
TX_ADDR          = 0xffffffffff
RX_PW_P0-6       = 0xff 0xff 0xff 0xff 0xff 0xff
EN_AA            = 0xff
EN_RXADDR        = 0xff
RF_CH            = 0xff
RF_SETUP         = 0xff
CONFIG           = 0xff
DYNPD/FEATURE    = 0xff 0xff
Data Rate        = 1MBPS
Model            = nRF24L01
CRC Length       = 16 bits
PA Power         = PA_MAX

TMRh20, 2015-06-23:
OK, this seem to point to basic SPI functionality not working I guess. The main difference between mlsorenson BBB support and this one is this uses kernel driven toggling of the CS pin with SPIDEV. 

One thing I noticed on the RPi was that it simply would not work if the CS pin had been left enabled as a GPIO pin. This actually caused me a fair bit of grief in trying to figure out kernel driven CS toggling... (simplest solution: reboot, reinstall this fork and test)

Other than something odd like that, I'm not too sure, because there isn't a whole lot of difference in the SPI functionality, and this was confirmed working on BBB.

velobiker, 2015-07-21:
You wrote:This actually caused me a fair bit of grief in trying to figure out kernel driven CS toggling... (simplest solution: reboot, reinstall this fork and test).
I tried reinstall you fork but result is negative. I liked your library of top-level Network-master. I would like to connect and you library RF24.
According to http://elinux.org/BeagleBone_Black_Enable_SPIDEV
pinctrl-single,pins = <
                                0x190 0x33      /* mcasp0_aclkx.spi1_sclk, INPUT_PULLUP | MODE3 */
                                0x194 0x33      /* mcasp0_fsx.spi1_d0, INPUT_PULLUP | MODE3 */
                                0x198 0x13      /* mcasp0_axr0.spi1_d1, OUTPUT_PULLUP | MODE3 */
                                0x19c 0x13      /* mcasp0_ahclkr.spi1_cs0, OUTPUT_PULLUP | MODE3 */
Perhaps instead 0x19c 0x13      /* mcasp0_ahclkr.spi1_cs0, OUTPUT_PULLUP | MODE3 */
I must reassign spi1_cs0 (GPIO pin 115)?

TMRh20, 2015-09-24:
Sorry for the delayed reply, must have missed this one. The best I can do is point to the thread [here](https://github.com/TMRh20/RF24/issues/69) regarding BBB porting, or suggest trying MRAA.

closed 2015-11-06 by TMRh20