nRF24/RF24 #123: Python Timing Problem

amboleos, 2015-06-19:
Hello everybody

I am working on a raspberry pi gateway which bridges NRF24 network to my cloud server. I have already microcontroller (Pic 24FJ64gb002) based NRF24 circuits and they works well each others. I am using this driver with python in raspberry. My problem is about long waits in driver functions.

At my test, there are a microcontroller based circuit and raspberry pi. I took some screenshots below when raspberry pi sends a package to circuit and circuit sends an ack to raspberry. This ack is a software ack which I create and you can think that it is same as sending a package to raspberry pi.

Green wave is  Circuit IRQ Pin (reciever IRQ Pin)
Yellow wave is Raspberry Pi IRQ pin (Sender IRQ Pin)
Blue wave show the function execution time.

Related code and timing picture are below, In this code, Raspberry sends a package to circuit. Also you can check how much time lasts between stopListening() and startListening().

```python
            send_end_time = time.time()
            if(len(raw_Rf_send_list)>0 and (not (radio.available()) and (send_end_time-send_start_time>0.01))):
                
                GPIO.output(21, False) # BLUE WAVE IS FALSE NOW
                radio.stopListening()
                for packet in raw_Rf_send_list:
                    printlist.append("r:" + str(packet.retry))
                    if(packet.retry >1):
                        rawRfsend(packet.recPayload,packet.reciever,packet.pipe)
                    else:
                        rawRfsend(packet.recPayload,packet.reciever,packet.pipe)
                        raw_Rf_send_list.remove(packet)

                    packet.retry -=1 
                    send_start_time= time.time()
                    break
```
```python
def rawRfsend (packet,addr,pipe):
    printlist.append("p"+str(pipe)+">"+ByteToHex(ByteArrayToStr(packet))) 
    if(pipe ==1  or pipe ==2):
        wAddr = 0xA7A7000000 + addr[0]*0x10000+addr[1]*0x100 + pipe
    else: 
        wAddr = pipe_broadcast 
    radio.openWritingPipe(wAddr)
    radio.write(ByteArrayToStr(bytearray(packet)))
    radio.startListening()
    GPIO.output(21, True)# BLUE WAVE IS TRUE NOW
```
[image]

As you see, receiver tries to send an ack when raspberry is still in TX mode. Raspberry IRQ pin is shows us when it finished the transmit and after that it must execute the startlistening() function to avoid missing a packet. I looked inside the source code but I am not sure why it waits so much. To show deeply which function takes how much time to execute, I put some photos with related function names.

Function execution time calculated like 
```python
GPIO.output(21, False)
#function
GPIO.output(21, True)
```

[image]

mz-fuzzy, 2015-06-19:
Wow, what an impressive measurement.
Well, I'd expect the radio instructions executed a bit faster, but it's quite slow hardware (I guess RPi 1, right?) with OS and python as a scripting language running on top of it.

Can you please measure the instruction duration using C++ code to see what part of slowness is caused by python/wrapper?

amboleos, 2015-06-19:
I am using raspberry b+ v1.2 . I think it is not a big deal to write strings and drive the SPI for this processer.I can execute same function with c++ and show the result tomorrow.

TMRh20, 2015-06-22:
> As you see, receiver tries to send an ack when raspberry is still in TX mode.

Well technically, if you have gotten a data_sent interrupt from the receiving radio module, it would seem to mean the RPi has already switched back to RX mode, received the packet, and sent back an ACK response. I would assume this would point to GPIO speeds, since it may take time after the command was executed to change state?. Even in the C++ lib, general SPI_DEV performance (instead of BCM driver) showed quite a difference when it was changed from using GPIO for the CS pin vs kernel driven toggling. It still uses GPIO for the CE pin toggling as well, which may be impacting the results? Any additional information on this might be pretty useful.

closed 2015-09-24 by TMRh20