nRF24/RF24 #124: RF24Mesh with teensy 3.1 problrms

epa64, 2015-06-22:
Hello,

I continu my Mesh  sensors project with teensy 3.1 an RF24L01+.
I try with the latest version and i have some compilation problems.
The Librairie RF24Network-Development don't compile with errors : RF24Network.h:804:22: error: expected ']' before ';' token and /RF24Network.h:811:41: error: expected unqualified-id before '-' token.
I continu to tested.
With the old version of libraries  (1- April) RF24Network-Development and  RF24Network-Development, the sample RF24Mesh_Example_Master and RF24Mesh_Example have a problem with the mesh.begin() of node RF24Mesh_Example.
####################################################
# Ok the node find the chanel
66244 MSH: Poll response > -64dbm
66244 MSH: Got poll from level 0
66246  MSH: Request address this node: 04
# The node recieve your address
Set address 04 rcvd 04
# the node don't exit to the mesh.begin(). Why ?
66817 MSH: No poll response from level 1
66994 MSH: No poll response from level 2
68579 MSH: No poll response from level 3
70523 MSH: Poll response > -64dbm
70523 MSH: Got poll from level 0
70525  MSH: Request address this node: 04
 Set address 04 rcvd 04

I search the problem ..

Eric Papet
Best regards

TMRh20, 2015-06-22:
Fixed compile error:  https://github.com/TMRh20/RF24Network/commit/c88ccdc804a2d8611fc10f43b593aeb11491794b

If the node goes back to polling immediately, its because the following write and connection test failed, so it tries to renew the address. I would assume a communication issue between the nodes, since it tests OK for me?

epa64, 2015-06-22:
Ok thank's for compilation issues.

I work with the new version of RF24-Development and RF24-Mesh. (Teensy 3.1)

The Samples of  RF24-Development helloword_tx ans helloword_rx works fine except the write method return always 0.

When i try  RF24-Mesh sample RF24Mesh_Example_Master and RF2AMesh_Example or RF24_Example_Node2Node, the node goes back to polling immediately, i don't know why ??
129319 MSH: Poll > -64dbm 
129319 MSH: Got poll from level 0
129320 MSH: Request address 
129324 Set address:    4
129863 MSH: No poll from level 1
129940 MSH: No poll from level 2

It shall seem that the node does not go out of the begin method (), it's possible and why ?

I continu to search !!

Best regards
Eric Papet

epa64, 2015-06-22:
The node loop in RF24Mesh.begin() method, becaus the requestAddress return always 0.
I lookup that the write method to RF25Network-Development return always 0.
while( !network.write(header,0,0) ){
		delay(55);
		printf("Retry register address...\n\r");
		if(registerAddrCount++ >= 6 ){ printf("Retry register address Exit with code 0\n\r");return 0; }
	}
	printf("Retry register address Exit with code 1\n\r");
    return 1;

NODE DEBUG:
12050 MSH: Poll > -64dbm 
12050 MSH: Got poll from level 0
12051 MSH: Request address 
12055 Set address:    4
Retry register address...
Retry register address...
Retry register address...
Retry register address...
Retry register address...
Retry register address...
Retry register address...
Retry register address Exit with code 0

Thank's for Help

Best regard's

Eric Papet

TMRh20, 2015-06-22:
I'm not sure, it works fine for me from RPi to AVR to ARM/Due devices. My best suggestion would be to try with the RF24 examples. If they are not working, then the problems may be external.

epa64, 2015-06-22:
The RF24_Master samples work fine.
pingpair_dyn OK;
transfert OK;

epa64, 2015-06-22:
The R24NetworkDevelopment  helloword_rx and helloword_tx works but the write return method is 0.
TX Sending...                          |     RX Received
Counter :#30                           |     packet #30
Data: 16512                            |     Recieve Data : 16512 at 16512 ms
network.write() return state:0   |    From node :1
failed.                                       |    To node :0

Why ?

I try another sample .

epa64, 2015-06-22:
The same test with  R24Network_Master work better 
TX Sending...                                |       RX Received
Counter :#24                                |       packet #24
Data: 14997                                 |       Recieve Data : 14997 at 14997 ms
network.write() return state:1        |
ok.                                                |

The write method return 1 (true) is not the same return with  R24Network-Development (0 false)
Why ?

TMRh20, 2015-06-24:
Apologies for the late reply, but I don't really have any idea here. I can't seem to recreate the same problem, so I'm assuming it is something different with the hardware (Teensy, RF24 module, or something odd)

It seems as though you've narrowed it down to RF24Network_Dev. All I can think of is there is the one setting, radio.enableDynamicAck(); that is used in RF24Network_Dev that is not used in any of the RF24 examples or RF24Network_Master. Very easy to test, just add ```radio.enableDynamicAck();``` to the gettingstarted sketch after radio.begin();

Beyond that, I'd be interested to see the printDetails() results when called before and after network.begin on the helloworld_tx sketch, as well as the output from enabling the following in RF24Network_config.h:
```
    #define SERIAL_DEBUG
    #define SERIAL_DEBUG_MINIMAL
    #define SERIAL_DEBUG_ROUTING
```
I assume the printf output works on Teensy?

epa64, 2015-06-25:
Thanks for your replies.
Ok I try to test Rf24Network_Master samples with radio.enableDynamicAck(); and printDetails with Rf24Network_DEV.

- When i uncoment #define SERIAL_DEBUG in RF24Network_config.h the sketch does no work, the Serial device doawn after 2 seconds.
- I noticed some difference between  RF24Network_DEV and Master 
	With Dev [Writing 16 bytes 0 blanks] (test02_rf24network_devlopment_tx)
	with Master [Writing 32 bytes 0 blanks] (test01_rf24network_master_tx)
After write debug code i see that the RF24::txStandBy(timeout) return 0 (false) with RF24Network_Devlopment and & (true) with Master.
Another problem when i uncoment #SERIAL_DEBUG in RF24Network_Devlopment RF24Natwork_config.h the skech stop at setpu() network.begin().
The printf method does not work ???

###################################
test02_rf24network_devlopment_tx
###################################
Counter :#1
Data: 2762
write_register(00,0e)
write_register(02,3f)
write_register(11,20)
[Writing 16 bytes 0 blanks]
write_register(07,10)
write_register(07,10)
write_register(00,0f)
write_register(07,70)
network.write() return state:0
failed.
TX Sending...
Counter :#2
Data: 3892
write_register(00,0e)
write_register(02,3f)
write_register(11,20)
[Writing 16 bytes 0 blanks]
write_register(07,10)
write_register(07,10)
write_register(00,0f)
write_register(07,70)
network.write() return state:0
failed.
The receive skech catch data also.

#########################################
test01_rf24network_master_tx
##########################################
Counter :#3
Data: 4938
write_register(00,0e)
write_register(02,3f)
write_register(11,20)
[Writing 32 bytes 0 blanks]
EPA ------>RF24::txStandBy(uint32_t timeout, bool startTx) entry ==> 0
EPA ----> RF24Network(MASTER)::write_to_pipe exit status =>
1
write_register(00,0f)
write_register(07,70)
network.write() return state:1
ok.
TX Sending...
Counter :#4
Data: 6040
write_register(00,0e)
write_register(02,3f)
write_register(11,20)
[Writing 32 bytes 0 blanks]
EPA ------>RF24::txStandBy(uint32_t timeout, bool startTx) entry ==> 0
EPA ----> RF24Network(MASTER)::write_to_pipe exit status =>
1
- exit status is true.

=====> rasio.printDetails before ans after network.begin()
#######################################
RF24_MASTER + RF24Network_MASTER
RF24Network/examples/helloworld_rx/
#######################################

BEFORE BEGIN
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe7e7e7e7e7 0xc2c2c2c2c2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
AFTER BEGIN
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f03c 0xf0f0f0f05a
RX_ADDR_P2-5	 = 0x69 0x96 0xa5 0xc3
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3f
EN_RXADDR	 = 0x3f
RF_CH		 = 0x5a
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
Received packet #0 at 7519
 
#######################################
RF24_MASTER + RF24Network_Development
RF24Network/examples/helloworld_tx/
######################################
BEFORE BEGIN
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe7e7e7e7e7 0xc2c2c2c2c2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
AFTER BEGIN
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xcccccc3ccc 0xcccccc3c3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3f
EN_RXADDR	 = 0x3f
RF_CH		 = 0x5a
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x05
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
Sending...EPA ----> RF24Network(DEV)::write_to_pipe radio.stopListening ...
EPA ----> RF24Network(DEV)::write_to_pipe exit status =>
0
failed.
Sending...EPA ----> RF24Network(DEV)::write_to_pipe radio.stopListening ...
EPA ----> RF24Network(DEV)::write_to_pipe exit status =>
0
failed.

- The receiver catch the data with Master ans Dev network lib.

Thans for your help.

Have you got a teensy 3.1 ?

I can sent your  a teensy 3.1 if you want.

Best regards
Eric Papet
OPEN-ROC project (LGPL)

epa64, 2015-06-26:
Hi,
I just add #define printf Serial.printf in RF24Network_Development and DEBUG work fine.

##########################################
MASTER
##########################################
TEST_MESH_MASTER --> LOOP cpt :0
TEST_MESH_MASTER --> NETWORK NOT AVAILABLE ... 
TEST_MESH_MASTER --> LOOP cpt :1
TEST_MESH_MASTER --> NETWORK NOT AVAILABLE ... 
TEST_MESH_MASTER --> LOOP cpt :2
TEST_MESH_MASTER --> NETWORK NOT AVAILABLE ... 
TEST_MESH_MASTER --> LOOP cpt :3
TEST_MESH_MASTER --> NETWORK NOT AVAILABLE ... 
TEST_MESH_MASTER --> LOOP cpt :4
TEST_MESH_MASTER --> NETWORK NOT AVAILABLE ... 
TEST_MESH_MASTER --> LOOP cpt :9
11040: MAC Received on 0 id 41 from 04444 to 0100 type 194

11040: NET message 0924

11040: MAC Sending to 04444 via 04444 on pipe 0

11040: NET Pipe 0 on node 04444 has address cccccc3ecc

11041: MAC Sent on cccc3ecc ok

11042: MAC Received on 0 id 45 from 04444 to 0100 type 194

11042: NET message 0924

11042: MAC Sending to 04444 via 04444 on pipe 0

11042: NET Pipe 0 on node 04444 has address cccccc3ecc

11043: MAC Sent on cccc3ecc ok

11044: MAC Received on 0 id 49 from 04444 to 0100 type 194

11044: NET message 0924

11044: MAC Sending to 04444 via 04444 on pipe 0

11044: NET Pipe 0 on node 04444 has address cccccc3ecc

11045: MAC Sent on cccc3ecc ok

TEST_MESH_MASTER --> NETWORK NOT AVAILABLE ... 
TEST_MESH_MASTER --> LOOP cpt :10

########################################
NODE
#########################################
 EPA_RF24Mesh.begin(), renewAddres Node ID:1 EPA_RF24Mesh.renewAddress() call ...
setup_address node=04444 mask=07777 parent=0444 pipe=04

5027: NET Pipe 5 on node 04444 has address 3e3e3e3ee3

5027: NET Pipe 4 on node 04444 has address 3e3e3e3e3e

5027: NET Pipe 3 on node 04444 has address 3e3e3e3ece

5028: NET Pipe 2 on node 04444 has address 3e3e3e3e33

5028: NET Pipe 1 on node 04444 has address 3e3e3e3e3c

5028: NET Pipe 0 on node 04444 has address cccccc3ecc

5029: NET Sending id 1 from 04444 to 0100 type 194

5029: MAC Sending to 00 via 00 on pipe 0

5029: NET Pipe 0 on node 00 has27897 MSH: No poll from level 0
27921: NET Sending id 338 from 04 to 0100 type 194

27921: MAC Sending to 01 via 01 on pipe 0

27921: NET Pipe 0 on node 01 has address cccccc3ccc

27922: MAC Sent on cccc3ccc ok

27948 MSH: No poll from level 1
27990: NET Sending id 339 from 04 to 0100 type 194

27990: MAC Sending to 010 via 010 on pipe 0

27990: NET Pipe 0 on node 010 has address cccccc33cc

27991: MAC Sent on cccc33cc ok

28017 MSH: No poll from level 2
28081: NET Sending id 340 from 04 to 0100 type 194

28081: MAC Sending to 0100 via 0100 on pipe 4

28081: NET Pipe 4 on node 0100 has address cc3cc3c33e

28082: MAC Sent on 3cc3c33e ok

28108 MSH: No poll from level 3
28198: NET Sending id 341 from 04 to 0100 type 194

28198: MAC Sending to 00 via 00 on pipe 0

28198: NET Pipe 0 on node 00 has address ccccccccc3

28199: MAC Sent on ccccccc3 ok

28225 MSH: No poll from level 0
28237: NET Sending id 342 from 04 to 0100 type 194

28237: MAC Sending to 01 via 01 on pipe 0

28237: NET Pipe 0 on node 01 has address cccccc3ccc

28238: MAC Sent on cccc3ccc ok

28244: MAC Received on 0 id 297 from 00 to 04 type 194

28244: NET message 0924

28244 MAC: System payload rcvd 194
28244 MSH: Poll > -64dbm 
28244 MSH: Got poll from level 1
28244: NET Sending id 342 from 04 to 00 type 195

28245: NET message 00 00 00 04 

28245: MAC Sending to 00 via 00 on pipe 4

28245: NET Pipe 4 on node 00 has address cccccccc3e

28246: MAC Sent on cccccc3e ok

28246 MSH: Request address 
28248: MAC Received on 0 id 329 from 00 to 04 type 194

28248: NET message 0004

28248 MAC: System payload rcvd 194
28272 MSH: No address response from level 1
28290: NET Sending id 343 from 04 to 0100 type 194

28290: MAC Sending to 010 via 010 on pipe 0

28290: NET Pipe 0 on node 010 has address cccccc33cc

28291: MAC Sent on cccc33cc ok

Best regards
Eric Papet

TMRh20, 2015-06-26:
Are you testing solely on Teensy devices, or do you have any other devices (Arduino Uno, RPi, etc?) to test with?  If it works with other devices, then it would have to be something related directly to the Teensy, and I would probably need to get ahold of one...

epa64, 2015-06-26:
Yes i have a UNO R3.
In April 2015 the MESH skecth work fine whith Teensy.
I test with Arduino.
Can you want to send you à Teensy3.1.

Best regards
Eric Papet

TMRh20, 2015-06-30:
Well I guess if it works with UNO etc, and not with Teensy, but that seems a bit odd. If so, email me tmrh20@gmail.com and we can figure something out.

closed 2015-09-24 by TMRh20