nRF24/RF24 #125: hybrid with ESP8266 carrying Arduino Bootloader

WERAQS, 2015-06-22:
using raspberry pi became too expensive on my sensor networks to carry information to the web and I want to try something simpler and cheaper, I think it is possible to burn code to esp8266 modules with new arduino bootloader and use nRF24L01 with them. nRF24 connected to the ESP will collect data and upload to the internet over simple wifi connection, I'm browsing through codes and wanted to get your ideas on it..

https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide

https://github.com/esp8266/Arduino

*for anyone never heard of ESP8266, they might seem small, but they can beat the Arduino :)

TMRh20, 2015-06-22:
Well, from the github link you provided, it looks like Arduino IDE integration is pretty much all there, including full SPI functionality of the new IDE. In this case, I think it will work "out of the box" as long as it is detected as an Arduino and not something else, and you use the hardware SPI pins.

From there it kind of gets interesting, because I'm not quite sure of the best solution, but it has a TCP/IP stack, and could probably work as a router to some degree. All in all, I think it could be as simple as modifying a few things to have it working with RF24,RF24Network and RF24Mesh. It might take some doing to look at how it could work as a master RF24Ethernet node, but the lower layers should be easy.

WERAQS, 2015-06-23:
yes it seems like a perfect fit, in fact, I don't want to change anything on lower layers.. they don't have to know whats changed, lol. I just want to deactivate Raspberry Pi from the network (because I'll be expanding the network to 250 nodes soon, I have to get 4 more raspberry pi to safely keep up with their logging and routing. my nodes using dual IDs and dual Channels to communicate with the base, one for 1 raspberry pi A+ which has usb memory as logger, and another name tag for Raspberry Pi B+ which routes data to the SQL server.

I'll be testing other features of the ESP8266 chips, unluckily one of my ESP-03's ceramic antenna damaged and I don't know if it still works. from what I can see it is impossible to set nRF24 routing with ESP-01 modules (because they have 2 GPIO pins available - but that also reminds me this -> http://nerdralph.blogspot.com.tr/2014/01/nrf24l01-control-with-3-attiny85-pins.html we've seen many impossibilities become reality with nRF24s thanks to developers like you)

I'll be ordering some ESP-03's which seems to have perfect price to range ratio and test the codes one by one..

ppotok, 2015-07-04:
Hello,
WERAQS, please how did you get to work ESP? I tried it with nodemcu v2 and I get error:

In file included from /home/potok/sketchbook/libraries/RF24/RF24.cpp:9:0:
/home/potok/sketchbook/libraries/RF24/nRF24L01.h:34:21: error: expected unqualified-id before numeric constant
 #define STATUS      0x07
                     ^
/home/potok/Arduino/arduino-1.6.5/hardware/esp8266com/esp8266/tools/sdk//include/c_types.h:58:3: note: in expansion of macro 'STATUS'
 } STATUS;
   ^
Error compiling.

------------------------------
Sorry for posting here, I cant find your email.

WERAQS, 2015-07-06:
I didn't, not yet, I'm compiling my thesis now.

crcastle, 2015-07-12:
@ppotok I got the same error and was able to fix it (and a bunch of other compile-time errors) by hacking on the RF24 library.  I replaced `#define STATUS 0x07` with `#define NRF_STATUS 0x07` and then in RF24.cpp replaced all occurrences of `STATUS` with `NRF_STATUS` (be careful not to edit `FIFO_STATUS`).

I then made some additional changes to RF24.cpp to deal with some `printf` compile problems.

Then I had to disable the watchdog timer on the ESP8266.

Now it's compiling and printing Serial.println()s but RF24 functionality is not working correctly.

Here is my code on 1) the ESP8266 with NRF24L01+, 2) serial output from ESP8266, and 3) sensor node I'm using to send test data to the NRF24L01+ on the ESP8266: https://gist.github.com/crcastle/93ebe3b42f3ab021639b

crcastle, 2015-07-13:
Made some progress.  I can now use the nRF24L01+ with the ESP8266 as the MCU!!!

In addition to:
* Renaming `STATUS` as `NRF_STATUS` in RF24.cpp (mentioned above)
* Converting all `printf_P` calls to `printf` in RF24.cpp (mentioned above)
* And changing `#include <avr/pgmspace.h>` to `#include <pgmspace.h>` in RF24_config.h (didn't to mention above)

I made a mistake with CE pin wiring and `RF24 radio` instantiation.  See this [gist comment](https://gist.github.com/crcastle/6bab21445c857993d1d8#comment-1491254) for more details.

I also "sprinkled" some `yield()`s around my code to make sure the ESP8266 has ample time to handle it's wifi and tcp/ip operations.  This is a requirement unique to the ESP8266.

TMRh20, 2015-07-13:
The [Updates](https://github.com/TMRh20/RF24/archive/Updates.zip) branch should now contain the changes per @crcastle, seems to compile fine for ESP8266, and is available for testing. Thanks for the input & testing!

crcastle, 2015-07-14:
Thanks @TMRh20!  I will test this out when I get some extra time, but I have decided that I will need to use an ATmega328P MCU to interface between the ESP8266 and nRF24L01+ (instead of using the ESP8266 as the MCU).  The problem for me is power consumption -- my requirements are for this module to run >300 days on user-replaceable batteries.  Even using D cell batteries with 15000 mAH I'll only get a week to a couple weeks.

The ESP8266 cannot be woken up from deep sleep with an interrupt, deep sleep being necessary to get idle power consumption to <100μA.  It's not clear to me if this is a hardware limitation or a software limitation in Espressif's SDK.  I'm hoping it's the latter, but for now, I'm going to use the ATmega328P to switch ESP8266 power on and off as the nRF24L01+ receives sensor data.

Regardless, @ppotok, I bet the [Updates](https://github.com/TMRh20/RF24/archive/Updates.zip) build will fix your problem!

WERAQS, 2015-07-15:
developments! great! 

it is possible to wake up ESP8266 using ch_pd or sth interrupt but need to load the same software or wait for the connection to the WiFi. since 328p costs less than $1.3 (smd) and been tested relentlessly for almost a year at network, it is possible to use. my main intention was to replace Raspberry Pi's job with it and from what I've seen, its almost done

ps. the perfect battery match to use them with 18650 in parallels with ebay item -> 291487681101 cases 3 of them perfectly fits on a board if you design yourself with either end as an antenna

ppotok, 2015-07-15:
Wow, I love you guys!
I tried to get to work Gettingstarted example, but ESP-12E (Nodemcu V2) Is getting continously restarted. with error : ,rst cause:4, boot mode:(3,6)

It do reset even if the module is not connected :-(

just to be sure with connection
CE - 7
CSN - 8
SCK - CLK (its named as SCLK)
MOSI - SD1 
MISO - SD0

I am sorry fort bothering you, but I didn't find anyone else with same problem :-(

crcastle, 2015-07-16:
@ppotok I connected as described here http://d.av.id.au/blog/esp8266-hardware-spi-hspi-general-info-and-pinout/ and had to add one additional pin for CE.  I used GPIO2.

crcastle, 2015-08-06:
@TMRh20 just used the [Updates](https://github.com/TMRh20/RF24/archive/Updates.zip) branch code.  It compiled successfully and works great on my ESP8266.  Thanks for adding support for that!

WERAQS, 2015-08-10:
WOW! I've been away for a while to write the thesis and EXACTLY! it compiles but have to remove some folders, I have all those boards and had to remove them for the operation. [image]

to make it compile, delete those folders (for windows 8.1);
C:\Users\<user name>\AppData\Roaming\Arduino15\packages\intel\hardware (if you have Galileo or other i586 restore it afterwards) 
C:\Users\<user name>\AppData\Roaming\Arduino15\arduino\hardware\sam ((if you have due or zero restore it afterwards) 
and I don't know why but, delete this too C:\Arduino-1.6.5-r2\hardware\arduino (restore it if you are going to use any arduino board)

TMRh20, 2015-08-12:
@WERAQS  it seems odd that you would need to delete those folders, can you post the compile errors you get with them restored?

WERAQS, 2015-08-13:
@TMRh20 Don't worry, I have multiple versions of Arduino software installed on this os, some of the older versions seems to work better and compile faster. On the debug side, disabling printf.h makes all codes compile.

In file included from pingpair_ack.ino:20:0:
C:\Users\i7\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\libraries\RF24/printf.h: In function 'void printf_begin()':
C:\Users\i7\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\libraries\RF24/printf.h:31:29: error: 'fdevopen' was not declared in this scope
   fdevopen( &serial_putc, 0 );

Multiple libraries were found for "SPI.h"
 Used: C:\Users\i7\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\libraries\SPI
 Not used: C:\Users\i7\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\libraries\SPI
Multiple libraries were found for "RF24.h"
 Used: C:\Users\i7\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\libraries\RF24
 Not used: C:\Users\i7\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\libraries\RF24

maribeiro, 2015-08-23:
Any luck getting the radio.printDetails() working on the ESP8266 ?

TMRh20, 2015-08-25:
@maribeiro As a guess, have you tried changing the PROGMEM related stuff, ie: ``` Serial.print(F("" blah")); ``` to ``` Serial.print(" blah"); ``` Just a guess based on the notion that it works ok with RF24Network.

pat2on, 2015-09-10:
@TMRh20 Thank you for your great work on RF24 and RF24Network libs and sharing your expertise! 
Unfortunately I cannot make any of examples running on ESP8266..
After spending quite a lot of time on this topic I decided to post my first help-comment here. 
Let's take helloworld_tx: 
The device restarts on each trial to send data spiting on serial:
====================================
RF24Network/examples/helloworld_tx/
Sending...
Exception (3):
epc1=0x40203006 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4022e04c depc=0x00000000

ctx: cont 
sp: 3ffe9c70 end: 3ffe9f20 offset: 01a0

>>>stack>>>
3ffe9e10:  00000022 3fffc6fc 00000001 00000000  
3ffe9e20:  00000000 00000000 3ffe8c60 40202492  
3ffe9e30:  00010000 3ffe8f00 4020182c 00000000  
3ffe9e40:  00000000 00000000 00000000 ffdfffff  
3ffe9e50:  ffffffff 3fffc6fc 00000059 00000008  
3ffe9e60:  00000038 3ffe9ef0 3ffe8c60 4020275d  
3ffe9e70:  3ffe9ef8 3ffe8e2c 00000001 40202aa0  
3ffe9e80:  40201124 3ffe8c60 3ffe9ef0 00000018  
3ffe9e90:  00000008 3ffe9ef0 3ffe8c60 402027c8  
3ffe9ea0:  00000000 00000038 3ffe9ef8 3ffe9f4c  
3ffe9eb0:  3fffdc20 3ffe8c60 3ffe9f78 4020527d  
3ffe9ec0:  3ffe8830 00000001 3ffe8c60 3ffe9f4c  
3ffe9ed0:  3fffdc20 3ffe8c60 3ffe9f78 40202955  
3ffe9ee0:  3fffdc20 3ffe8c60 3ffe9f78 40201d80  
3ffe9ef0:  00000001 00000001 000007d0 00000000  
3ffe9f00:  00000000 00000000 3ffe9f44 4020184a  
3ffe9f10:  00000000 00000000 3ffe8f00 40100378  
<<<stack<<<
================================
I suppose it's not power related (using external battery pack - tested 2 different types). 
Wiring like above (http://d.av.id.au/blog/esp8266-hardware-spi-hspi-general-info-and-pinout/) with CE -> GPIO2 and CS -> GPIO15.
Corrected all "Serial.print(F(".
Tried with ESP07, ESP12 and NodeMcu v1.0. 
Playing with different pins and nothing. 
Where is my bloody mistake? I am quite new with ESP...

cacatpunctro, 2015-09-17:
Use the needed files from https://github.com/mysensors/Arduino/tree/development/libraries/MySensors/utility
uncomment #include <MyConfig.h> from RF24_config.h

TMRh20, 2015-09-18:
@cacatpunctro Thanks much for the info regarding the mySensors fork, but I would encourage users to contribute the needed changes to this repository instead of copying specific files to support certain hardware.

Based on the provided info, I just corrected what I'm sure is a mistake regarding using %S instead of %s regarding the ESP8266, and there seems to be only one other main change regarding the use of progmem in openReadingPipe whatHappened and stopListening functions. If the ESP works with the current mysensors fork and still has problems with this for, then the remaining changes would be very easy to include here.

Please let me know!

cacatpunctro, 2015-09-18:
RF24/examples/GettingStarted
*** PRESS 'T' to begin transmitting to the other node
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK

Exception (3):
epc1=0x40202d6e epc2=0x00000000 epc3=0x00000000 excvaddr=0x4022d66c depc=0x00000000

ctx: cont 
sp: 3ffe9b30 end: 3ffe9d10 offset: 01a0

>>>stack>>>
3ffe9cd0:  00000000 00000000 00000016 401015f9  
3ffe9ce0:  402017ed 3ffe8006 3ffe8c0c 3ffe9d3c  
3ffe9cf0:  3fffdc20 00000000 3ffe9d34 4020184a  
3ffe9d00:  00000000 00000000 3ffe8cf0 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
RF24/examples/GettingStarted
*** PRESS 'T' to begin transmitting to the other node

cacatpunctro, 2015-09-18:
I'm trying to send data from/to esp/arduni using esp>nrf24 --- nrf24<arduino

There is something really broken with the esp8266 arduino implementation or with this port of the library. It all seems related to PROGMEM or stdio redirection.

Using the mysensors lib, alltough it doesn't crash my esp8266, GettingStarted example allways fails to send/receive packets.

Now, using the pingpair_ack example I am able to send data from arduino to esp, but esp crashes when switching to transmit mode. 

Hope it helps

TMRh20, 2015-09-18:
Yeah, this is a *big* help.

Since the two forks are still pretty closely aligned, I decided to try two ways of doing things. 

1. In the [updates branch](https://github.com/TMRh20/RF24/archive/Updates.zip) I redefined all of the needed PROGMEM related variables to be handled the same as they are on devices like Arduino Due that do not have progmem. In theory, this should rule out any progmem related problems, since the mysensors fork seems to leave in a bunch of progmem related code, while removing other parts.
*note* If you enable verbose output during compilation (file > preferences in Arduino IDE) you should receive warnings about PROGMEM vars being redefined. The warnings can be removed, but are a clear indication that things are being redefined as wanted.

2. In [this gist](https://gist.github.com/TMRh20/109d09623597df479e99/archive/48a6702d8ed6ce6bae177bde37dc405bd5a2feec.zip) I added modified RF24.cpp and RF24_config.h files to align with the current ESP changes in the mysensors fork. In theory, this should provide the same or very similar results as the current mysensors code, unless I messed something up. 

Also, if printf works, you could try recompiling with ``` #define SERIAL_DEBUG ``` uncommented in RF24_config.h, and (with either fork) it should give us a good idea of what exactly is happening before the crashing. 

So, two more things to try if anybody is interested.

boguslawb, 2015-09-19:
Maybe some data types may be differently defined  and/or aligned in memory in esp8266 ?

cacatpunctro, 2015-09-20:
With the Updates branch I got the exact same as with the mysensors: GettingStarted no longer crashes the ESP but it doesn't work (packet failed to transmit), pingpair_ack crashes the ESP as soon as T is pressed, but works if it receives.

TMRh20, 2015-09-23:
@cacatpunctro The thing that bugs me is that @crcastle seems to report ok operation, without any crashing. I would begin to assume that, with it working on one and not the other, something else has to be different. ( pins, config, etc, I dont know?? )

Beyond me speculating or you reviewing pins etc, it might be worth adding some debugging code using serial print to find out where or what is the cause. It will take me some time to do this, so it might just be easier for me to order an ESP at this point. I probably should lol.

cacatpunctro, 2015-09-23:
That wouldn't be a bad ideea :) Get a nodedevkit since it's got pins soldered and stuff :)
I'm only trying the examples, board is connected using HSPI. Some ppl using the mysensors are reporting resets too (I never used the full lib, only tried the nrf part).
I'm sorry I can't be of any help since looking at the code nothing stands out to be wrong.

TMRh20, 2015-09-23:
Thanks for the info, now its mostly a matter of shipping time.

Also, just a thought, because I'm not sure if that made any difference for @maribeiro , and nobody has specified I think, but if using the examples, ensure removal all progmem and printf related references including ``` printf ``` and ``` Serial.print(F" ``` statements of course.

crcastle, 2015-09-23:
Hi guys, just saw I got mentioned.  If it's useful in debugging, here's my setup so you can compare.  I've had this running for over a month non-stop now.

I'm using an [Adafruit Huzzah](https://www.adafruit.com/product/2471) and [this nRF24L01+ radio](http://amzn.com/B00E594ZX0).  It's serving as my RF24Network gateway.  It receives sensor data from other nRF24L01+ nodes (currently just 3 nodes plus the gateway) and sends that sensor data to an HTTP API.

Here's a pic:
[image]

Wiring details:
DC +5V in going straight to the Huzzah (it has a regulator)
Huzzah 3V3 out pin to power the nRF24L01+
Code and SPI pin connection details: https://gist.github.com/crcastle/9dd9dcad86eee7e5b4a2

cacatpunctro, 2015-09-24:
Well, here is my setup, not as pretty as @crcastle 
I have also tried with a 5V adaptor going directly to the board.

I downloaded RF24 + RF24Network, opened "examples/rf24network/helloworld_tx" and changed pins to RF24 radio(2,15); 

There is no module to receive what it's beeing sent ...

It gets to Sending... then loops

n 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
RF24Network/examples/helloworld_tx/
Sending...
Exception (3):
epc1=0x40203042 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4022e088 depc=0x00000000

ctx: cont 
sp: 3ffe9c70 end: 3ffe9f20 offset: 01a0

[image]

maribeiro, 2015-09-24:
I've managed to get it working by using only the RF24Network examples.
The examples from the RF24 folder (the GettingStarted one for example) gives me that boot loop you're getting too.
I cant remember if the changing to Serial.print  (as recommended previously by @TMRh20) made the RF24 GettingStarted example to work...IIRC I dont think so but I'll try them again tomorrow. 
Nevertheless the RF24Network examples work and that's the ones I use as a start for my sketches.

cyberp, 2015-10-02:
Hi all,
thanks for all comments and the code posted here. 
That all helped me a lot.
I used the code from the updates branch and my ESP-12 crashes like cacatpunctro's, when i try to send something. Receiving data works fine.
I managed transmitting data without reset of the ESP when i changed one line of code in **RF24.cpp**:

```
void RF24::stopListening(void)
...
// comment this out:
//write_register(EN_RXADDR,read_register(EN_RXADDR) | _BV(pgm_read_byte(&child_pipe_enable[0]))); // Enable RX on pipe0

// replace with
write_register(EN_RXADDR,read_register(EN_RXADDR) | _BV(ERX_P0)); // Enable RX on pipe0
```
It seem's that all the pgm_read_write stuff is dangerous.

TMRh20, 2015-10-23:
So I finally received my NodeMCU and tested it out with RF24, and I have no problems using the master branch.

I'm using a nodemcu v0.9 (ESP-12) module & followed [this tutorial](http://benlo.com/esp8266/esp8266QuickStart.html) to get up and running.

Once working with Arduino. I connected the SPI pins as shown in [this pinout](http://www.14core.com/wp-content/uploads/2015/06/Node-MCU-Pin-Out-Diagram1.png) and used the same RF24 config as shown above ( RF24 radio(2,15) )

Gettingstarted (RF24)
```
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
Now sending
Sent 8091017, Got response 8089131, Round-trip delay 1886 microseconds
Now sending
Sent 9093285, Got response 9091446, Round-trip delay 1839 microseconds
Now sending
Sent 10095507, Got response 10093692, Round-trip delay 1815 microseconds
*** CHANGING TO RECEIVE ROLE -- PRESS 'T' TO SWITCH BACK
Sent response 93585244
Sent response 94591564
Sent response 95595328
Sent response 96599140
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
Now sending
Sent 18915178, Got response 18913362, Round-trip delay 1816 microseconds
Now sending
Sent 19917405, Got response 19915590, Round-trip delay 1815 microseconds
```

helloworld_tx (RF24Network):
```
Sending...ok.
Sending...ok.
Sending...ok.
Sending...ok.
Sending...ok.
Sending...ok.
```

The main thing I'm not understanding about the different changes folks are making, is that the defines such as ``` #define pgm_read_word(p) (*(p)) ``` are effectively removing any usage of PROGMEM, so the changes in the master branch should have the exact same effect as the changes listed in the above post by @cyberp . This is the same method used on Raspberry Pi, Arduino Due, and all devices that do not use program memory.

??? All seems good to me

maribeiro, 2015-10-29:
@TMRh20  I can confirm that GettingStarted example now works fine on my nodemcu too ;)
One thing I miss is the radio.printDetails() function, because it helps while debugging to see if there's a problem with connections.

TMRh20, 2015-10-31:
@maribeiro Added in [master branch](https://github.com/TMRh20/RF24/archive/master.zip).  ```Serial.begin()``` needs to be called of course, and ```radio.printDetails()``` or ```printf()``` will now send output to serial.

maribeiro, 2015-10-31:
@TMRh20  many thanks for your work, I  really appreciate it!!!

WERAQS, 2015-11-07:
a small trick but it works, if you are complaining about 0x707070707E radio channel default, just run setup twice on your nodes.

  radio.begin();
    delay(2);
    radio.powerUp();
    radio.enableDynamicPayloads();
    radio.setDataRate(RF24_1MBPS);
    radio.setPALevel(RF24_PA_MAX);
    radio.setChannel(120);
    radio.setRetries(5,15);
    radio.openWritingPipe(0xA0A0A0A0A1LL); 
    radio.openWritingPipe(0xA0A0A0A0A2LL); 
    delay(25);
    radio.powerDown();
    delay(25);
    radio.begin();
    delay(2);
    radio.powerUp();
    radio.enableDynamicPayloads();
    radio.setDataRate(RF24_1MBPS);
    radio.setPALevel(RF24_PA_MAX);
    radio.setChannel(15);
    radio.setRetries(15,15);
    radio.openWritingPipe(0xB0B0B0B0B1LL); 
    radio.openWritingPipe(0xB0B0B0B0B2LL); 

void loop..

TMRh20, 2015-11-11:
I've been playing around a bit with these ESPs, and they are pretty cool, but I wish development was a bit further along. It seems to support Station + AP mode at once, but its looking like handling routing etc. is mostly out of the question for now, which is a bit of a bummer. I was hoping to chain some together and have routing similar to RF24Network. It also looks like they recently released the source code for the IP stack, but not for the full interface??

With that in mind, I just managed to get the latest [git version](https://github.com/esp8266/Arduino#using-git-version-) working with the whole RF24 stack and have been playing around with dual network interfaces on the ESP using RF24Ethernet. Out of curiosity, I'm probably going to have to look at the LWIP stack now, since this would provide two interfaces, and could be adapted to other hardware etc, or the ESP itself once/if the code is made available.

I've searched and tested etc., and I can't seem to make anything happen with IP forwarding or routing between a softAP and a wifi connection to my router. I'm guessing this just isn't possible right now??

someburner, 2015-11-12:
Not sure if this helps at all, but I wrote a library for RFM69 modules (referenced above), for use with the Native Espressif IoT SDK (1.4). It's capable of doing OTA flashing of a remote Moteino node (~15 kB) in one go so it works pretty well, albeit the code being sloppy. 

As @boguslawb mentioned, the ESP8266's SPI interface (and RAM) must be read form/written to with 32-bit aligned data. Not sure if the Esp-Arduino platform handles this or not, but definitely something to consider (if it hasn't been already). The SPI library linked to in my repo is a good example of how to do this.

Also as mentioned, it is best to avoid blocking loops wherever possible so as not to trip the WDT. It looks like esp-Arduino has a timer scheduler but to be safe you could always use something like SimpleTimer (http://playground.arduino.cc/Code/SimpleTimer) to help you set up timers and then just call the polling function timer.run() in your main loop. If execution of your .ino is paused by delay() then doing this could speed up code in your main loop.

itsjustvenky, 2015-11-17:
[image]

Thank you so much for your work. It works like a charm....

kartakur, 2015-12-04:
I have Wemos D1, still geting error pls help. What should I download or what pins I should connect.

WERAQS, 2015-12-04:
@kartakur we use google and "reading" yes! actual reading!

madshobye, 2015-12-09:
For those who are struggling with the different version of nodemcu and the pin configurations of spi i suggest you look at this:

http://www.mysensors.org/build/esp8266_gateway

[image]

closed 2015-12-14 by Avamander

kalibatta, 2018-06-17:
@TMRh20
 I see this issue was closed but it still exists. I just used your latest repo but was still getting the same until i followed **cyberp** comment
 

> void RF24::stopListening(void)
> ...
> // comment this out:
> //write_register(EN_RXADDR,read_register(EN_RXADDR) | _BV(pgm_read_byte(&child_pipe_enable[0]))); // Enable RX on pipe0
> 
> // replace with
> write_register(EN_RXADDR,read_register(EN_RXADDR) | _BV(ERX_P0)); // Enable RX on pipe0

Kindly update to save others

Avamander, 2018-06-17:
@kalibatta 
Are you using the latest version of the ESP8266 core?

kalibatta, 2018-06-18:
No I am using 2.3.0 Ok will try updating the esp core and i see. Otherwise thanks for the nice library am just praying that you apply the same stack on espwifi mesh - i like the addressability in your library than the painless mesh for esp