nRF24/RF24 #130: [noob] debug connection between nrf24l01 and arduino

joihn, 2015-08-18:
Hello

I have problem of connection between the module and the arduino (maybe bad jumper...)

Is there a way to check this connection with your library ? unlike the originial library of maniacbug, the gettingstarted don't say anything...

uggima, 2015-08-18:
just getting back into these libraries so couldn't say exactly, but would think you could find a way to read a setting/register from the nrf24 module.

Had a quick look and found https://github.com/TMRh20/RF24/blob/master/RF24.h#L1088-L1088 hope it helps

TMRh20, 2015-08-20:
Can also use printf.h as with the maniacbug lib. See the [GettingStarted_CallResponse.ino](https://github.com/TMRh20/RF24/blob/master/examples/GettingStarted_CallResponse/GettingStarted_CallResponse.ino) for an example. 

1. Uncomment ``` #include <printf.h> ```
2. Uncomment ``` printf_begin(); ``` after serial.begin
3. Add ``` radio.printDetails(); ```

joihn, 2015-09-01:
thanks for your help, I still have the problem, "response timed out"

Here are the output of the "radio.printDetails();" maybe someone can help me to analyse it :)
both card display the same output

RF24/examples/GettingStarted_CallResponse
*** PRESS 'T' to begin transmitting to the other node
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x65646f4e31 0x65646f4e32
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x65646f4e31
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x03
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x06
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_LOW

TMRh20, 2015-09-02:
If it looks the same on both devices, you may just need to set the identifier:

```
/***      Set this radio as radio number 0 or 1         ***/
bool radioNumber = 0;
```

Avamander, 2015-09-09:
@TMRh20 The OP has not posted a while. It can be assumed it is solved. Could you close the issue?

TMRh20, 2015-09-10:
Aye.

closed 2015-09-10 by TMRh20

joihn, 2016-01-08:
sorry for the big delay I got busy ;)
thanks you, it was the identifier, it works now :D YAY

Avamander, 2016-01-09:
I just noticed it too. radio.printDetails(); does not print anything.

joihn, 2016-01-09:
example.

add `<#include <printf.h>>`
Uncomment printf_begin(); after serial.begin
Add radio.printDetails();

WelshFred, 2017-07-24:
I cannot get this library to work on two Due's I am currently using Getting Started program downloaded last week, Modified date Dec 2014. Added radio.printDetails(); as first line in void loop() and I get the following output:
Device set to 1 using radio.printDetails();
Sent response 0
= 0xff RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1. = 0xffffffffff 0xffffffffff
RX_ADDR_P2-5. = 0xff 0xff 0xff 0xff
TX_ADDR.. = 0xffffffffff
RX_PW_P0-6. = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA.. = 0x00
EN_RXADDR. = 0xff
RF_CH.. = 0xff
RF_SETUP. = 0xff
CONFIG.. = 0xff
DYNPD/FEATURE. = 0xff 0xff
Data Rate. = 1MBPS
Model.. = nRF24L01
CRC Length. = 16 bits
PA Power. = PA_MAX
Sent response 50331648
 RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1. = 0xffffffffff 0xffffffffff
RX_ADDR_P2-5. = 0xff 0xff 0xff 0xff
TX_ADDR.. = 0xffffffffff
RX_PW_P0-6. = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA.. = 0x00
EN_RXADDR. = 0xff
RF_CH.. = 0xff
RF_SETUP. = 0xff
CONFIG.. = 0xff
DYNPD/FEATURE. = 0xff 0xff
Data Rate. = 1MBPS
Model.. = nRF24L01
CRC Length. = 16 bits
PA Power. = PA_MIN
Sent response 0

Device set to 0 using radio.printDetails();
*** PRESS 'T' to begin transmitting to the other node
Sent response 0
= 0xff RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1	 = 0x0000000000 0x0000000000
RX_ADDR_P2-5	 = 0xc0 0x00 0x00 0x00
TX_ADDR		 = 0x0000000000
RX_PW_P0-6	 = 0xff 0xff 0xff 0xff 0xff 0xff
EN_AA		 = 0xff
EN_RXADDR	 = 0xff
RF_CH		 = 0xff
RF_SETUP	 = 0xff
CONFIG		 = 0xff
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = Disabled
PA Power	 = PA_MAX
STATUS		 = 0xff RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1	 = 0xffffffffff 0xffffffffff
RX_ADDR_P2-5	 = 0xff 0xff 0xff 0xff
TX_ADDR		 = 0x0000000000
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0xff
EN_RXADDR	 = 0xff
RF_CH		 = 0xff
RF_SETUP	 = 0xff
CONFIG		 = 0xff
DYNPD/FEATURE	 = 0xff 0xff
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = Disabled
PA Power	 = PA_MIN
Sent response 0
To give you as much detail as I can set as default pins for SPI MOSI pin 51, MISO pin 50, SCK pin 52, CE pin 9 and  CSN pin 10 {note Due can be pin 4,10,53} If I use pin 53 get:
SeîRF24/examples/GettingStarted
*** PRESS 'T' to begin transmitting to the other node
Which is what I would expect, press 'T' and nothing happens. If by changing CSN to pin 10 get the following:
SeîRF24/examples/GettingStarted
*** PRESS 'T' to begin transmitting to the other node
Sent response 4194304
Sent response 251723776
Sent response 520093696
Sent response 251658240
Sent response 4194304
Sent response 4194304
Sent response 1594818560
However I have not pressed a key yet, both do the same. I never see this on pin 53 nor pin 10 *** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK".
 YES I USE CAP 'T' when I'm on pin 53 because of this line in your code: if ( c == 'T' && role == 0 ){   
On pin 10 as I have said just runs on both. These nRFL24's are on a separate power lines with grounds common. 
Please help!

Avamander, 2017-07-24:
@WelshFred Wiring is incorrect, module is faulty, or your pin definitions are wrong.