nRF24/RF24 #133: rf24Network tx error when using mixed module nrf24L01+ and nrf24L01+PA/LNA

PeterSek, 2015-09-02:
I am using latest rf24/24Network library. I noticed 'strange' behavior in that TX fails (return false) when there is a mismatch between trx radios - one trx is nrf24L01+ and the other nrf24L01+PA/LNA. 
Issue is regardless which one tries to transmit.
But same code works OK when BOTH are same either nrf24L01+ or BOTH nrf24L01+PA/LNA

Also earlier library prior to recent commit does not seem to have this problem
TRX radios are about 1 meter appart, LOW power

  //RADIO
  SPI.begin();
  radio.begin();
  radio.setRetries(15, 15);

  network.begin(/*channel*/ 90, /*node address*/ thisNode);
  radio.setDataRate (RF24_250KBPS);
  //radio.setPALevel(RF24_PA_MIN);
  setPALevel(getPALevel());       //set to MIN/LOW
  radio.printDetails();

STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xcccccccee3
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3f
EN_RXADDR	 = 0x3f
RF_CH		 = 0x5a
RF_SETUP	 = 0x21
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x05
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MIN

TX part
  RF24NetworkHeader header(/*to node*/ otherNode);
  done = network.write(header, &payload, sizeof(payload));
  if (!done) {
    Serial.println ("TX.ERROR");
  }

TMRh20, 2015-09-02:
One of the biggest differences between the new and the old is the use of dynamic payloads. This allows the radio to send smaller packets as well as knowing how long each received payload was.

Unfortunately, a number of 'clone' modules do not support dynamic payloads it seems, so you may have to revert to the 'old' functionality to have it work with mixed modules.

Easy way to test:
1. Open RF24Network_config.h in a text editor
2. Comment out ``` #define ENABLE_DYNAMIC_PAYLOADS ```
3. Recompile & Test

With dynamic payloads disabled, the radio will send/receive 32 bytes in every transmission, and when using fragmentation, will only be able to determine how many fragments, not the size of the re-assembled payload.

This can also be tested using the core RF24 library example, pingpair_dyn.ino

PeterSek, 2015-09-03:
Thank for reply.

when when I use mixed nrf24l01+ and nrf24l01+PA/LNA
RF24 library example pingpair_dyn.ino
WORKS OK

when when I use mixed nrf24l01+ and nrf24l01+PA/LNA
rf24Network example helloworld_tx/rx.ino
With RF24Network_config.h Comment out #define ENABLE_DYNAMIC_PAYLOADS
and just in case I also added radio.enableDynamicPayloads(); before/after
network.begin
Partially works RX receive the packet but TX 'sending...failed'

RF24Network/examples/helloworld_tx/
STATUS = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1 = 0xcccccc3ccc 0xcccccc3c3c
RX_ADDR_P2-5 = 0x33 0xce 0x3e 0xe3
TX_ADDR = 0xcccccccc3c
RX_PW_P0-6 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA = 0x3f
EN_RXADDR = 0x3f
RF_CH = 0x5a
RF_SETUP = 0x01
CONFIG = 0x0f
DYNPD/FEATURE = 0x3f 0x05
Data Rate = 1MBPS
Model = nRF24L01+
CRC Length = 16 bits
PA Power = PA_MIN
Sending...failed.
Sending...failed.

RF24Network/examples/helloworld_rx/
STATUS = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5 = 0x33 0xce 0x3e 0xe3
TX_ADDR = 0xe7e7e7e7e7
RX_PW_P0-6 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA = 0x3f
EN_RXADDR = 0x3f
RF_CH = 0x5a
RF_SETUP = 0x01
CONFIG = 0x0f
DYNPD/FEATURE = 0x3f 0x05
Data Rate = 1MBPS
Model = nRF24L01+
CRC Length = 16 bits
PA Power = PA_MIN
Received packet #0 at 2000
Received packet #1 at 4000

Now, as soon as I change radios on both ends to be same: both + or botrh
pa/lna
Send and receive works OK
Below both are PA/LNA

RF24Network/examples/helloworld_tx/
STATUS = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1 = 0xcccccc3ccc 0xcccccc3c3c
RX_ADDR_P2-5 = 0x33 0xce 0x3e 0xe3
TX_ADDR = 0xcccccccc3c
RX_PW_P0-6 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA = 0x3f
EN_RXADDR = 0x3f
RF_CH = 0x5a
RF_SETUP = 0x01
CONFIG = 0x0f
DYNPD/FEATURE = 0x3f 0x05
Data Rate = 1MBPS
Model = nRF24L01+
CRC Length = 16 bits
PA Power = PA_MIN
Sending...ok.
Sending...ok.

RF24Network/examples/helloworld_rx/
STATUS = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5 = 0x33 0xce 0x3e 0xe3
TX_ADDR = 0xe7e7e7e7e7
RX_PW_P0-6 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA = 0x3f
EN_RXADDR = 0x3f
RF_CH = 0x5a
RF_SETUP = 0x01
CONFIG = 0x0f
DYNPD/FEATURE = 0x3f 0x05
Data Rate = 1MBPS
Model = nRF24L01+
CRC Length = 16 bits
PA Power = PA_MIN
Received packet #0 at 2000
Received packet #1 at 4000

It also works OK whdn both ends are basic+ module

here are small mods to sample code
void setup(void)
{
  Serial.begin(115200);
  Serial.println("RF24Network/examples/helloworld_tx/");
  printf_begin();

  SPI.begin();
  radio.begin();
  radio.enableDynamicPayloads();
  network.begin(/*channel*/ 90, /*node address*/ this_node);
  radio.setPALevel(RF24_PA_MIN);
  radio.enableDynamicPayloads();
  radio.printDetails();
}

Hope this helps
Peter

TMRh20, 2015-09-04:
Thanks for the info, I'll see if I can replicate issues with my PA+LNA modules, since I haven't been using them much as of late.

PeterSek, 2015-09-04:
Hi
Appreciate this
If you like me to run any tests just let me know. I am taking all hardware
with me next week when I am away from Melbourne
BTW I can use all PA modules but it would be beneficial if we have cross
TRX compatibility
Strange thing is that with recent test it partially started working one
direction: received gets it but sender things the transmission failed.
Could be timing or some wired confing being/or not being send to radio.
Also please also keep in mind all this worked OK prior to recent DEV merge
BTW I have some +mini modules (1x8with2mm pin spacing) I will try to check
these too
FYI I am away next week so response may be slow and time permitting I will
run a dif pre/post DEV merge
Once again thanks for looking into this
Peter

boguslawb, 2015-09-19:
Hi
Can you check this link ? https://www.squirrel-labs.net/blog/getting-the-best-out-of-nrf24l01/#page=1
It may be related and more important it may unlock all potential of  nrf24L01+PA/LNA module. Please let me know if your library is already optimised for this module. I checked code but I don't see if CE is kept high during all ACK. WHat I spotted is that in startFastwrite there is optional parameter to start with CE high but seems unused, because all invocations of this method do not use this param.

TMRh20, 2015-09-19:
The main issue here should now be fixed in the [development branch](https://github.com/TMRh20/RF24Network/archive/Development.zip)

@boguslawb 
Yes this issue is addressed in the library.
In standard write() function, CE is left high until ACK or timeout.

In all other (fastWrite, writeBlocking) except for startWrite() , the CE pin is left high until radio.txStandBy() is called, which drops the radio back to standby-I mode and sets CE LOW.

TMRh20, 2015-09-19:
@boguslawb  Also, what you spotted in startFastWrite is an option to write data to the radio FIFO, but not send. This is experimental, and allows users to write two or possibly three payloads to the radio before actually initiating transmission.

boguslawb, 2015-09-19:
Sorry, I'm quite new to the github, how is that development branch different to the pulling zip archive of them master ? Is that issue for LNA+PA module addressed in master repository ?

TMRh20, 2015-09-19:
> Sorry, I'm quite new to the github, how is that development branch different to the pulling zip archive of them master ?

The code has been updated. 

> Is that issue for LNA+PA module addressed in master repository ?

Yes, this main issue (https://github.com/TMRh20/RF24/issues/133) pertains to RF24Network only. The issue you mention with LNA + PA modules has to do with the core RF24 driver, and was fixed a long long time ago in a galaxy far far away.

boguslawb, 2015-09-19:
Just one more question here please .Did you tested improvement in LNA+PA range with modifications of your library against original RF24 from maniacbug ?

TMRh20, 2015-09-19:
Nope.

closed 2015-09-23 by TMRh20