nRF24/RF24 #134: Using multiple SPI-Devices next to the nrf24l01+

Oitzu, 2015-09-11:
I wanted to use multiple SPI-Devices on the SPI-Bus and had the problem that the lines
```
_SPI.setBitOrder(MSBFIRST);
_SPI.setDataMode(SPI_MODE0);
_SPI.setClockDivider(SPI_CLOCK_DIV2);
```
are problematic to other SPI-Devices.

Removing this lines solves the problem.

But the big question is:
Why is the rf24 changing bitrate, mode and clockdivider if this doesn't seem necessary? Or is it necessary in some cases?

See also: http://forum.mysensors.org/topic/1961/using-nrf24l01-with-other-devices-together-on-one-spi-bus/3

TMRh20, 2015-09-11:
Hmm.. with the introduction of the [SPI Transaction API](https://www.arduino.cc/en/Tutorial/SPITransaction) I guess this is not needed anymore. With the old API, the bitOrder, dataMode and clockDivider were set before every write, to allow different devices using different settings to operate at the same time on the SPI bus.

Technically, it used to be necessary,  but I'm thinking its probably getting called twice per write when the transaction code is being used. (Arduino IDE 1.6.1+) I'm curious as to why this would cause problems with other devices, since they should be using their required settings regardless of what RF24 is doing. All in all, those lines can probably be removed, but may affect users compiling without the Arduino IDE or using an older version.

Oitzu, 2015-09-11:
Hm... i don't know.. maybe other libs are trusting that the standard spi settings are used and fail because the rf24 lib sets the global spi settings to something else?

I tried this module: http://de.aliexpress.com/item/ArduCAM-M-2MP-Camera-Shield-OV2640-2MP-SPI-Camera-module/32307177889.html?ws_ab_test=201407_5,201444_5,201409_3

The lib doesn't seem to set any specific spi settings.

Other users complaining about problems with the cc3000 module, the W5100 ethernet module and spi touch screens.
Maybe some or all are related to this issue.

TMRh20, 2015-09-23:
It all has to do with the SPI settings I'm sure. In the [Updates branch]( https://github.com/TMRh20/RF24/archive/Updates.zip ] I've implemented a fix due to your comment, because technically this partly was/is a bug, if using Arduino IDE 1.6.3+ I think (that includes the transaction API )

This should be moved into master soon, but I am way behind on dealing with issues. (Sorry for the late reply btw)

Interestingly enough, the W5100 ethernet library as an example, is not yet using the SPI Transaction API, so if there are still problems between the two libraries while using the updates branch + Arduino IDE 1.6.5+ its probably a good idea to report an issue on the affected lib, and suggest an update to make use of the new SPI API.

Note [here](https://www.arduino.cc/en/Reference/SPISetBitOrder) it is suggested that all new libraries utilize the SPI transaction code.

Take a looks [here](http://dorkbotpdx.org/blog/paul/spi_transactions_in_arduino) for more info on the SPI transaction API.

closed 2015-11-06 by TMRh20