nRF24/RF24 #136: Malformed payloads

DrAtari, 2015-09-15:
I have a problem with malformed payloads with CRC-16 enabled (default). 

Maybe it's not a problem with your library, but you may know what I have to look for.

When I use 2 modules to communicate with eachother there are no problems. When a third module repeats the payload it received, then a malformed payload may be received by one of the modules.

For instance if I send the string "chro", the repeating MCU will receive the proper string and Serial.println() shows that it tries to send the right string. The receiving module will receive the string "chrï¿½" however. 

Or it repeats the string "d,1442332738,17.0,24.5", but the receiving module will receive this string: "d,1442#72738,17.ï¿½,:4.5" (note that it's showing the same characters as above: "ï¿½")

I tried setting autoack to false, but the problem persists. Data rate is RF24_250KBPS with dynamic payloads enabled. Also tried using flush_tx() after stopListening() before write(). 

Sometimes it looks like 2 seperate payloads get merged, sometimes only the first 7 characters of a 15+ character string get transmitted. 

Any idea how to fix this?

TMRh20, 2015-09-15:
>Any idea how to fix this?

Good detailed description of the problem, but without seeing some code, not really. Per [this](https://en.wikipedia.org/wiki/Wikipedia:Reference_desk/How_to_ask_a_software_question) no1 & 2 are good, but I need no3.

PhillyNJ, 2015-09-15:
Hi, may I ask if the code is being compiled using 2 different compilers?

Regards,

Phil

Sent from Phil's iPhone

DrAtari, 2015-09-16:
I've posted an example here:
http://pastebin.com/q4Ffq6TP

Yes, the code is compiled using 2 different compilers. The Teensy 3.1 and ATMega328 code is compiled with Arduino (with Teensyduino addon), the RPi code is compiled with gcc.

PhillyNJ, 2015-09-16:
Is the RPI the "receiver"? I suspect the issue is how each compiler
compiles "char const *txt".

DrAtari, 2015-09-16:
Yes, in this example it is. But both Teensy and RPi sometimes receive malformed payloads from the repeater. 

I'll try using a Arduino Pro Mini later, to rule out it's a hardware (capacitor) problem. 

What would you change about the char const *txt?

PhillyNJ, 2015-09-16:
I think its a compiler issue and nothing to do with the RF24. See this
thread
http://www.avrfreaks.net/forum/compiling-structsfloats-arduino-ide-vs-atmel-studio-7-beta-issue

I had a similar issue when passing a enum in a struct from a device that
was compiled in the Arduino IDE (receiver) and the Atmel Studio (sender).
Both use different compliers. My issue was related to how an enum was
compiled.

Have you tried just passing an a byte or int and see if its malformed?

I have had good luck using struct when passing data (no enums). It makes
for a great way to see who is the sender and how to handle the data
accordingly.

e.g.

typedef struct {

    int Id; // sender

    int Water;

    float Outside;

    float Battery;

    float Garage;

    float Temp;

}

Header_t;

friksa, 2015-09-16:
@DrAtari have you tried adding this line before startListening?

````
radio.txStandBy(timeoutPeriod);  //Partially blocking standby, blocks until success or max retries. FIFO flushed if auto timeout reached
````

TMRh20, 2015-09-17:
@DrAtari Your code is incomplete.

@friska It might need to be made clearer in the documentation, but ``` radio.txStandBy ``` is only needed when using write functions that leave the radio in Standby-II mode, like ``` radio.fastWrite() ``` and ``` radio.writeBlocking() ```

DrAtari, 2015-09-17:
Thanks for the answers. I'll provide a working code later, where irrelevant parts are removed. Using a Pro Mini instead of ATMega didn't change anything, so I think I can rule out a capacitor issue.

The most recent bug on the receiving side was that one module sent the payload "pir,ready", which was received by one of the receivers. But the other receiver received the payload "d,1442514", which is actually part of a payload it received a few minutes before. It has the same payload length as "pir,ready". 

So either there is something wrong with my code, or there is some problem with the reading pipes. Maybe not getting flushed properly after radio.stopListening() and radio.startListening()? It gets the right dynamic payload size of the current payload, but the data from a previous payload?

DrAtari, 2015-09-21:
Ok, here's the full code which reproduces the problem. It's enough to observe stdout on the RPi to observe it. Normally it should show these 4 log messages:

[19:54:42]: got init: pir,ready
[19:54:53]: got init: chro
[19:54:55]: unknown payload recieved: tb,234,234
[19:54:57]: sending data:d,1442801851,14,23.8 size:21

After letting it run for 7 minutes the RPi and/or the Teensy received some malformed packets (or something else went wrong):
[19:59:44]: unknown payload recieved: d,1442801 (this payload gets sent by the Teensy after it got the d,... payload from the RPi)
[19:59:34]: unknown payload recieved: temp,316 (this malformed? payload got sent by the Teensy)

RPi:
http://pastebin.com/4CrRi7qQ
Compile with
g++ -Ofast -mfpu=vfp -mfloat-abi=hard -march=armv6zk -mtune=arm1176jzf-s  -Wall -I../ -lrf24-bcm -lwiringPi -std=c++11 fixme_rpi.cpp -o fixme_rpi

ATmega328:
http://pastebin.com/6ELCpQP3
(compiled with Arduino IDE)

Teensy 3.1:
http://pastebin.com/A7QMNRGf
(compiled with Arduino IDE with Teensyduino)

TMRh20, 2015-09-22:
@DrAtari Thanks for sharing. :p The most clear and present problem has to do with your addressing. To put it simply, all listening pipes on all devices must be unique. Technically, there are a number of minor issues, which require a bit of a detailed understanding of the radio modules to address. 

Depending on your goals, I would suggest taking a look at RF24Network and/or RF24Mesh, since they were intended to allow more complex communication scenarios, without having to spend time reading through the datasheet, and understanding addressing, retries, and how the nrf chip identifies valid payloads. 

As a rule with this library, (and computers in general) they basically either work or do not work. If they do not work, it is usually not the fault of the hardware or the driver software. Issues like data corruption at the RF24 layer are *usually* an obvious code problem.

DrAtari, 2015-09-22:
You mean the fact that 2 devices open the same reading pipe could be a problem? Would it change something if I take care of acks manually? Maybe it's less work than using RF24Network/Mesh. Will have a look at it though. Thanks.

TMRh20, 2015-09-22:
Yeah, the only time you want to share reading pipes is with auto-ack disabled and/or using selective acknowledgments when writing to those addresses. (note: with mixed modules there can be problems when using dyn payloads + dyn acks )

> Would it change something if I take care of acks manually? Maybe it's less work than changing the code to use RF24Network/Mesh.

Yes and no? Per above, you can share reading pipes if not using acks. One of the main reasons I started working on building an RF24 communication stack was to understand the OSI model, and why everything is built using these layers. In the end, there are many very good reasons.

IMHO, it is usually easier in the end to work with "the appropriate layer to suit your goals". 

For example, you will also generally want to at least utilize some sort of 'header' for the payload including something like a counter that increments every send and ensures that the CRC is different each time. You are likely have missing payloads otherwise, especially if repeating duplicate payloads, with multicast etc...

DrAtari, 2015-09-23:
Ok. I guess I'll switch to RF24Mesh then, to avoid a mess and easily add more modules later.

If one module in the mesh is switched off, the payloads will still reach the target, right (given they are in range)? And every module is addressable with a fixed address?

Does the RF24Mesh lib cause traffic while not using radio.networkavailable() or other methods? As I prefer to only have them send data when my code tells them to send data.

TMRh20, 2015-09-24:
> If one module in the mesh is switched off, the payloads will still reach the target, right (given they are in range)? I mean will the network find a new route, if one module in the old route is missing? 

As long as there is a path, basically, yes. The main idea of RF24Mesh is that nodes will regularly check for connectivity, so if a node goes down or is moved, other nodes can find a new path.

The matter of verifying connectivity etc, is left mostly up to you, via the checkConnection() and renewAddress() functions. You typically want to call checkConnection() at defined intervals (say every 30-seconds to a minute, or longer even) so that if the master node goes offline, it can be left offline for a min. or two, and all nodes will reconnect once started.

The other potential failure, is if a node goes offline or is moved. This can also be handled by the timed poll above, or you could also keep track of failed writes, and verify connectivity after every failure, or a defined number of failures.

See the ongoing [discussion here](https://github.com/TMRh20/RF24Mesh/issues/27#issuecomment-142598798)

> Does the RF24Mesh lib cause traffic while not using radio.networkavailable() or other methods? As I prefer to only have them send data when my code tells them to send data.

Per above, it is entirely up to you how traffic is managed, and when data is sent. Any node will only send data when either:
1. Your code initiates traffic.
2. Another node is relaying traffic through it. (Handled in mesh.update())

> Also, one of my modules only gets its power turned on for a few seconds every 10 minutes. Will it be able to send the data to the master node in the mesh easily?

This should work fairly easily, although this is something I haven't tested extensively. You mostly just want to release the address of the node going down, so the master is not assigning children to that node. When powered up, the address would need to be renewed.

closed 2015-10-20 by Avamander