nRF24/RF24 #139: Particle Core/Photon implementation

rlogiacco, 2015-09-22:
The parent library from maniacbug has been succesfully ported for the Particle Core / Photon, a wifi enabled device based on STM32 (https://github.com/technobly/SparkCore-RF24).

I wonder if you are interested into making that port _official_ as I believe this is now considered the main RF24 library. I volunteer to provide the testing ground on both Core and Photon devices if you don't own them.

TMRh20, 2015-09-22:
Yeah, I'm happy to add any additional hardware support here, but coding it can be very difficult without the hardware. 

I'm not very familiar with these devices, but it looks like there is an [Arduino core](https://github.com/rogerclarkmelbourne/Arduino_STM32) available, so the RF24 driver would detect it similar to a Due I think. Without the Arduino core, I assume more changes would be needed.

rlogiacco, 2015-09-23:
I honestly believe you will understand much more from having a look at the repo I posted above. The Particle Core and Photon are much different from Arduino, they are proper IoT devices running a firmware which maintains the device connected to a private cloud. The most common way to compile their software is via an online editor and compiler which pushes the new firmware over the air...

I understand developing without the corresponding hardware is near impossible, but in this case you can program the thing on your PC and run on my hardware :+1: 

I believe it will be easier if you just have a look... And in case you don't believe it's feasible I can keep using the porting mentioned above (which originates from maniacbug's RF24 as well). My intention was to consolidate the platforms under one umbrella, but it is not mandatory at all :wink:

TMRh20, 2015-09-25:
I did look at the repo, but didn't realize the full nature of the device I guess. One of the main goals of this repo is to support as wide a range of hardware as possible, in the easiest manner so no complaints there. ( I do need to take a hard look at the portability aspect and how that can be cleaned up, because I tend to learn as I go a lot of the time, and its a bit messy again )

The code seems very similar, so I'm thinking its just a matter of modifying a few defines etc. and a matter of learning enough to understand the make file and just make the thing work I guess.

I'll see how soon I can throw some code together for this, if you really want me in control of your device lol. If you need to take discussion offline, feel free to use tmrh20@gmail.com but for development purposes, I like to have a record of the issues and what was discussed on github.

Also, I tend to be a bit blunt most of the time, but thanks a lot for all of your help and contributions lately! Very much appreciated.

stewarthou, 2015-10-30:
Hi @rlogiacco, I just ported this library for Photon. you can check here [An implementation of TMRh20's optimized RF24 Library for the Particle Photon.](https://github.com/stewarthou/Particle-RF24)

I have published to Particle Build Web IDE, you should be able to find it in Community Libraries soon, but in the mean time, include it manually.

I'm still including more examples from this library to the ported version. but you can try the [GettingStarted](https://github.com/stewarthou/Particle-RF24/blob/master/firmware/examples/GettingStarted/GettingStarted.ino) now.

If you discover any issues, please report.

rlogiacco, 2015-10-30:
Thanks, I'll give it a try very soon!

closed 2015-11-25 by TMRh20

TMRh20, 2015-11-25:
@stewarthou I hope you don't mind, but I added a link to your fork to the RF24 documentation.