nRF24/RF24 #143: RF24 1.01 is not shown in Arduino IDE

harryboo, 2015-10-01:
Hi,

version 1.01 is not shown in arduino ide and gives the following message:

'arch' folder is no longer supported! See http://goo.gl/gfFJzU for more information

version 1.0 is working without problems.

Thanx

TMRh20, 2015-10-01:
Hmm, v1.0 is missing a libraries properties file, so I thought it wouldn't show up in the library manager? That probably explains why it works though??

To address this, I attempted another release as v1.02 using the updates branch for testing it. I renamed the 'arch' directory and related references to 'utilitiy' per the pre-1.5 rules, so the lib should remain backward compatible with older IDE versions. **Instead of RF24/arch it is now RF24/utility**

The 1.6.5 IDE seems to be picking up RF24 in library manager, but it seems to have the same problem, and doesn't display. 

According to [this](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list) the new release should be picked up automatically at some point, so I think its a matter of waiting to see...

TMRh20, 2015-10-01:
Ok, scratch that all.

It was getting too messy with different version numbers with no code changes  I realized I could just delete the old releases and tags, so I recreated v1.0 and v1.0.1 with separate tags and properties files. I'm assuming they will be picked up properly now???

harryboo, 2015-10-01:
Thanx, this works now. Also the examples are now displayed.

closed 2015-10-01 by harryboo