nRF24/RF24 #144: Python 3 compatible ?

doudz, 2015-10-11:
Hi, 
I'm trying to compile the python wrapper on Rpi for python 3.4 (on raspbian jessie).
I successfully compile it doing 
python3 setup.py build
sudo python3 setup.py install
 but I tried to "import RF24" I got the following error
```
ImportError: /usr/lib/arm-linux-gnueabihf/libboost_python-py27.so.1.55.0: undefined symbol: PyClass_Type
```
How use the right libboost extension ?

TMRh20, 2015-10-11:
There was a thread that talked about it, but I cannot seem to find it.

Currently only python 2.7 is supported, with a bit of info [here]( http://tmrh20.github.io/RF24/Python.html )  ( Technically, you only need the libboost python module, not all as per the info )

doudz, 2015-10-12:
Thank you for this answer.
But the problem is the system is taking libboost_python-py27.so.1.55.0 instead of libboost_python-py34.so.1.55.0
Maybe because libboost_python.so is a symbolic link to libboost_python-py27.so.1.55.0
I think there's something missing somewhere to force using py34 instead of the default py27

doudz, 2015-10-12:
accord to this site http://www.joernesdohr.com/tutorials/using-the-rf24-library-for-nrf24l01-on-raspberry-pi-with-python-3/
we need to use boost_python3 instead of boost_python
so I update setup.py like this :

```
from distutils.core import setup, Extension
import sys

BOOST_LIB = 'boost_python'
if sys.version_info >= (3, ):
    BOOST_LIB = 'boost_python3'

module_RF24 = Extension('RF24',
            libraries = ['rf24-bcm', BOOST_LIB],
            sources = ['pyRF24.cpp'])

setup(name='RF24',
    version='1.0',
    ext_modules=[module_RF24]
      )
```
and since libboost_python3.so is missing (on raspbian jessie), I link libboost_python-py34.so.1.55.0 to libboost_python3.so manually
```
ln -s libboost_python-py34.so.1.55.0 libboost_python3.so
```
after that I compile and install
```
python3 setup.py build
sudo python3 setup.py install
```
And it works ! :)
```Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RF24
>>> RF24
<module 'RF24' from '/usr/local/lib/python3.4/dist-packages/RF24.cpython-34m.so'>
>>> 
```

TMRh20, 2015-10-12:
Nice! If you are comfortable using github etc. feel free to submit your additions via a pull request, else I think you've provided enough info to replicate this. I haven't gotten to try it out myself, but definitely will as soon as I can.

mz-fuzzy, 2015-10-12:
Just for reference: https://github.com/TMRh20/RF24Network/issues/44

There are issues with unicode strings in python3, I'm not sure whether, beside RF24Network, it concerns also RF24, I'd say so. So fixing of this is probably also needed for having python3 supported.

mz-fuzzy, 2015-10-23:
I've prepared support for python3 in my forked repo:
https://github.com/mz-fuzzy/RF24/tree/python3

What is handled there:
- using right boost lib as @doudz suggested
- issue with using 'string' for message handling as pionted out by @joernesdohr - this needed to be adapted for python3; now, 'bytearray' is used ('bytes' also accepted on input). 
- backward compatibility with python2 is kept; if you pass/expect strings, they should be converted implicitly
- if you have old python RF24 scripts, they should run as they are in python2
- example pingpair_dyn.py adapted so it can run on both python2 and python3

It is now tested only by dry runs as I don't have RPi currently available. I'd like to ask python3-support-thirsty RF24 followers for cooperation - please try to fetch my 'python3' branch, build, install and test, and please report your results to me, thanks.

joernesdohr, 2015-10-23:
Thanks for your efforts @mz-fuzzy! I don't have access to my hardware over the weekend, but I'll take your branch out for a spin next week as soon as possible.

doudz, 2015-10-23:
Good news ! I'll try it this morning

doudz, 2015-10-23:
It works well on my "pure" python implementation of MySensors library.
I only check reading data, I'll check writing soon.
Thanks you !!

TMRh20, 2015-10-31:
@mz-fuzzy Tested & seems to be working great for me, just had to link libboost_python3.so per @doudz info above. Thanks!

TMRh20, 2015-11-14:
The [python documentation](http://tmrh20.github.io/RF24/Python.html) looks like it could use a bit of attention, but I'm so not the expert here. I can give it a shot, but figure it might work out better with input from folks with better python skills. It can be edited by modifying the text at the bottom of the RF24.h file if anybody is up for it.

joernesdohr, 2015-11-14:
Sorry for the delay. I can verify that the adapted RF24 lib works with Python 3.2. I'm currently looking into adapting the changes to RF24Network. (see https://github.com/TMRh20/RF24Network/issues/44#issuecomment-156603332 for this)

closed 2015-12-14 by Avamander