nRF24/RF24 #148: Detaching this library from maniacbug's

Avamander, 2015-10-20:
@TMRh20 

Should I or should I not contact Github to detach this "fork" from maniacbug's? This library is so far ahead it does not deserve the name "fork".

Github's help page says this:"To detach the fork and turn it into a standalone repository on GitHub, contact GitHub support."

t3chguy, 2015-10-20:
Not that my opinion matters much in this scenario but I agree that this library isn't really a fork anymore, and therefore should not be labelled as such.

TMRh20, 2015-10-20:
I didn't even know that was a thing, but it seems reasonable to me if that's what you both think.

Avamander, 2015-10-20:
Support says that unfortunately you @TMRh20 have to contact Gihub support.

TMRh20, 2015-10-20:
Hmm, you'd think they would be able to use my agreement here, but request sent anyway.

Avamander, 2015-10-20:
It has been done.

closed 2015-10-20 by Avamander

TMRh20, 2015-10-20:
FYI, github support (wow, that was fast) suggested an extraction, which also brings all the children of this fork along for the ride, so I went with that.

t3chguy, 2015-10-20:
I would like to thank @Avamander and @TMRh20 and any other contributors and collaborators for supporting such a useful Library. I have only had minor use of it but have bought a batch of various RF24L01+ Modules to play with and know that I have everything I need available because of your efforts.

Thanks

fab672000, 2015-10-20:
I also agree that this version is so much more than a fork now, great job @TMRh20 and @Avamender !