nRF24/RF24 #150: Deleting unused branches

Avamander, 2015-10-20:
Would it be possible that you'd, @TMRh20, delete old unused branches in this repository?

Avamander, 2015-10-20:
I deleted those few branches that had not been touched four years.

iamthemovie, 2015-10-23:
I think you deleted something that was an updated documentation?

Avamander, 2015-10-23:
@iamthemovie 

I deleted nothing related to documentation, if I did, do tell what exactly and where it was and I'll restore it. 

FYI the branches that do not exist on this repo compared to maniacbug's were:
* interrupts
* p_enable
* write_bug  
* arduino-1.0

iamthemovie, 2015-10-23:
Sorry sorry panicing, I'm not sure what's going on but the GH Pages Docs have changed completely changed for me, I used to see a table with the pinouts on this page (which differed from the Manicbug repo) and the examples folder reflected the documentation:

[image]

Is not reflected here:

[image]

And this page (which used to look different and have the table with pinouts and where they go):

http://tmrh20.github.io/RF24/

looks like maniacbugs original GH page?

Avamander, 2015-10-23:
This seems to be unrelated to my changes. Because I really did not touch anything related to documentation. You can compare the [old repo](https://github.com/Avamander/RF24/branches/all) and [new repo](https://github.com/TMRh20/RF24/branches/all)  you can see the four branches I removed. 

The documentation page has definitely changed, (see this)[https://github.com/TMRh20/tmrh20.github.io/blob/master/RF24/index.html] for example, it was generated this year. I think this might have been caused by the "deforking" that was done. I can try removing the gh-pages branch that seems to override the newer documentation page.

iamthemovie, 2015-10-23:
Sorry if it's unrelated, I just assumed it might have been something to do with this. I guess you could try that yes! That example looks like the page from the beginning of the week :)

iamthemovie, 2015-10-23:
(Thanks by the way)

Avamander, 2015-10-23:
Paging @TMRh20, it seems that somehow the old documentation is overriding the new documentation, could you please investigate?

TMRh20, 2015-10-23:
I was just about to post on how odd this seems lol, considering that github.io is its own repository. 

If you clone the repo, or just look at it, all the documentation is correct in the [actual source code area](https://github.com/TMRh20/tmrh20.github.io) 

I'll try regenerating the documentation with doxygen, and re-push it to github maybe, and see if the system picks it up correctly hehe.

Avamander, 2015-10-23:
Try disabling gihub pages from this repo's settings?

Avamander, 2015-10-23:
It seems that the website has updated itself to the latest version.

iamthemovie, 2015-10-23:
Thanks guys :) Also, random question, are either of you in any IRC channels just for chatting about this library - I do love it.

TMRh20, 2015-10-23:
@avamander Regarding your prev. question, the only branch I would want to keep is the updates branch, since I like to use it for any changes that may require additional testing etc. before moving into master, kind of like a development branch.

@iamthemovie Not really, right now I'm barely keeping up with this kind of stuff.

Avamander, 2015-10-23:
@TMRh20 Thanks for your assistance. It seems all OK now.

closed 2015-10-23 by Avamander

Avamander, 2020-10-13:
@TMRh20 Is there a problem?

TMRh20, 2020-10-13:
Yes I’m just putting things on pause as I wrap some things up.

Avamander, 2020-10-13:
@TMRh20 

> Yes I’m just putting things on pause as I wrap some things up.

Do the libraries have to be put on pause or could I continue to keep an eye on them?