nRF24/RF24 #156: SPI bus with two transceivers

santi-pdp, 2015-11-02:
Hello,

I am working on a project where I try to make two transceivers work in the same SPI bus in a Raspberry pi A+. The problem I am facing now is that both transceivers seem to block each other when running the gettingstarted_example in separate processes. As it can be seen in the following logs, both devices are connected to different GPIO and CE at their CE and CSN respectively:
================ SPI Configuration ================
CSN Pin          = CE0 (PI Hardware Driven)
CE Pin           = Custom GPIO22
Clock Speed      = 8 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xe7e7e7e7e7 0xc2c2c2c2c2
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0xe7e7e7e7e7
RX_PW_P0-6       = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x03
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX
###############################################
================ SPI Configuration ================
CSN Pin          = CE1 (PI Hardware Driven)
CE Pin           = Custom GPIO27
Clock Speed      = 8 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xe7e7e7e7e7 0xc2c2c2c2c2
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0xe7e7e7e7e7
RX_PW_P0-6       = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x03
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX 

When both devices are executed they block and no output is available, either at receiver or at transmitter.Any clue of what can be happening or what approach should be taken to make both devices work in the same Raspberry pi?

Thanks in advance

TMRh20, 2015-11-02:
> The problem I am facing now is that both transceivers seem to block each other when running the gettingstarted_example in separate processes.

The SPI BUS is managed via the CS pins to choose which device is active. If running in two separate processes, you need to ensure only one device accesses SPI at once. This may be easier using SPIDEV. I've never tested this though, dual radios has only been tested with a single process accessing both devices.

t3chguy, 2015-11-02:
I'd imagine the best solution would be to write a sort of middleware application which would manage the SPI bus on behalf of the other two processes, queueing actions to be handled and also returning data when it is ready. It would be much easier within a single process though.

closed 2015-11-06 by TMRh20

santi-pdp, 2015-11-06:
Thank you guys, it was because of that indeed. I am using both devices within the same process in a successful way now.