nRF24/RF24 #157: Question: Cross-Compilation for RPi

iamthemovie, 2015-11-02:
I've been looking into setting up my compiler tool chain on my Mac to support compiling for the RPi2. I've had a quick look at the Makefile and I'm just wondering if this codebase is designed solely for compilation on the Pi itself and if anyone does cross compilation on Linux / Windows / Mac for the broadcom chip on either of the Pis so far?

TMRh20, 2015-11-02:
The Makefile is designed for compilation on the RPi/Linux. Mainly, it looks for the BCM hardware in /proc/cpuinfo but this could easily be modified. I don't have much experience regarding cross compilation on other devices though.

iamthemovie, 2015-11-02:
Yeah I saw that, I might take a crack at it and see where I get to. Thanks for the info :)

iamthemovie, 2015-11-02:
Did you do anything significant to the BCM2835 library?

TMRh20, 2015-11-03:
Nope. I can't remember exactly without doing a compare, but there would only be very minimal changes if any. This is probably a good reminder to update the BCM driver sometime soon too.

iamthemovie, 2015-11-05:
Happy to look at updating it if you want. I pulled version 1.46 and integrated into a my own now heavily modified RPi specific version of this repo but I can try port it back in if you want?

TMRh20, 2015-11-05:
Sure, I'm also curious about your 'heavily modified RPi specific version'. Contributions are always welcome.

iamthemovie, 2015-11-05:
It's heavily modified in the sense that I hate not knowing how things work so I tend to take them apart and put them back together so I don't feel like "magic" is happening. I love this library so I just wanted to play around with it, removed all the compiler directives for everything but RPi/ARM/Linux, updated the bcm library version to 1.46 and ensured I could compile it on my mac :) working on my Pi2 so far, just in the middle of testing it.

TMRh20, 2015-12-18:
Thanks, but it looks like the bcm driver has already hopped a couple more versions, so I updated to v1.48  https://github.com/TMRh20/RF24/commit/18daa4c4ddaef1c5bb3048c0a891c45030955c46

closed 2015-12-18 by TMRh20