nRF24/RF24 #158: Update python wrapper to use minimal RF24 constructor

TMRh20, 2015-11-06:
Reminder to update python wrapper to use minimal RF24 constructor, for use/compatibility with both BCM and SPIDEV usage.

mz-fuzzy, 2015-11-13:
@TMRh20, any plan with this? I guess best would be to check whether RF24 on the system is compiled with BCM support (is that evident from RF24.h?) and depending on that include/exclude BCM-related stuff in the python wrapper?

TMRh20, 2015-11-13:
Heh, no real plan, just a rough idea. My thoughts initially revolved around removing all the BCM specific stuff from the C++ & python examples, and fixing all the documentation to refer to the standard linux constructor ( like RF24 (22,0) ) while still actually supporting the BCM initialization, just that it would become a mostly undocumented option.

This would mainly be to allow some transition time and prevent more confusion around the constructor. Eventually, I think it's probably a good idea to remove the BCM specific stuff altogether, and stick with GPIO + SPI-BUS numbers for Linux devices.

mz-fuzzy, 2016-01-27:
Python wrapper now exposes BCM specific stuff only if it's in C++ RF24, 95926b2c9aeedd1cdae011a1bb0f54f40b9a7726

mz-fuzzy, 2016-02-17:
From my point of view it's resolved, closing.

closed 2016-02-17 by mz-fuzzy