nRF24/RF24 #159: AES library

WERAQS, 2015-11-07:
has anyone successfully encrypted the packages with http://forum.arduino.cc/index.php?topic=88890.0 ?

Avamander, 2015-11-08:
I doubt it. But I'm planning to look at signing packets like MySensors is doing.

iamthemovie, 2015-11-08:
I have been dabbling with TEA, http://www.movable-type.co.uk/scripts/tea.html for sending data.

WERAQS, 2015-11-08:
http://www.libelium.com/ sensors using the same kind of encryption, I'm thinking about it but have to combine, encrypt, send, decrypt and break the data to get the results. I have to code something unique for the package layout I think, because whenever I try to send more than 4 bytes on a tab delimited package it fails. (ie. 9999 ok, 10001 nope.)

Avamander, 2015-11-08:
I think signing packets would be a better idea, just like MySensors do.

spaniakos, 2015-11-08:
i have a repo using AES that encryps packets, i dotn have a example of RF24 and AES online , but i have some on my laptop that i used to encrypt using AES/3DES/SHA1/SHA256/MD5

iamthemovie, 2015-11-08:
Signing packs = good call either way :)

pat1, 2015-11-13:
see at https://github.com/TMRh20/RF24Network/pull/3

closed 2015-12-13 by Avamander